JMRI/JMRI #3105: LE Java Exception for Block Content Labels

dsand47, 2017-02-20:
While testing JMRI 4.7.1 I was able to create a java exception.

[java exception.txt](https://github.com/JMRI/JMRI/files/786512/java.exception.txt)

The error is easy to re-create:

1. Create a new block contents label or remove the text from an existing one.
2. Right click on the block contents label and select **Properties**.
3. Click on **OK**, **Apply** or **Cancel**.

The **Cancel** stack is a bit different, but it has the same first line.

If the block has a value, the error does not occur.  

Dave Sand

geowar1, 2017-02-20:
Excellent… (It wasn't me; you didn't see me; you can't prove a thing… ;-)
(git blame FTW!)
I'll get the fix submitted ASAP. Thanks.

geowar1, 2017-02-20:
Ok, the fix is in PR#3084
Regress when you can and let me know if we can close this.

dsand47, 2017-02-20:
I will check it out once it is in 4.7.2ish.

In the meantime, take a look at issue #3107.

Dave Sand

dsand47, 2017-03-07:
The error no longer occurs at 4.7.2ish.

Dave Sand

closed 2017-03-07 by dsand47