JMRI/JMRI #3107: Layout Editor issues at 4.7.1

dsand47, 2017-02-20:
I have identified a number of issues at 4.7.1:
* Ghost layout blocks — I don’t know if these are an issue or something that shows up due to close monitoring during testing.
* **Failure to create LayoutBlock** errors while drawing the track segments with no block selected.
* When a track segment is deleted and re-added using the block name, the corresponding layout block is not saved, resulting in startup errors.
* The edit mode block contents icon should be visible after the add.  It should not require a save/stop/start sequence.

The attached zip file contains a PDF describing the test cases in detail, the test case xml files and a python script that checks the current state of layout blocks.

[Block Issues 4.7.1.zip](https://github.com/JMRI/JMRI/files/788465/Block.Issues.4.7.1.zip)

Dave Sand

geowar1, 2017-02-21:
Can you provide steps to reproduce each of these? I'm not understanding or able to reproduce the issues.

•Ghost layout blocks: I assume phantom layout blocks are showing up in the blocks table that aren't in your layout?
Playing around I was getting phantom turnouts & sensors; but only if my layout was powered up. It looks like JMRI was creating turnouts and sensors based on messages from my layout…?¿?

•Failure to create LayoutBlock errors while drawing the track segments with no block selected.

Unable to reproduce… (I created a few anchors and connected them with track segments without a block selected; deleted and re-added them (w/new block names typed in textfield)  and I got no errors. Saved & restored same… )

•When a track segment is deleted and re-added using the block name, the corresponding layout block is not saved, resulting in startup errors.

Nothing is saved unless you save your profile or panel… but this hasn't changed from previous clients. As best as I can follow your instructions… I'm seeing the new block in the saved panel. And it's re-created when the panel is re-loaded.

•The edit mode block contents icon should be visible after the add. It should not require a save/stop/start sequence.

Likewise this has always been the case… We've been discussing this issue on the developers forum IIRC since January 22nd. See: GitHub issue #2910.

dsand47, 2017-02-21:
I will cover the first two items tonight.  Refer to the attached screen shots.

**step 1a.png**
This is starting point.  I have defined sensors, blocks, some Logix and have a script running.  You can see that there are no layout blocks defined.

**step 1b.png**
I switch to edit mode, place two end bumpers and two anchor points.  I then draw the first track segment.  The first error occurs.

**step 1c.png**
I do the remaining track segments.  2 more errors.  I also check the layout block state.  Still none.

**step 1d.png**
I return to display mode.  We now have layout blocks.  Why?
The only action between 1c and 1d is switching the panel from edit to display mode.  The following steps will clear the layout blocks:
1. Starting in display mode, save/stop/start.  See **step 1e.png**.
2. While still in display mode, save/stop/start.  See **step 1f.png**.
The layout blocks are gone.  The ghosts appear during the transition from edit to display.

BTW:  For LocoNet connections I know that sensors and turnouts can be created by JMRI doing an initial poll.  I use RR-CirKits xxxxMan cards and these always report status during start up.  Since the layout is 45 miles away, I normally work in simulation mode.

[step 1.zip](https://github.com/JMRI/JMRI/files/789174/step.1.zip)

Dave Sand

dsand47, 2017-02-21:
This covers the 3rd issue:  The missing layout block.

**step 2a.png**
This is the starting point using **Block Test 5.xml**.  I have completed the panel configuration which includes blocks assigned for each track segment, color coding, occupancy sensor icons and block content labels.

**step 2b.png**
The plan is to add a second track segment to B-Blk-3.  The first step is to delete the track segment so that an anchor point can be inserted before the end bumper.  The layout block list is still valid.

**step 2c.png**
2 track segments have been added with block name **B-Blk-3** in the combo box.  The alternate track color is displayed due to the new **_selected block_** feedback.

**step 2d.png**
I have not been able to turn off the alternate track color feedback.  Removing the block name from the combo box does not work.  Here I just selected another block.  You can see that layout block B-Blk-3 is still valid since the track color is blue.  Another method that works is exit/enter edit mode.

**step 2e.png**
The panel xml file has been saved. The current contents:
[image]
I have included the entire xml file with the screen shots.

My theory is that when the track segment was deleted, a **_do not save_** flag was set for layout block B-Blk-3.  When it was subsequently used for the added track segments, the **_do not save_** flag was not cleared.

**step 2f.png**
When PanelPro is re-started, the missing layout block errors occur since the track segments point to a layout block that is no longer present.

**step 2g.png**
After manually updating the track segments, everything is good.

[step 2.zip](https://github.com/JMRI/JMRI/files/790588/step.2.zip)

Dave Sand

geowar1, 2017-02-21:
There is no "do not save" flag; The block logic maintains a "use count"; if on save that value is zero then the block isn't saved. My guess is that that use count isn't being bumped by the sensor use.

> I have not been able to turn off the alternate track color feedback

Yeah, I need to add a menu item to enable/disable that behavior (default=disabled).
If you want to write that up as a separate issue please do; I'll take assignment and fix that.

geowar1, 2017-02-21:
> For LocoNet connections I know that sensors and turnouts can be created by JMRI doing an initial poll. I use RR-CirKits xxxxMan cards and these always report status during start up. 
I use these as well… really like them over any of the alternatives I've tried before. ;-)

dsand47, 2017-02-21:
I have added getUseCount() to my script.  Very interesting.

When the XML file is loaded, the use count is correct.  When a track segment is deleted, the use count is decremented.  When a track segment is added to a block using either the block option on the tool bar or after the fact using the track segment edit option the use count is **NOT** incremented.

Sample scenario:
1. Layout block with 2 track segments, count 2.
2. Delete **track segment 2**, count 1.
3. Add **track segment 2** back with the block selected, count 1.
4. Delete **track segment 2**, count 0.

But one **track segment 1** still refers to the layout block.  

Dave Sand

dsand47, 2017-02-21:
I just discovered another problem.

**block add 1.png**
This is the initial startup using **Block Test 0.xml**.

**block add 2.png**
I add a block manually to the block table.
- A layout block is created with a use count of 1.
- 2 Failure to create error messages are created.

There appear to be some very strange logic paths.

[new block.zip](https://github.com/JMRI/JMRI/files/791575/new.block.zip)

Dave Sand

dsand47, 2017-03-07:
These issues have been resolved.

Dave Sand

closed 2017-03-07 by dsand47