JMRI/JMRI #3109: XML Panel failed at SignalGroup

silverailscolo, 2017-02-20:
Failed loading, stating:
```
/Users/me/Library/Preferences/JMRI/Testpanel_RG1965rev1d.xml Exception: org.jdom2.input.JDOMParseException: Error on line 315: cvc-complex-type.2.4.b: The content of element 'signalgroup' is not complete. One of '{appearance, aspect, signalHead}' is expected.
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
org.jdom2.input.JDOMParseException: Error on line 315: cvc-complex-type.2.4.b: The content of element 'signalgroup' is not complete. One of '{appearance, aspect, signalHead}' is expected.
```
Line 315 etc. reads:
```
  <signalgroups class="jmri.managers.configurexml.DefaultSignalGroupManagerXml">
    <signalgroup systemName="IG2" userName="Seinmast 2# knipper" signalMast="NS Testseinmast">
      <systemName>IG2</systemName>
      <appearance valid="Limited" />
      <appearance valid="Restricting" />
    </signalgroup>
  </signalgroups>
```
That implies at least 1 SignalHead is expected, but when creating a SigGroup the user is free to NOT select one, save the file and voila.
I think it would be best (easiest) to set min=0 for SignalHeads in SignalGroups (such a Group won't do much, but some users might wish to add the head at a later moment, save first and hope the Panel willl open to let them finish set up).
The alternative would be to force the user to pick a (singleoutput) SignalHead when creating an SignalGroup.
Panel file Testpanel_RG1965rev1d.xml attached (renamed to .txt)
[Testpanel_RG1965rev1d.xml.txt](https://github.com/JMRI/JMRI/files/788642/Testpanel_RG1965rev1d.xml.txt)

bobjacobsen, 2017-02-20:
I think I've been working the same issue, see comments in #3110

bobjacobsen, 2017-02-21:
Looking at some test cases, I think that "no signalheads" is sort-of whacky, not really the purpose of this feature, but will still work.  So I recommend allowing the schema to accept zero there.

closed 2017-03-12 by bobjacobsen

silverailscolo, 2017-03-13:
This still fails in 4.7.2ish when opening a configuration containing a newly created SignalGroup without a SignalHead, but including an aspect:
``The content of element 'signalgroup' is not complete. One of '{aspect, signalHead}' is expected.``

Relevant part:
```
  <signalgroups class="jmri.managers.configurexml.DefaultSignalGroupManagerXml">
    <signalgroup systemName="R12" userName="mygroup" signalMast="virtmast1">
      <systemName>R12</systemName>
      <aspect valid="Clear" />
      <aspect valid="Approach" />
    </signalgroup>
  </signalgroups>
```
The systemName element might be the cause.

Should I open a new Issue instead? Don't see a Reopen button in GitHub.
[teststoregroup.xml.txt](https://github.com/JMRI/JMRI/files/837515/teststoregroup.xml.txt)

bobjacobsen, 2017-03-13:
Thanks for the test file!  I'll add a new PR shortly to cover this. No need to open a new Issue.

bobjacobsen, 2017-03-13:
Should we respin 4.7.2 for this? (I lean to yes)

KenC57, 2017-03-13:
Since we've not got the builds too far afield yet, I'd say yes too, just to
cover as many things as possible.

-ken c