JMRI/JMRI #3110: XML Panel failed at SignalMastRepeeater

silverailscolo, 2017-02-20:
When loading this panel at statup I receive a different warning (see Issue #3109) but validation from within DP flags:
``Value 'BothWay' is not facet-valid with respect to enumeration (...)``

[image]
Now, BothWay seems to be in the list, but the returns withing the single quotes look strange.
This Repeater may not be worth much, as I was just testing the interface at the time (May 2016, release 4.3.5).
The XML fragment concerned reads:
```
    <signalmastrepeater>
      <masterMast>i11</masterMast>
      <slaveMast>drv4B</slaveMast>
      <enabled>true</enabled>
      <update>BothWay</update>
    </signalmastrepeater>
```

bobjacobsen, 2017-02-20:
The problem is that the schema is based on `xs:string`, but should be based on `xs:token`:
```
          <xs:restriction base="xs:token">
            <xs:enumeration value="BothWay"/>
            <xs:enumeration value="MasterToSlave"/>
            <xs:enumeration value="SlaveToMaster"/>
          </xs:restriction>
```
The hint is how the value was expanded with newlines while being matched to an `xs:string`. I'll put together a PR.

bobjacobsen, 2017-02-20:
There are more problems in that file:
```
The content of element 'signalgroup' is not complete. One of '{appearance, aspect, signalHead}' is expected.
```
This file says it was written by JMRI 4.3.5. Has it been edited manually?

bobjacobsen, 2017-02-20:
So this might be a code problem.

The signalgroups-2-9-6.xsd schema requires that there be at least one `signalHead` element inside a `signalgroup` element, but that's not present in the sample file:
```
  <signalgroups class="jmri.managers.configurexml.DefaultSignalGroupManagerXml">
    <signalgroup systemName="IG2" userName="Seinmast 2# knipper" signalMast="NS Testseinmast">
      <systemName>IG2</systemName>
      <appearance valid="Limited" />
      <appearance valid="Restricting" />
    </signalgroup>
  </signalgroups>
```

The `jmri/managers/configurexml/DefaultSignalGroupManagerXml.java` configXML `store(..)` code stores the SignalHeads that are contained, but I don't know whether there's always guaranteed to be at least one.  Is that a contract that should be enforced by the schema?

The `load(..)` code seems to be happy if there are no signalHead elements: it checks for that, and proceeds if there are none.

Somebody who understands this part of the code should comment:  Should there always be at least one signalHead? Now?  In past versions?

bobjacobsen, 2017-02-21:
The part above about signalgroup and zero signalhead elements is covered in more detail in #3109

PR #3113 (2017-02-22): Fix Schema for signalmastrepeater Element
Closes #3110
closed 2017-02-21 by silverailscolo