JMRI/JMRI #3126: Can't independently invoke some migrated PackageTests

bobjacobsen, 2017-02-23:
`./runtest.csh jmri.jmrix.pi.PackageTest` runs to completion, but doesn't run any tests.

It looks like `jmri.jmrix.pi.PackageTest` is somehow partially migrated from JUnit3 to JUnit4.

I think it's seeing that there's a `main(..)`, and successfully launching it, but then something goes wrong.

pabender, 2017-02-23:
@bobjacobsen,

That particular package actually never existed as a JUnit3 package.

The problem here isn't that the tests aren't running, but there isn't any output when the tests succeed. 

This is a logging level issue that I haven't taken the time to figure out.  If there is an error, then the log.error message in the for loop prints output.  The log.info statement "Success" doesn't ever print.

If you pull the code in #3127, you'll see I added some System.out.println statements to the class.  Those actually print.

I'm not going to merge the code that way, but it does illustrates the problem.

Paul

bobjacobsen, 2017-02-23:
Thanks! I think I understand, but let me check.

With older packages, the behavior was something like this:
```
% ./runtest.csh jmri.jmrix.nce.PackageTest
.......Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
..................................
.........................................
.........................................
..........
Time: 4.616

OK (133 tests)
```

So that's what I was expecting.  With this package, it's running the tests silently:
```
% ./runtest.csh jmri.jmrix.pi.PackageTest
Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
```

But it's really running the tests still.  To confirm that, I changed a test to deliberately fail and got:
```
% ./runtest.csh jmri.jmrix.pi.PackageTest
Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
ERROR - testDefaultSystemName(jmri.jmrix.pi.RaspberryPiSensorManagerTest): real object returned  [main] jmri.jmrix.pi.PackageTest.main()
```
Which is a different format than before, but carries all the information (though I'll miss the feedback a little).

rhwood, 2017-02-23:
Just out of curiosity what is the reason for not using the TextRunner in the main method for some tests, but leaving it in others? I think that if we used it uniformly everywhere, we'd have a consistent experience for all tests.

pabender, 2017-02-23:
On 02/23/2017 06:08 AM, Randall Wood wrote:
> Just out of curiosity what is the reason for not using the TextRunner in
> the main method for some tests, but leaving it in others? I think that
> if we used it uniformly everywhere, we'd have a consistent experience
> for all tests.

Because there isn't a TextRunner for JUnit4 that I can find.

The textui runner used elsewhere is a JUnit3 object.

Lately when I've been building a new PackageTest, I've actually been 
leaving the main out altogether.

The JUnit4 style PackageTest classes that have a main (like the 
jmri.jmrix.pi.PackageTest class Bob started this discussion with) were 
some of the earlier ones I converted.

Like @bobjacobsen, I missed the feedback, but I think leaving the main 
out to get the feedback is only a short term solution.  Eventually we 
probably want to discontinue using JUnit3 at all.

Paul

rhwood, 2017-02-24:
Didn't realize a TextRunner was no longer available. A different question is this:

The JUnit4 runner (```org.junit.runner.JUnitCore.main()```) produces the output @bobjacobsen is expecting. Why do the some main() methods in JUnit4 test classes suppress that output?

(That is, if I change
```
// Main entry point
     static public void main(String[] args) {
         org.junit.runner.Result result = org.junit.runner.JUnitCore
                  .runClasses(PackageTest.class);
         for(org.junit.runner.notification.Failure fail: result.getFailures()) {
             log.error(fail.toString());
         }
          //junit.textui.TestRunner.main(testCaseName);
          if (result.wasSuccessful()) {
              log.info("Success");
          }
      }
```
to
```
    static public void main(String[] args) {
        org.junit.runner.JUnitCore.main(PackageTest.class.getName());
    }
```
I get the expected output.)

pabender, 2017-02-24:
On 02/24/2017 05:58 AM, Randall Wood wrote:
> The JUnit4 runner (|org.junit.runner.JUnitCore.main()|) produces the
> output @bobjacobsen <https://github.com/bobjacobsen> is expecting. Why
> do the some main() methods in JUnit4 test classes suppress that output?

<snip>

> |static public void main(String[] args) {
> org.junit.runner.JUnitCore.main(PackageTest.class.getName()); } |
>
> I get the expected output.)

Because I didn't realize you could run it that way.

Paul

closed 2017-07-03 by bobjacobsen