JMRI/JMRI #3132: Suspicious comparison

AppChecker, 2017-02-24:
Hi!

Please look these code fragments:
 [java/src/jmri/managers/JmriUserPreferencesManager.java](https://github.com/JMRI/JMRI/blob/d833bd7deac6059398ff9434798e7d8a82edcc31/java/src/jmri/managers/JmriUserPreferencesManager.java#L1156-L1158)

```
                    if (window.getAttribute("locX") != null && window.getAttribute("locX") != null) {
                        double x = window.getAttribute("locX").getDoubleValue();
                        double y = window.getAttribute("locY").getDoubleValue();
                        ....
                    }
```
window.getAttribute("locX")  is checked twice.
Probably it should be:
```
                    if (window.getAttribute("locX") != null && window.getAttribute("locY") != null) {
                        double x = window.getAttribute("locX").getDoubleValue();
                        double y = window.getAttribute("locY").getDoubleValue();
                        ....
                    }
```

This possible defect found by[AppChecker](https://cnpo.ru/en/solutions/appchecker.php)

PS
There are two places, where left and right side of binary expression is same:
[java/src/jmri/jmrit/display/layoutEditor/LayoutEditorFindItems.java, line 119-120](https://github.com/JMRI/JMRI/blob/d833bd7deac6059398ff9434798e7d8a82edcc31/java/src/jmri/jmrit/display/layoutEditor/LayoutEditorFindItems.java#L119-L120)
[java/src/jmri/jmrit/display/layoutEditor/LayoutEditorFindItems.java, line 143-144](https://github.com/JMRI/JMRI/blob/d833bd7deac6059398ff9434798e7d8a82edcc31/java/src/jmri/jmrit/display/layoutEditor/LayoutEditorFindItems.java#L143-L144)

But I think in this case it is normal.

geowar1, 2017-02-24:
On Feb 24, 2017, at 3:21 AM, appchecker <notifications@github.com> wrote:
> Please look these code fragments:
> java/src/jmri/managers/JmriUserPreferencesManager.java <https://github.com/JMRI/JMRI/blob/d833bd7deac6059398ff9434798e7d8a82edcc31/java/src/jmri/managers/JmriUserPreferencesManager.java#L1156-L1158>
>                     if (window.getAttribute("locX") != null && window.getAttribute("locX") != null) {
>                         double x = window.getAttribute("locX").getDoubleValue();
>                         double y = window.getAttribute("locY").getDoubleValue();
>                         ....
>                     }
> window.getAttribute("locX") is checked twice.
> Probably it should be:
> 
>                     if (window.getAttribute("locX") != null && window.getAttribute("locY") != null) {
>                         double x = window.getAttribute("locX").getDoubleValue();
>                         double y = window.getAttribute("locY").getDoubleValue();
>                         ....
>                     }
> This possible defect found byAppChecker <https://cnpo.ru/en/solutions/appchecker.php>AppChecker FTW!

> PS
> There are two places, where left and right side of binary expression is same:
> java/src/jmri/jmrit/display/layoutEditor/LayoutEditorFindItems.java, line 119-120 [...]
That doesn’t look right… seems redundant (you can say that again… but don’t) but from the code I can’t tell what was intended…
Looking thru the git history for those lines… they’ve always been that way… (no-one “broke” it ;-)

Looks to me like someone copied the lines like 178-9 (that checked both directions):

                if (p.getEastBoundSignal().equals(bean.getSystemName())
                        || p.getWestBoundSignal().equals(bean.getSystemName())) {

and changed it to only check one direction… twice… ;-)

I have to wonder if the compiler (tokenizer? what’s the “Java” term here?) optimizes that out.
-- 
Enjoy,
George Warner, (408)249-7533
eMail/iChat: <geowar1@mac.com>

bobjacobsen, 2017-02-24:
> if (window.getAttribute("locX") != null && window.getAttribute("locX") != null) {
> double x = window.getAttribute("locX").getDoubleValue();
> double y = window.getAttribute("locY").getDoubleValue();

That first line is doing the null-check to prevent an NPE in the 2nd and third lines.  But it’s got a typo in it.

FindBugs probably flags this, either as the 3rd line is an unprotected reference or the 1st line is redundant.

bobjacobsen, 2017-03-13:
The section of LayoutEditorFindItems highlighted above is a pile of inconsistent conventions. Sometimes a variable is compared against both a system name and a user name. Sometime just against the system name.  The pieces highlighted:

```
                if (p.getEastBoundSignal().equals(bean.getSystemName())
                        || p.getEastBoundSignal().equals(bean.getSystemName())) {
```
which looks like a cut&paste error when trying to set up "match on system name or user name".  But that's not certain, because in other places `getEastBoundSignal` is only compared to a system name.  A bit of consistency would be good here...

KenC57, 2017-08-15:
PR #3931 fixed these lines.

closed 2017-08-15 by KenC57