JMRI/JMRI #3136: Path objects duplicated in 4.7.1

pnbgit, 2017-02-25:
I have loaded the panel I am working on into 4.7.1.  I was 
working on it using JMRI 4.6. It loaded ok and seems to function as normal.

However, when I saved the panel, without making any changes, into a new 
file, I noticed that the new XML file has had about 750 lines added by 
4.7.1 compared to the one saved by JMRI 4.6, increasing from 12480 to 
13226 lines - about a 6% increase.

Doing a diff compare, most of the additions seem to duplicate path 
objects within block objects, for example:

   <block systemName="IB111" length="0.0" curve="0" userName="Down Main 
Longfield Av App">
       <systemName>IB111</systemName>
       <userName>Down Main Longfield Av App</userName>
       <permissive>no</permissive>
       <occupancysensor>BD: Down Main Longfield Av 
Approach</occupancysensor>
       <path todir="64" fromdir="128" block="IB43" />
       <path todir="128" fromdir="64" block="IB42" />
       <path todir="128" fromdir="64" block="IB42" />
       <path todir="64" fromdir="128" block="IB43" />
  </block>

The second two path lines were added by the update to 4.7.1, and 
duplicate the two path objects that were already there.

I also found some of the duplicated path objects had beansetting 
objects.  I don't know what these are for, but again they just duplicate 
previous path entries:

   <block systemName="IB117" length="0.0" curve="0" userName="Fiddle 
Relief 13">
       <systemName>IB117</systemName>
       <userName>Fiddle Relief 13</userName>
       <permissive>no</permissive>
       <path todir="64" fromdir="128" block="IB116">
         <beansetting setting="2">
           <turnout systemName="Fiddle Up Relief Arrival 13" />
         </beansetting>
       </path>
       <path todir="128" fromdir="64" block="IB126" />
       <path todir="128" fromdir="64" block="IB126" />
       <path todir="64" fromdir="128" block="IB116">
         <beansetting setting="2">
           <turnout systemName="Fiddle Up Relief Arrival 13" />
         </beansetting>
       </path>
  </block>

Again, the second two path objects have been added by 4.7.1 and 
duplicate the first two.

I can upload my panel if nobody else can duplicate this problem, but it does reference quite a lot of custom icons which will need to be copied to the right place for the panel to load without errors.

bobjacobsen, 2017-02-25:
The last time this came up, the fix was to make sure the `equals(..)` and hash operations were doing the right thing.  IIRC, some of the code was expecting 'same object' for equality, and some was expecting 'same content'.  First step might be to look for recent changes in that area of the code (and/or write or check the tests)

pnbgit, 2017-02-26:
Is there somebody who can take a look at this?  As a complete Newbie who hasn't yet got any familiarity with the JMRI code, or even worked out how to build it yet, I'm afraid I wouldn't have a clue where to start looking - I posted this in the hope that my testing feedback would be useful.  I guess the only clue is that it must be something that has been changed since 4.6, as the problem wasn't there then.

bobjacobsen, 2017-02-27:
See #2939, #2976 and the much earlier #678

bobjacobsen, 2017-02-27:
Also #682 which seems directly on point.

geowar1, 2017-04-01:
Can you regress this on 4.7.2 or .3? If not please close; if so please comment to that effect. TIA.

closed 2017-05-07 by bobjacobsen