JMRI/JMRI #3150: Panel will not load on test ver. 4.71

JLang1951, 2017-02-27:
I thought that I would join Git Hub so I could post this problem related to Test version 4.71 on loading my panel files. I’ve attached one of my layout editor files that you can use as a test. This panel file has worked over a number of years up to ver. 4.57. I tried this panel when ver. 4.6 came out and that’s where the problems started so I reverted back to 4.57 and all was well again so I tried yesterdays test release of 4.71 to see if things were fixed but the panel still won’t load.

System setup: NCE simulated, CMRI simulated.

JMRI system console while opening on v.4.57:
[Console v4.57.txt](https://github.com/JMRI/JMRI/files/803918/Console.v4.57.txt)
JMRI system console while opening on v.4.71 before loading panel:[console v4.71 no panel load.txt](https://github.com/JMRI/JMRI/files/803920/console.v4.71.no.panel.load.txt)
JMRI system console while opening on v.4.71 after loading panel:[Console v4.71 Panel load.txt](https://github.com/JMRI/JMRI/files/803931/Console.v4.71.Panel.load.txt)
Panel file:
[J_Lang_LE_panel.zip](https://github.com/JMRI/JMRI/files/803947/J_Lang_LE_panel.zip)

If there is any other thing that I can do or test please let me know as I am more than willing to help.
J. Lang

bobjacobsen, 2017-02-27:
Thank you for all the information!

In addition to the panel, there's a problem with loading some of the decoder definition files.  This is a big unexpected.  The files that show a problem are:
```
Pane_wow_snd.xml
wowDieselCV.xml

enum-primemover.xml
enumDiesel04.xml
```
The first two of those are files we distribute with JMRI (although your versions might be different). We don't (currently) distribute files by the second names.

They might be in the JMRI program directory `C:\Program Files (x86)\JMRI\xml `, or in a user directory, maybe `C:\Users\BillyBob\JMRI\`.

If you could find those and add them to this PR, I'd be happy to look at that too.  They're probably messing up your ability to use the DecoderPro part of JMRI.

Thanks again

bobjacobsen, 2017-02-27:
For others reading this, the problem is this element:
```
    <layoutturnout ident="TO1" turnoutname="NT283" blockname="CP 2831" blockbname="CP 2831" blockcname="CP 2832" blockdname="CP 2832" type="5" connectaname="T28" connectbname="T29" connectcname="T46" connectdname="T27" continuing="2" disabled="no" disableWhenOccupied="no" xcen="200.0" ycen="60.0" xa="170.0000457763672" ya="50.0" xb="209.9999747412262" yb="50.0" xc="229.9999542236328" yc="70.0" xd="190.0000252587738" yd="70.0" ver="1" class="jmri.jmrit.display.configurexml.LayoutTurnoutXml">
      <signalAMast>2831E</signalAMast>
      <signalDMast>2832E</signalDMast>
      <sensorA>CP2831W</sensorA>
      <sensorD>CP2832W</sensorD>
    </layoutturnout>
```
and ones like it.  The `EditorLayoutTurnoutType` schema (in `xml//schema/types/editors.xsd`) doesn't have a definition for the `<sensorD>` element even though `java/src/jmri/jmrit//display/layoutEditor/configurexml/LayoutTurnoutXml.java` reads and writes it.

JLang1951, 2017-02-27:
As far as the wow decoder defs. in decoder pro mmosher added the TCS Wow diesel v4 definitions to GitHub last month.
[tcs-wow-diesel-v4.zip](https://github.com/JMRI/JMRI/files/804221/tcs-wow-diesel-v4.zip)
I installed those according to the readme file as I had some of those decoders to program so that's probably why they showed up in the console. They work fine when I added them to v.4.57. Thanks for looking into this.
J.Lang

bobjacobsen, 2017-02-27:
Thanks for the info on the TCS files.  Will check them as part of merging #2906

mmosher5501, 2017-02-27:
enumDiesel04.xml
enum-primemover.xml
are new include files

Pane_wow_snd.xml
wowDieselCV.xml
were revised, along with
CV1_CV99_wow.xml

referenced by new file
TCS_WOW_Diesel04.xml
 with small change to 
TCS_WOW_steam04.xml  

Work and validate in 4.6

bobjacobsen, 2017-02-27:
Agreed.  #3153 includes those and validates just fine. I'm hoping that installing a new (development) version with them will fix this for @JLang1951

JLang1951, 2017-02-27:
Bob, Just an FYI on my end I edited  EditorLayoutTurnoutType  in   xml//schema/types/editors.xsd and added the  sensorD element. All my panels now seem to work just fine now. Just wanted to let you know. Thanks for all you do.
J.L.

bobjacobsen, 2017-02-28:
Thanks for your help with this.

There's a development build at http://jmri.tagadab.com/jenkins/job/Development/job/Packages/1312/ that has both the schema fix, and updated versions of the TCS files.  When you get a chance could you try that and see if (a) the validation problem is still gone and (b) the messages about the decoder files are gone?  

Thanks again.

JLang1951, 2017-02-28:
Yes everything has cleared up on the console after I uninstalled the previous version and installed the new build. All my panels are working fine and the decoders showed up after a rebuild decoder index. Again thanks for fixing this and to mmosher for providing the new TCS files.
John Lang

closed 2017-02-28 by JLang1951