JMRI/JMRI #3156: Print Roster entry fails

silverailscolo, 2017-02-28:
In DP4.7.1 on macOS _and_ Win7, selecting an old or newly created roster entry and choosing File menu > Print preview > Selected Roster Entry, silently fails with ``InvocationTarget java.lang.reflect.InvocationTargetException``. Will check this out later today.
Console shows:
```
2017-02-28 10:35:04,482 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-02-28 10:35:04,499 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\e.broerse\JMRI\log\messages.log [main]
2017-02-28 10:35:04,500 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\me\JMRI\log\session.log [main]
2017-02-28 10:35:04,505 apps.Apps                             INFO  - PanelPro version 4.7.1-R1b7dd65 starts under Java 1.8.0_121 on Windows 7 x86 v6.1 at Tue Feb 28 10:35:04 CET 2017 [main]
2017-02-28 10:35:06,093 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3ed13931 [main]
2017-02-28 10:35:06,510 node.NodeIdentity                     INFO  - Using jmri-5CC5D4543D91-3ed13931 as the JMRI Node identity [main]
2017-02-28 10:35:06,565 jmrix.ConnectionConfigManager         INFO  - No local configuration found. [main]
2017-02-28 10:35:08,194 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-13]
2017-02-28 10:35:08,204 configurexml.ConnectionConfigXml      INFO  - CAN Simulator Started [main]
2017-02-28 10:35:08,511 logix.WarrantPreferences              INFO  - Found Warrant preferences file: C:\Users\me\JMRI\My_JMRI_Railroad\signal\WarrantPreferences.xml [main]
2017-02-28 10:35:08,661 util.HelpUtil                         WARN  - JavaHelp: File help/nl/JmriHelp_nl.hs not found, dropping to default [main]
2017-02-28 10:35:08,761 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2017-02-28 10:35:08,761 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\me\JMRI\My_JMRI_Railroad\ [main]
2017-02-28 10:35:08,761 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\me\JMRI\My_JMRI_Railroad\ [main]
2017-02-28 10:35:08,762 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\me\JMRI\ [main]
2017-02-28 10:35:08,762 util.FileUtilSupport                  INFO  - File path home: is C:\Users\me\ [main]
2017-02-28 10:35:08,762 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
InvocationTarget java.lang.reflect.InvocationTargetException
2017-02-28 10:36:24,092 jmrit.XmlFile                         INFO  - No C:\Users\me\JMRI\My_JMRI_Railroad\roster\test471.xml file to backup [AWT-EventQueue-0]
InvocationTarget java.lang.reflect.InvocationTargetException
``

bobjacobsen, 2017-03-01:
Added better logging to java/src//jmri/util/swing/GuiUtilBase.java and now see the traceback (which goes through a bunch of GUI code) is due to:
```
     [java] 	at jmri.jmrit.roster.PrintRosterEntry.<init>(PrintRosterEntry.java:135)
     [java] 	at jmri.jmrit.roster.swing.RosterFrame.printLoco(RosterFrame.java:873)
     [java] 	at jmri.jmrit.roster.swing.RosterFrame.remoteCalls(RosterFrame.java:887)
```

bobjacobsen, 2017-03-01:
Sorry for the bad debug info in the original, for some reason that was sending just a bit to System.out instead of the whole exception to logging.

silverailscolo, 2017-03-02:
Still not really working, as an  empty list displays:
[image]

bobjacobsen, 2017-03-02:
What programmer file do you have defined?  That looks like either one isn't selected, or the "Basic" has been selected on a decoder that doesn't fill those out.

silverailscolo, 2017-03-02:
When opened from within a roster entry (Basic tab shown) a long list of all tabs appears in this dialog (new layout is my PR; the checkbox "Function List" is added separately  by this method). Not at the train Mac, will debug some more later.

bobjacobsen, 2017-03-02:
"Print entry" seems to print through the context of the selected programmer definition I.e. Basic or Advanced. If you've got Basic selected, it won't print all CVs.

silverailscolo, 2017-03-03:
Indeed "Basic" seems to be passed as the programming method, though I haven "Comprehensive" selected in Prefs. It then tries reading from xml programmer, but in contrast to calling PrintPanes() from within an opened programmer, nothing is available in the RawPaneList when trying to print (preview) a selection from the Roster pane. I suspect a length of 1 for this container is an bug resulting from the while-has loop, as it's an empty list of panes (tabs).
PrintRosterEntry tries to read it using jmri.jmrit.symbolicprog.configurexml so I'm wondering (will have to inspect a programmer SML file) why no "pane" elements are loaded.

silverailscolo, 2017-03-04:
Fixed in PR #3166. It was a case of XML structure being different than expected, but error check not noticing.

PR #3163 (2017-03-01): Fix roster entry print fail when no pane name
Closes #3156 

Print would fail when one or more panes didn't have a specific name.  Also improved some logging.
closed 2017-03-05 by silverailscolo