JMRI/JMRI #3159: LayoutEditor feedback from JMRI 4.7.1

bobjacobsen, 2017-02-28:
From JMRIusers:

The Layout Editor in PanelPro is the one that I am **least likely** to use, so this comment may have a simple response. Only one of our 22 panels uses that editor and the Ops guy who created it has asked for a modification. Since I was deep into that task, it seemed a good place for comparison. 

In version 4.7.1 the “Tool Bar” layout seems to be better arranged with a logical classification that what is available in 4.6 and what I can remember from previous sallies into this realm. Providing the new option to move the Tool Bar to a side or the bottom of the Panel window is a really nice feature. 

The expanded list of Zoom options and remembering the Zoom setting from one session to the next are both great too. 

However, the font size for both the “Tool Bar” and the “Help Bar” is too small for me to read. Changing the display font size in the Preferences window, currently set to 20, does not impact a change in these two bars. Back to 4.6 at least for this effort or until this puzzle is unraveled. 

My screen is listed at 17 inches with a 1920 by 1980 pixel size and in the Windows Settings options menu, reads “Change the size of text, apps, and other items: 150%” to both help me read the usual messages and enlarge the icons on the touch screen so that my fat-finger can often only hit one of the choices rather than several. 

When the position of the Tool Bar is changed, the JMRI System Console indicates a font size for the Layout Editor as shown here: 

2017-02-28 11:27:01,997 util.Log4JUtil INFO - ****** JMRI log ******* [main] 
2017-02-28 11:27:02,012 util.Log4JUtil INFO - This log is appended to file: C:\Users\cliffaa-rr\JMRI\log\messages.log [main] 
2017-02-28 11:27:02,012 util.Log4JUtil INFO - This log is stored in file: C:\Users\cliffaa-rr\JMRI\log\session.log [main] 
2017-02-28 11:27:02,028 apps.Apps INFO - PanelPro version 4.7.1-R1b7dd65 starts under Java 1.8.0_121 on Windows 10 amd64 v10.0 at Tue Feb 28 11:27:02 PST 2017 [main] 
. . 
. 
2017-02-28 11:28:29,316 layoutEditor.LayoutEditor INFO - Font size: 9.0 [AWT-EventQueue-0] 

Thus far, I have not been able to figure out what is needed at this end to change that font size, and in 4.6 that need never arises. 

Thanks for all the good work and positive support. 

Cliff Anderson in Baja SoCal

geowar1, 2017-02-28:
On Feb 28, 2017, at 12:45 PM, Bob Jacobsen <notifications@github.com> wrote:
> The Layout Editor in PanelPro is the one that I am least likely to use, so this comment may have a simple response. Only one of our 22 panels uses that editor and the Ops guy who created it has [...]
I didn’t like that changing that preference forced a re-launch of the app… so I ignored it and put a “ToolBar Font Size” menu item in the “Options” menu. Depending on what build you have you may or may not see that…

Try this one and let me know what you think: <https://github.com/JMRI/JMRI/pull/3145 <https://github.com/JMRI/JMRI/pull/3145>>.
> My screen is listed at 17 inches with a 1920 by 1980 pixel size and in the Windows Settings options menu, reads “Change the size of text, apps, and other items: 150%” to both help me read the usual messages and enlarge the icons on the touch screen so that my fat-finger can often only hit one of the choices rather than several.
> 
> 2017-02-28 11:28:29,316 layoutEditor.LayoutEditor INFO - Font size: 9.0 [AWT-EventQueue-0]
> 
> Thus far, I have not been able to figure out what is needed at this end to change that font size, and in 4.6 that need never arises.
> 
Check your “Options” menu for a “ToolBar” sub-menu… if you don’t have it you have an older build… ;(

Thank you for your feedback… While I have opinions (read “Mac Bias”) about how things should be I recognize that it’s not _MY_ app… so other users opinions matter. Period!

So please feel free to speak up and let me know how I may improve your experience.

-- 
Enjoy,
George Warner, (408)249-7533
eMail/iChat: <geowar1@mac.com>

Cliff-Anderson, 2017-07-30:
Please forgive me for letting this slide.  

As of version 4.9.1 the Options -> ToolBar -> list includes "ToolBar Side >" and "Wide ToolBar" and "Dropdown Lists Display Order...>" but no "ToolBar Font Size" option.

However, the current sizes of the font in the two toolbars do appear very readable.  

Perhaps this issue is moot, especially since nobody else seems to have noticed it.

Thanks,
Cliff

geowar1, 2017-09-22:
The toolbar fonts item was disabled as per [](https://github.com/JMRI/JMRI/pull/3145#issuecomment-283940658).

geowar1, 2017-10-29:
Can this be closed?

closed 2017-10-30 by bobjacobsen