JMRI/JMRI #3160: MemoryIcons with text not properly rotating

bobjacobsen, 2017-02-28:
From jmriusers:

When I display a Memory Variable with a text value with no rotation it displays and updates fine. If I then rotate the variable through 90 degrees (angle doesn't seem to matter), the text is rotated. However, as soon as the value is updated the text is updated but goes back to zero degrees.

If a panel is saved and then reload all values are shown at zero degrees. The xml saves the rotation value ok (degrees="90" in the memoryicon segment), but doesn't seem to use it

<memoryicon memory="IM7" x="160" y="160" level="4" forcecontroloff="false" hidden="no" positionable="true" showtooltip="true" editable="true" degrees="90" size="12" style="1" red="0" green="0" blue="0" hasBackground="no" justification="left" selectable="no" class="jmri.jmrit.display.configurexml.MemoryIconXml" defaulticon="program:resources/icons/misc/X-red.gif">
<tooltip>IM7 (IM:AUTO:0007)</tooltip>
</memoryicon>

Connection to a block (in order to implement train tracking) does not seem to make any difference.

I must be missing something here, as I've seen this working on Steve's YouTube video https://www.youtube.com/watch?v=ykUPOtYVn-o https://www.youtube.com/watch?v=ykUPOtYVn-o

bobjacobsen, 2017-03-01:
> PanelPro version 4.7.1-R1b7dd65 starts under Java 1.8.0_121 on Linux i386 v4.8.0-39-generic

mstevetodd, 2017-03-01:
In the video referenced, I (SteveT) am not using MemoryIcons, I am using BlockContentsIcons.

petecressman, 2017-03-09:
This is not reproducible with a MemoryIcon in either Panel Editor or Control Panel Editor.  Someone familiar with Layout Editor should look into this.

closed 2021-04-14 by stale[bot]