JMRI/JMRI #3171: Error 302 Reading Indexed CVs with DCS240

dheap, 2017-03-02:
A user has reported Error 302 "programmer busy" when attempting to read Indexed CVs with a DCS240.
https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/136143
https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/136411

User-submitted LocoNet Monitor: 
```
DecoderPro read CV16.2.257 via DP in Direct Byte mode
Programming Request: Write Byte in Direct Mode on Service Track to CV31 
value 16 (0x10, 10000).
LONG_ACK: The Slot Write command was accepted.
Programming Request: Write Byte in Direct Mode on Service Track to CV32 
value 2 (0x2, 10).
LONG_ACK: The Slot Write command was rejected.
Programming Track Request: Read Byte in Direct Mode on Service Track 
variable CV257 value 255 (0xff, 11111111).
LONG_ACK: The Slot Write command was rejected.
Programming Response: Write Byte in Direct Mode on Service Track to CV31 
value 16 (0x10, 10000).
```

It appears the programmer is sending the CV32 Write command before the Programming Response from the  CV31 Write command has been received, causing the "programmer busy" message.

I don't have access to the hardware.

Dave

bobjacobsen, 2017-03-03:
It would be really helpful to get a trace with the "raw data" turned on.  What _might_ be happening is that the DCS240 (unlike earlier units) will send a reply message when the write is complete (unlike the previous "blind write").  Perhaps bits in the LONG_ACK will specify that, or perhaps we'll need some command station - specific code.

If there's nothing different about the LONG_ACK, the next will be to get the bytes from the initial Programming Request and send them from the Send Packet tool.  That'll let the trace show the full sequence of whatever the DCS240 is going to send back.

dheap, 2017-03-03:
I don't have access to DCS240 hardware so we may have to get back to the original poster to see if he is prepared to do further tests and logs.

ValPistilli, 2017-03-06:
Here is the data from Write full sheet. DCS240

[EF 0E 7C 6B 00 00 00 00 00 1C 23 7F 7F 36]  Programming Request: Write Byte in Direct Mode on Service Track to CV29 value 35 (0x23, 100011).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[BB 7F 00 3B]  Request Command Station Ops Switches.
[E7 0E 7C 6B 00 00 02 47 00 1C 23 7F 7F 7B]  Programming Response: Write Byte in Direct Mode on Service Track to CV29 value 35 (0x23, 100011).
[EF 0E 7C 6B 00 00 00 00 00 42 02 7F 7F 49]  Programming Request: Write Byte in Direct Mode on Service Track to CV67 value 2 (0x2, 10).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 42 02 7F 7F 04]  Programming Response: Write Byte in Direct Mode on Service Track to CV67 value 2 (0x2, 10).
[EF 0E 7C 6B 00 00 00 00 00 43 05 7F 7F 4F]  Programming Request: Write Byte in Direct Mode on Service Track to CV68 value 5 (0x5, 101).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 43 05 7F 7F 02]  Programming Response: Write Byte in Direct Mode on Service Track to CV68 value 5 (0x5, 101).
[EF 0E 7C 6B 00 00 00 00 00 44 07 7F 7F 4A]  Programming Request: Write Byte in Direct Mode on Service Track to CV69 value 7 (0x7, 111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 44 07 7F 7F 07]  Programming Response: Write Byte in Direct Mode on Service Track to CV69 value 7 (0x7, 111).
[EF 0E 7C 6B 00 00 00 00 00 45 0C 7F 7F 40]  Programming Request: Write Byte in Direct Mode on Service Track to CV70 value 12 (0xc, 1100).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 45 0C 7F 7F 0D]  Programming Response: Write Byte in Direct Mode on Service Track to CV70 value 12 (0xc, 1100).
[EF 0E 7C 6B 00 00 00 00 00 46 10 7F 7F 5F]  Programming Request: Write Byte in Direct Mode on Service Track to CV71 value 16 (0x10, 10000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 46 10 7F 7F 12]  Programming Response: Write Byte in Direct Mode on Service Track to CV71 value 16 (0x10, 10000).
[EF 0E 7C 6B 00 00 00 00 00 47 15 7F 7F 5B]  Programming Request: Write Byte in Direct Mode on Service Track to CV72 value 21 (0x15, 10101).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 47 15 7F 7F 16]  Programming Response: Write Byte in Direct Mode on Service Track to CV72 value 21 (0x15, 10101).
[EF 0E 7C 6B 00 00 00 00 00 48 1A 7F 7F 5B]  Programming Request: Write Byte in Direct Mode on Service Track to CV73 value 26 (0x1a, 11010).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 48 1A 7F 7F 16]  Programming Response: Write Byte in Direct Mode on Service Track to CV73 value 26 (0x1a, 11010).
[EF 0E 7C 6B 00 00 00 00 00 49 1E 7F 7F 5E]  Programming Request: Write Byte in Direct Mode on Service Track to CV74 value 30 (0x1e, 11110).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 49 1E 7F 7F 13]  Programming Response: Write Byte in Direct Mode on Service Track to CV74 value 30 (0x1e, 11110).
[EF 0E 7C 6B 00 00 00 00 00 4A 23 7F 7F 60]  Programming Request: Write Byte in Direct Mode on Service Track to CV75 value 35 (0x23, 100011).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4A 23 7F 7F 2D]  Programming Response: Write Byte in Direct Mode on Service Track to CV75 value 35 (0x23, 100011).
[EF 0E 7C 6B 00 00 00 00 00 4B 28 7F 7F 6A]  Programming Request: Write Byte in Direct Mode on Service Track to CV76 value 40 (0x28, 101000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4B 28 7F 7F 27]  Programming Response: Write Byte in Direct Mode on Service Track to CV76 value 40 (0x28, 101000).
[EF 0E 7C 6B 00 00 00 00 00 4C 2F 7F 7F 6A]  Programming Request: Write Byte in Direct Mode on Service Track to CV77 value 47 (0x2f, 101111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4C 2F 7F 7F 27]  Programming Response: Write Byte in Direct Mode on Service Track to CV77 value 47 (0x2f, 101111).
[EF 0E 7C 6B 00 00 00 00 00 4D 33 7F 7F 77]  Programming Request: Write Byte in Direct Mode on Service Track to CV78 value 51 (0x33, 110011).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4D 33 7F 7F 3A]  Programming Response: Write Byte in Direct Mode on Service Track to CV78 value 51 (0x33, 110011).
[EF 0E 7C 6B 00 00 00 00 00 4E 3A 7F 7F 7D]  Programming Request: Write Byte in Direct Mode on Service Track to CV79 value 58 (0x3a, 111010).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4E 3A 7F 7F 30]  Programming Response: Write Byte in Direct Mode on Service Track to CV79 value 58 (0x3a, 111010).
[EF 0E 7C 6B 00 00 00 00 00 4F 41 7F 7F 07]  Programming Request: Write Byte in Direct Mode on Service Track to CV80 value 65 (0x41, 1000001).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4F 41 7F 7F 4A]  Programming Response: Write Byte in Direct Mode on Service Track to CV80 value 65 (0x41, 1000001).
[EF 0E 7C 6B 00 00 00 00 00 50 48 7F 7F 11]  Programming Request: Write Byte in Direct Mode on Service Track to CV81 value 72 (0x48, 1001000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[... cut ...]
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 02 5C 5B 7F 7F 41]  Programming Response: Write Byte in Direct Mode on Service Track to CV93 value 219 (0xdb, 11011011).
[EF 0E 7C 6B 00 00 00 00 02 5D 7F 7F 7F 29]  Programming Request: Write Byte in Direct Mode on Service Track to CV94 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 02 5D 7F 7F 7F 64]  Programming Response: Write Byte in Direct Mode on Service Track to CV94 value 255 (0xff, 11111111).

ValPistilli, 2017-03-06:
Here is Read Full sheet

[EF 0E 7C 2B 00 00 00 00 02 1C 7F 7F 7F 28]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV29 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 1C 23 7F 7F 3B]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV29 value 35 (0x23, 100011).
[EF 0E 7C 2B 00 00 00 00 02 42 7F 7F 7F 76]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV67 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 42 02 7F 7F 44]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV67 value 2 (0x2, 10).
[EF 0E 7C 2B 00 00 00 00 02 43 7F 7F 7F 77]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV68 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 43 05 7F 7F 42]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV68 value 5 (0x5, 101).
[EF 0E 7C 2B 00 00 00 00 02 44 7F 7F 7F 70]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV69 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 44 07 7F 7F 47]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV69 value 7 (0x7, 111).
[EF 0E 7C 2B 00 00 00 00 02 45 7F 7F 7F 71]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV70 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 45 0C 7F 7F 4D]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV70 value 12 (0xc, 1100).
[EF 0E 7C 2B 00 00 00 00 02 46 7F 7F 7F 72]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV71 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 46 10 7F 7F 52]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV71 value 16 (0x10, 10000).
[EF 0E 7C 2B 00 00 00 00 02 47 7F 7F 7F 73]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV72 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 47 15 7F 7F 56]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV72 value 21 (0x15, 10101).
[EF 0E 7C 2B 00 00 00 00 02 48 7F 7F 7F 7C]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV73 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 48 1A 7F 7F 56]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV73 value 26 (0x1a, 11010).
[EF 0E 7C 2B 00 00 00 00 02 49 7F 7F 7F 7D]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV74 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 49 1E 7F 7F 53]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV74 value 30 (0x1e, 11110).
[EF 0E 7C 2B 00 00 00 00 02 4A 7F 7F 7F 7E]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV75 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 4A 23 7F 7F 6D]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV75 value 35 (0x23, 100011).
[EF 0E 7C 2B 00 00 00 00 02 4B 7F 7F 7F 7F]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV76 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 4B 28 7F 7F 67]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV76 value 40 (0x28, 101000).
[EF 0E 7C 2B 00 00 00 00 02 4C 7F 7F 7F 78]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV77 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 4C 2F 7F 7F 67]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV77 value 47 (0x2f, 101111).
[EF 0E 7C 2B 00 00 00 00 02 4D 7F 7F 7F 79]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV78 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 4D 33 7F 7F 7A]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV78 value 51 (0x33, 110011).
[EF 0E 7C 2B 00 00 00 00 02 4E 7F 7F 7F 7A]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV79 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 4E 3A 7F 7F 70]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV79 value 58 (0x3a, 111010).
[... cut ...]
[E7 0E 7C 2B 00 00 02 47 02 5C 5B 7F 7F 01]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV93 value 219 (0xdb, 11011011).
[EF 0E 7C 2B 00 00 00 00 02 5D 7F 7F 7F 69]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV94 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 02 5D 7F 7F 7F 24]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV94 value 255 (0xff, 11111111).
[BB 7F 00 3B]  Request Command Station Ops Switches.

ValPistilli, 2017-03-06:
The DCS240 that I have was updated thursday. The owner said.

ValPistilli, 2017-03-06:
I'll have this unit for about two weeks if you need anything else checked. Let me know.
Val

ValPistilli, 2017-03-06:
I am now able to connect to the program track via the front panel USB port. Previous readings were with LocobufferUSB.
Here are readings for Speed Table read:
[EF 0E 7C 2B 00 00 00 00 02 1C 7F 7F 7F 28]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV29 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 1C 23 7F 7F 3B]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV29 value 35 (0x23, 100011).
[EF 0E 7C 2B 00 00 00 00 02 42 7F 7F 7F 76]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV67 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 42 02 7F 7F 44]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV67 value 2 (0x2, 10).
[EF 0E 7C 2B 00 00 00 00 02 43 7F 7F 7F 77]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV68 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 43 0B 7F 7F 4C]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV68 value 11 (0xb, 1011).
[EF 0E 7C 2B 00 00 00 00 02 44 7F 7F 7F 70]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV69 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 44 14 7F 7F 54]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV69 value 20 (0x14, 10100).
[EF 0E 7C 2B 00 00 00 00 02 45 7F 7F 7F 71]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV70 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 45 1E 7F 7F 5F]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV70 value 30 (0x1e, 11110).
[EF 0E 7C 2B 00 00 00 00 02 46 7F 7F 7F 72]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV71 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 46 27 7F 7F 65]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV71 value 39 (0x27, 100111).
[EF 0E 7C 2B 00 00 00 00 02 47 7F 7F 7F 73]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV72 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 47 30 7F 7F 73]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV72 value 48 (0x30, 110000).
[EF 0E 7C 2B 00 00 00 00 02 48 7F 7F 7F 7C]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV73 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[BB 7F 00 3B]  Request Command Station Ops Switches.
[E7 0E 7C 2B 00 00 02 47 00 48 3A 7F 7F 76]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV73 value 58 (0x3a, 111010).
[EF 0E 7C 2B 00 00 00 00 02 49 7F 7F 7F 7D]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV74 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 49 43 7F 7F 0E]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV74 value 67 (0x43, 1000011).
[EF 0E 7C 2B 00 00 00 00 02 4A 7F 7F 7F 7E]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV75 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 4A 4C 7F 7F 02]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV75 value 76 (0x4c, 1001100).
[EF 0E 7C 2B 00 00 00 00 02 4B 7F 7F 7F 7F]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV76 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 4B 56 7F 7F 19]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV76 value 86 (0x56, 1010110).
[EF 0E 7C 2B 00 00 00 00 02 4C 7F 7F 7F 78]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV77 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 4C 5F 7F 7F 17]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV77 value 95 (0x5f, 1011111).
[EF 0E 7C 2B 00 00 00 00 02 4D 7F 7F 7F 79]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV78 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 4D 69 7F 7F 20]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV78 value 105 (0x69, 1101001).
[EF 0E 7C 2B 00 00 00 00 02 4E 7F 7F 7F 7A]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV79 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[... cut ...]
[E7 0E 7C 2B 00 00 02 47 02 5C 75 7F 7F 2F]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV93 value 245 (0xf5, 11110101).
[EF 0E 7C 2B 00 00 00 00 02 5D 7F 7F 7F 69]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV94 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 02 5D 7F 7F 7F 24]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV94 value 255 (0xff, 11111111).

ValPistilli, 2017-03-06:
Here is Speed Table write:
[EF 0E 7C 6B 00 00 00 00 00 1C 23 7F 7F 36]  Programming Request: Write Byte in Direct Mode on Service Track to CV29 value 35 (0x23, 100011).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 1C 23 7F 7F 7B]  Programming Response: Write Byte in Direct Mode on Service Track to CV29 value 35 (0x23, 100011).
[EF 0E 7C 6B 00 00 00 00 00 42 02 7F 7F 49]  Programming Request: Write Byte in Direct Mode on Service Track to CV67 value 2 (0x2, 10).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[BB 7F 00 3B]  Request Command Station Ops Switches.
[E7 0E 7C 6B 00 00 02 47 00 42 02 7F 7F 04]  Programming Response: Write Byte in Direct Mode on Service Track to CV67 value 2 (0x2, 10).
[EF 0E 7C 6B 00 00 00 00 00 43 0B 7F 7F 41]  Programming Request: Write Byte in Direct Mode on Service Track to CV68 value 11 (0xb, 1011).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 43 0B 7F 7F 0C]  Programming Response: Write Byte in Direct Mode on Service Track to CV68 value 11 (0xb, 1011).
[EF 0E 7C 6B 00 00 00 00 00 44 14 7F 7F 59]  Programming Request: Write Byte in Direct Mode on Service Track to CV69 value 20 (0x14, 10100).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 44 14 7F 7F 14]  Programming Response: Write Byte in Direct Mode on Service Track to CV69 value 20 (0x14, 10100).
[EF 0E 7C 6B 00 00 00 00 00 45 1E 7F 7F 52]  Programming Request: Write Byte in Direct Mode on Service Track to CV70 value 30 (0x1e, 11110).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 45 1E 7F 7F 1F]  Programming Response: Write Byte in Direct Mode on Service Track to CV70 value 30 (0x1e, 11110).
[EF 0E 7C 6B 00 00 00 00 00 46 27 7F 7F 68]  Programming Request: Write Byte in Direct Mode on Service Track to CV71 value 39 (0x27, 100111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 46 27 7F 7F 25]  Programming Response: Write Byte in Direct Mode on Service Track to CV71 value 39 (0x27, 100111).
[EF 0E 7C 6B 00 00 00 00 00 47 30 7F 7F 7E]  Programming Request: Write Byte in Direct Mode on Service Track to CV72 value 48 (0x30, 110000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 47 30 7F 7F 33]  Programming Response: Write Byte in Direct Mode on Service Track to CV72 value 48 (0x30, 110000).
[EF 0E 7C 6B 00 00 00 00 00 48 3A 7F 7F 7B]  Programming Request: Write Byte in Direct Mode on Service Track to CV73 value 58 (0x3a, 111010).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 48 3A 7F 7F 36]  Programming Response: Write Byte in Direct Mode on Service Track to CV73 value 58 (0x3a, 111010).
[EF 0E 7C 6B 00 00 00 00 00 49 43 7F 7F 03]  Programming Request: Write Byte in Direct Mode on Service Track to CV74 value 67 (0x43, 1000011).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 49 43 7F 7F 4E]  Programming Response: Write Byte in Direct Mode on Service Track to CV74 value 67 (0x43, 1000011).
[EF 0E 7C 6B 00 00 00 00 00 4A 4C 7F 7F 0F]  Programming Request: Write Byte in Direct Mode on Service Track to CV75 value 76 (0x4c, 1001100).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4A 4C 7F 7F 42]  Programming Response: Write Byte in Direct Mode on Service Track to CV75 value 76 (0x4c, 1001100).
[EF 0E 7C 6B 00 00 00 00 00 4B 56 7F 7F 14]  Programming Request: Write Byte in Direct Mode on Service Track to CV76 value 86 (0x56, 1010110).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4B 56 7F 7F 59]  Programming Response: Write Byte in Direct Mode on Service Track to CV76 value 86 (0x56, 1010110).
[EF 0E 7C 6B 00 00 00 00 00 4C 5F 7F 7F 1A]  Programming Request: Write Byte in Direct Mode on Service Track to CV77 value 95 (0x5f, 1011111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4C 5F 7F 7F 57]  Programming Response: Write Byte in Direct Mode on Service Track to CV77 value 95 (0x5f, 1011111).
[EF 0E 7C 6B 00 00 00 00 00 4D 69 7F 7F 2D]  Programming Request: Write Byte in Direct Mode on Service Track to CV78 value 105 (0x69, 1101001).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4D 69 7F 7F 60]  Programming Response: Write Byte in Direct Mode on Service Track to CV78 value 105 (0x69, 1101001).
[EF 0E 7C 6B 00 00 00 00 00 4E 72 7F 7F 35]  Programming Request: Write Byte in Direct Mode on Service Track to CV79 value 114 (0x72, 1110010).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4E 72 7F 7F 78]  Programming Response: Write Byte in Direct Mode on Service Track to CV79 value 114 (0x72, 1110010).
[EF 0E 7C 6B 00 00 00 00 00 4F 7B 7F 7F 3D]  Programming Request: Write Byte in Direct Mode on Service Track to CV80 value 123 (0x7b, 1111011).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4F 7B 7F 7F 70]  Programming Response: Write Byte in Direct Mode on Service Track to CV80 value 123 (0x7b, 1111011).
[EF 0E 7C 6B 00 00 00 00 02 50 05 7F 7F 5E]  Programming Request: Write Byte in Direct Mode on Service Track to CV81 value 133 (0x85, 10000101).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[... cut ...]
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 02 5C 75 7F 7F 6F]  Programming Response: Write Byte in Direct Mode on Service Track to CV93 value 245 (0xf5, 11110101).
[EF 0E 7C 6B 00 00 00 00 02 5D 7F 7F 7F 29]  Programming Request: Write Byte in Direct Mode on Service Track to CV94 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 02 5D 7F 7F 7F 64]  Programming Response: Write Byte in Direct Mode on Service Track to CV94 value 255 (0xff, 11111111).

ValPistilli, 2017-03-06:
write 16 to CV 31

[BB 7F 00 3B]  Request Command Station Ops Switches.
[EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Programming Request: Write Byte in Direct Mode on Service Track to CV31 value 16 (0x10, 10000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 1E 10 7F 7F 4A]  Programming Response: Write Byte in Direct Mode on Service Track to CV31 value 16 (0x10, 10000).

ValPistilli, 2017-03-06:
Here is a read of TCS Wow 121. Speed table read.

[EF 0E 7C 2B 00 00 00 00 02 1C 7F 7F 7F 28]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV29 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 03 47 00 1C 23 7F 7F 3A]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV29 value 35 (0x23, 100011).
[EF 0E 7C 2B 00 00 00 00 02 42 7F 7F 7F 76]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV67 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 03 47 00 42 08 7F 7F 4F]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV67 value 8 (0x8, 1000).
[EF 0E 7C 2B 00 00 00 00 02 43 7F 7F 7F 77]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV68 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 43 10 7F 7F 57]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV68 value 16 (0x10, 10000).
[EF 0E 7C 2B 00 00 00 00 02 44 7F 7F 7F 70]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV69 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 44 18 7F 7F 58]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV69 value 24 (0x18, 11000).
[EF 0E 7C 2B 00 00 00 00 02 45 7F 7F 7F 71]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV70 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 45 20 7F 7F 61]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV70 value 32 (0x20, 100000).
[EF 0E 7C 2B 00 00 00 00 02 46 7F 7F 7F 72]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV71 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 46 28 7F 7F 6A]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV71 value 40 (0x28, 101000).
[EF 0E 7C 2B 00 00 00 00 02 47 7F 7F 7F 73]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV72 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 47 30 7F 7F 73]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV72 value 48 (0x30, 110000).
[EF 0E 7C 2B 00 00 00 00 02 48 7F 7F 7F 7C]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV73 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 48 38 7F 7F 74]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV73 value 56 (0x38, 111000).
[EF 0E 7C 2B 00 00 00 00 02 49 7F 7F 7F 7D]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV74 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 49 40 7F 7F 0D]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV74 value 64 (0x40, 1000000).
[EF 0E 7C 2B 00 00 00 00 02 4A 7F 7F 7F 7E]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV75 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 4A 48 7F 7F 06]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV75 value 72 (0x48, 1001000).
[EF 0E 7C 2B 00 00 00 00 02 4B 7F 7F 7F 7F]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV76 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 4B 50 7F 7F 1F]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV76 value 80 (0x50, 1010000).
[EF 0E 7C 2B 00 00 00 00 02 4C 7F 7F 7F 78]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV77 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 4C 58 7F 7F 10]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV77 value 88 (0x58, 1011000).
[EF 0E 7C 2B 00 00 00 00 02 4D 7F 7F 7F 79]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV78 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[BB 7B 00 3F]  Request Fast Clock information.
[E7 0E 7B 00 7F 7F 7F 4F 7F 00 70 58 06 0C]  Response Fast Clock is Frozen, rate is 0:1. Day 0, 23:0.0. Last set by ID 0x6 0x58 (856).
	Master: LocoNet 1.1; Track Status: On/Running; Programming Track: Busy
[E7 0E 7C 2B 00 00 02 47 00 4D 60 7F 7F 29]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV78 value 96 (0x60, 1100000).
[EF 0E 7C 2B 00 00 00 00 02 4E 7F 7F 7F 7A]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV79 value 255 (0xff, 11111111).
[... cut ...]
[E7 0E 7B 00 7F 7F 7F 47 7F 00 70 58 06 04]  Response Fast Clock is Frozen, rate is 0:1. Day 0, 23:0.0. Last set by ID 0x6 0x58 (856).
	Master: LocoNet 1.1; Track Status: On/Running; Programming Track: Available
[BB 7F 00 3B]  Request Command Station Ops Switches.
[BB 7B 00 3F]  Request Fast Clock information.
[E7 0E 7B 00 7F 7F 7F 47 7F 00 70 58 06 04]  Response Fast Clock is Frozen, rate is 0:1. Day 0, 23:0.0. Last set by ID 0x6 0x58 (856).
	Master: LocoNet 1.1; Track Status: On/Running; Programming Track: Available
[BB 7F 00 3B]  Request Command Station Ops Switches.

ValPistilli, 2017-03-06:
speed table Write TCS Wow 121

[EF 0E 7C 6B 00 00 00 00 00 1C 23 7F 7F 36]  Programming Request: Write Byte in Direct Mode on Service Track to CV29 value 35 (0x23, 100011).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 1C 23 7F 7F 7B]  Programming Response: Write Byte in Direct Mode on Service Track to CV29 value 35 (0x23, 100011).
[EF 0E 7C 6B 00 00 00 00 00 42 08 7F 7F 43]  Programming Request: Write Byte in Direct Mode on Service Track to CV67 value 8 (0x8, 1000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 42 08 7F 7F 0E]  Programming Response: Write Byte in Direct Mode on Service Track to CV67 value 8 (0x8, 1000).
[EF 0E 7C 6B 00 00 00 00 00 43 10 7F 7F 5A]  Programming Request: Write Byte in Direct Mode on Service Track to CV68 value 16 (0x10, 10000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 43 10 7F 7F 17]  Programming Response: Write Byte in Direct Mode on Service Track to CV68 value 16 (0x10, 10000).
[EF 0E 7C 6B 00 00 00 00 00 44 18 7F 7F 55]  Programming Request: Write Byte in Direct Mode on Service Track to CV69 value 24 (0x18, 11000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 44 18 7F 7F 18]  Programming Response: Write Byte in Direct Mode on Service Track to CV69 value 24 (0x18, 11000).
[EF 0E 7C 6B 00 00 00 00 00 45 20 7F 7F 6C]  Programming Request: Write Byte in Direct Mode on Service Track to CV70 value 32 (0x20, 100000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 45 20 7F 7F 21]  Programming Response: Write Byte in Direct Mode on Service Track to CV70 value 32 (0x20, 100000).
[EF 0E 7C 6B 00 00 00 00 00 46 28 7F 7F 67]  Programming Request: Write Byte in Direct Mode on Service Track to CV71 value 40 (0x28, 101000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 46 28 7F 7F 2A]  Programming Response: Write Byte in Direct Mode on Service Track to CV71 value 40 (0x28, 101000).
[EF 0E 7C 6B 00 00 00 00 00 47 30 7F 7F 7E]  Programming Request: Write Byte in Direct Mode on Service Track to CV72 value 48 (0x30, 110000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 03 47 00 47 30 7F 7F 32]  Programming Response: Write Byte in Direct Mode on Service Track to CV72 value 48 (0x30, 110000).
[EF 0E 7C 6B 00 00 00 00 00 48 38 7F 7F 79]  Programming Request: Write Byte in Direct Mode on Service Track to CV73 value 56 (0x38, 111000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 03 47 00 48 38 7F 7F 35]  Programming Response: Write Byte in Direct Mode on Service Track to CV73 value 56 (0x38, 111000).
[EF 0E 7C 6B 00 00 00 00 00 49 40 7F 7F 00]  Programming Request: Write Byte in Direct Mode on Service Track to CV74 value 64 (0x40, 1000000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 49 40 7F 7F 4D]  Programming Response: Write Byte in Direct Mode on Service Track to CV74 value 64 (0x40, 1000000).
[EF 0E 7C 6B 00 00 00 00 00 4A 48 7F 7F 0B]  Programming Request: Write Byte in Direct Mode on Service Track to CV75 value 72 (0x48, 1001000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4A 48 7F 7F 46]  Programming Response: Write Byte in Direct Mode on Service Track to CV75 value 72 (0x48, 1001000).
[EF 0E 7C 6B 00 00 00 00 00 4B 50 7F 7F 12]  Programming Request: Write Byte in Direct Mode on Service Track to CV76 value 80 (0x50, 1010000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4B 50 7F 7F 5F]  Programming Response: Write Byte in Direct Mode on Service Track to CV76 value 80 (0x50, 1010000).
[EF 0E 7C 6B 00 00 00 00 00 4C 58 7F 7F 1D]  Programming Request: Write Byte in Direct Mode on Service Track to CV77 value 88 (0x58, 1011000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4C 58 7F 7F 50]  Programming Response: Write Byte in Direct Mode on Service Track to CV77 value 88 (0x58, 1011000).
[EF 0E 7C 6B 00 00 00 00 00 4D 60 7F 7F 24]  Programming Request: Write Byte in Direct Mode on Service Track to CV78 value 96 (0x60, 1100000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4D 60 7F 7F 69]  Programming Response: Write Byte in Direct Mode on Service Track to CV78 value 96 (0x60, 1100000).
[EF 0E 7C 6B 00 00 00 00 00 4E 68 7F 7F 2F]  Programming Request: Write Byte in Direct Mode on Service Track to CV79 value 104 (0x68, 1101000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4E 68 7F 7F 62]  Programming Response: Write Byte in Direct Mode on Service Track to CV79 value 104 (0x68, 1101000).
[EF 0E 7C 6B 00 00 00 00 00 4F 72 7F 7F 34]  Programming Request: Write Byte in Direct Mode on Service Track to CV80 value 114 (0x72, 1110010).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 4F 72 7F 7F 79]  Programming Response: Write Byte in Direct Mode on Service Track to CV80 value 114 (0x72, 1110010).
[EF 0E 7C 6B 00 00 00 00 00 50 7C 7F 7F 25]  Programming Request: Write Byte in Direct Mode on Service Track to CV81 value 124 (0x7c, 1111100).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[... cut ...]
[E7 0E 7C 6B 00 00 02 47 02 41 00 7F 7F 07]  Programming Response: Write Byte in Direct Mode on Service Track to CV66 value 128 (0x80, 10000000).
[EF 0E 7C 6B 00 00 00 00 02 5E 00 7F 7F 55]  Programming Request: Write Byte in Direct Mode on Service Track to CV95 value 128 (0x80, 10000000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 02 5E 00 7F 7F 18]  Programming Response: Write Byte in Direct Mode on Service Track to CV95 value 128 (0x80, 10000000).
[BB 7F 00 3B]  Request Command Station Ops Switches.

dheap, 2017-03-06:
Thanks very much for those. I'm sure that will be a great help to us. There was no sign of overlap in your Speed table writes, but the raw codes are very interesting.

The LocoNet experts will have to look into this as I know very little about LocoNet. The return codes are not quite what I expected to see from my ancient LocoNet Personal Use Edition 1.0 Specification.

I would have expected to see `[B4 7F 01 35] LONG_ACK` rather than `[B4 6F 01 25] LONG_ACK` to indicate we need to wait for an `<E7>` reply before issuing another Programming Request.

The logs from the original poster showed overlaps:
Loconet Monitor: 
Loksound 4 21-pin decoder, loco address 1000 (not that it matters?)
DecoderPro read CV16.2.257 via DP in Direct Byte mode
```
Programming Request: Write Byte in Direct Mode on Service Track to CV31 
value 16 (0x10, 10000).
LONG_ACK: The Slot Write command was accepted.
Programming Request: Write Byte in Direct Mode on Service Track to CV32 
value 2 (0x2, 10).
LONG_ACK: The Slot Write command was rejected.
Programming Track Request: Read Byte in Direct Mode on Service Track 
variable CV257 value 255 (0xff, 11111111).
LONG_ACK: The Slot Write command was rejected.
Programming Response: Write Byte in Direct Mode on Service Track to CV31 
value 16 (0x10, 10000).
```

Let's see what the LocoNet experts come up with...

ValPistilli, 2017-03-06:
Here is write a four CV sequence to change a bell type.

[EF 0E 7C 6B 00 00 00 00 01 48 04 7F 7F 44]  Programming Request: Write Byte in Direct Mode on Service Track to CV201 value 4 (0x4, 100).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 03 47 01 48 04 7F 7F 08]  Programming Response: Write Byte in Direct Mode on Service Track to CV201 value 4 (0x4, 100).
[EF 0E 7C 6B 00 00 00 00 01 49 17 7F 7F 56]  Programming Request: Write Byte in Direct Mode on Service Track to CV202 value 23 (0x17, 10111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 03 47 01 49 17 7F 7F 1A]  Programming Response: Write Byte in Direct Mode on Service Track to CV202 value 23 (0x17, 10111).
[EF 0E 7C 6B 00 00 00 00 01 4A 00 7F 7F 42]  Programming Request: Write Byte in Direct Mode on Service Track to CV203 value 0 (0x0, 0).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 01 4A 00 7F 7F 0F]  Programming Response: Write Byte in Direct Mode on Service Track to CV203 value 0 (0x0, 0).
[EF 0E 7C 6B 00 00 00 00 01 4B 00 7F 7F 43]  Programming Request: Write Byte in Direct Mode on Service Track to CV204 value 0 (0x0, 0).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 01 4B 00 7F 7F 0E]  Programming Response: Write Byte in Direct Mode on Service Track to CV204 value 0 (0x0, 0).

ValPistilli, 2017-03-06:
Read full sheet Brake pane;

[BB 7B 00 3F]  Request Fast Clock information.
[E7 0E 7B 00 7F 7F 7F 47 7F 00 70 58 06 04]  Response Fast Clock is Frozen, rate is 0:1. Day 0, 23:0.0. Last set by ID 0x6 0x58 (856).
	Master: LocoNet 1.1; Track Status: On/Running; Programming Track: Available
[BB 7F 00 3B]  Request Command Station Ops Switches.
[BB 7B 00 3F]  Request Fast Clock information.
[E7 0E 7B 00 7F 7F 7F 47 7F 00 70 58 06 04]  Response Fast Clock is Frozen, rate is 0:1. Day 0, 23:0.0. Last set by ID 0x6 0x58 (856).
	Master: LocoNet 1.1; Track Status: On/Running; Programming Track: Available
[EF 0E 7C 2B 00 00 00 00 03 36 7F 7F 7F 03]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV183 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 01 36 20 7F 7F 13]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV183 value 32 (0x20, 100000).
[EF 0E 7C 2B 00 00 00 00 03 37 7F 7F 7F 02]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV184 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 03 47 01 37 1A 7F 7F 29]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV184 value 26 (0x1a, 11010).
[EF 0E 7C 2B 00 00 00 00 03 38 7F 7F 7F 0D]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV185 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 01 38 10 7F 7F 2D]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV185 value 16 (0x10, 10000).
[EF 0E 7C 2B 00 00 00 00 03 39 7F 7F 7F 0C]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV186 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 01 39 08 7F 7F 34]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV186 value 8 (0x8, 1000).
[EF 0E 7C 2B 00 00 00 00 03 3A 7F 7F 7F 0F]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV187 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 01 3A 03 7F 7F 3C]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV187 value 3 (0x3, 11).
[EF 0E 7C 6B 00 00 00 00 01 48 68 7F 7F 28]  Programming Request: Write Byte in Direct Mode on Service Track to CV201 value 104 (0x68, 1101000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 01 48 68 7F 7F 65]  Programming Response: Write Byte in Direct Mode on Service Track to CV201 value 104 (0x68, 1101000).
[EF 0E 7C 6B 00 00 00 00 01 49 0D 7F 7F 4C]  Programming Request: Write Byte in Direct Mode on Service Track to CV202 value 13 (0xd, 1101).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 01 49 0D 7F 7F 01]  Programming Response: Write Byte in Direct Mode on Service Track to CV202 value 13 (0xd, 1101).
[EF 0E 7C 6B 00 00 00 00 01 4B 64 7F 7F 27]  Programming Request: Write Byte in Direct Mode on Service Track to CV204 value 100 (0x64, 1100100).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 02 00 02 47 01 4B 64 7F 7F 68]  Programming Response: Write Byte in Direct Mode on Service Track to CV204 value 100 (0x64, 1100100).
[EF 0E 7C 2B 00 00 00 00 03 4A 7F 7F 7F 7F]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV203 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 01 4A 00 7F 7F 4F]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV203 value 0 (0x0, 0).
[EF 0E 7C 2B 00 00 00 00 03 4B 7F 7F 7F 7E]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV204 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 01 4B 0F 7F 7F 41]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV204 value 15 (0xf, 1111).
[EF 0E 7C 6B 00 00 00 00 01 48 68 7F 7F 28]  Programming Request: Write Byte in Direct Mode on Service Track to CV201 value 104 (0x68, 1101000).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 01 48 68 7F 7F 65]  Programming Response: Write Byte in Direct Mode on Service Track to CV201 value 104 (0x68, 1101000).
[EF 0E 7C 6B 00 00 00 00 01 49 0F 7F 7F 4E]  Programming Request: Write Byte in Direct Mode on Service Track to CV202 value 15 (0xf, 1111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 01 49 0F 7F 7F 03]  Programming Response: Write Byte in Direct Mode on Service Track to CV202 value 15 (0xf, 1111).
[EF 0E 7C 6B 00 00 00 00 01 4B 64 7F 7F 27]  Programming Request: Write Byte in Direct Mode on Service Track to CV204 value 100 (0x64, 1100100).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 02 00 02 47 01 4B 64 7F 7F 68]  Programming Response: Write Byte in Direct Mode on Service Track to CV204 value 100 (0x64, 1100100).
[... cut ...]
[E7 0E 7C 2B 00 00 02 47 01 4A 02 7F 7F 4D]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV203 value 2 (0x2, 10).
[EF 0E 7C 2B 00 00 00 00 03 4B 7F 7F 7F 7E]  Programming Track Request: Read Byte in Direct Mode on Service Track variable  CV204 value 255 (0xff, 11111111).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 03 4B 3B 7F 7F 77]  Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set  CV204 value 187 (0xbb, 10111011).

dheap, 2017-03-06:
OK, I had another look at the documentation and can now see why `B4 6F `rather than` B4 7F` . But the `01` certainly says we need to wait for an `<E7>` reply before issuing another Programming Request, and we were not doing that in the Indexed CV Read sequence from Mick Moignard.  

> I would have expected to see `[B4 7F 01 35] LONG_ACK` rather than `[B4 6F 01 25] LONG_ACK` to indicate we need to wait for an `<E7>` reply before issuing another Programming Request.

I think we've probably got enough to go with now, unless you have any logs showing "rejected" or other errors.

mickmoignardOAC, 2017-03-06:
Here's my results, which show many Rejected errors, and has a screenshot at the end to show how DP reports the issues.
This DCS240 is a replacement for the original, by Digitrax, both of which showed the error I originally reported.  The new one is now running Digitrax firmware 0.3. It was tested while connected to the layout - which means a DCS100 as a booster, a Db150 ditto, UR92 (powered) and 4 UP3/5 unpowered panels, plus the computer and Locobuffer on Loconet.  

[DCS240 Programming Loksound 4 issues 2017-03-06.pdf](https://github.com/JMRI/JMRI/files/821496/DCS240.Programming.Loksound.4.issues.2017-03-06.pdf)

mickmoignardOAC, 2017-03-06:
Is it possible that the issue is decoder specific, to do with how long it takes for the decoder to respond?  I've not had issues programming Tsunamis with the DCS240/DP combo; I only get the issue with this Loksound 4.  From what Val says, he may have used a Tsunami? I have a Zimo, I'll see whether that works OK or not, in the same way, read and write the speed table pane.

ValPistilli, 2017-03-06:
I did the last test using a TCS Wow sound 121 decoder.

Val Pistilli
Webmaster
www.rpmvalleyforge.com
www.lehighvalleysgaugers.org
IN GOD WE TRUST

mickmoignardOAC, 2017-03-06:
I just tried the read/write speed table with a Zimo MX648R: no issues, but this decoder has no indexed Cvs.  I then tried read/write the Function map pane on a TSU-1100, which uses single-indirection indexed Cvs, again, no issues.  The only decoder seen it with is Loksound 4s with double-indexed CVs.

bobjacobsen, 2017-03-06:
Thanks.  That’s perfect. It’s showing a new form of LACK that JMRI doesn’t understand is different from the old forms.  Will get on fixing this.

Bob

> write 16 to CV 31
>
> [BB 7F 00 3B] Request Command Station Ops Switches.
> [EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07] Programming Request: Write Byte in Direct Mode on Service Track to CV31 value 16 [...]

bobjacobsen, 2017-03-06:
That read-speed-table sequence looks a little odd.  Can you please try something for me?  No trace needed.
- Using the DecoderPro speed table pane, write the first (left-most) value in the TCS Wow 121 speed table to some known value, say 10
- Then, using the single CV programmer, write CV67 (that same one) to some other value, say 15
- Then read it back using the DecoderPro speed table pane (e.g. read all on that pane).  Does it say 10, 15 or something else?
- Then read CV29 using the single CV programmer.  Does the read value change?
- Then read CV69 using the single CV programmer.  What does it say?

Thanks.

ValPistilli, 2017-03-06:
> 
> That read-speed-table sequence looks a little odd. Can you please try something for me? No trace needed.
> 
> Using the DecoderPro speed table pane, write the first (left-most) value in the TCS Wow 121 speed table to some known value, say 10
Using the touch pad, I could get it to 9. Wrote it. Read back 9
> Then, using the single CV programmer, write CV67 (that same one) to some other value, say 15
Wrote 15
> Then read it back using the DecoderPro speed table pane (e.g. read all on that pane). Does it say 10, 15 or something else?
Read 15
> Then read CV29 using the single CV programmer. Does the read value change?
Started with 35. Read 35
> Then read CV69 using the single CV programmer. What does it say?
69 read 24. Or did you mean 67?  Read 67=15
> Thanks.

Val Pistilli
Webmaster
www.rpmvalleyforge.com
www.lehighvalleysgaugers.org
IN GOD WE TRUST

ValPistilli, 2017-03-06:
These measurements were on a TCS Wow 121.

Val Pistilli
Webmaster
www.rpmvalleyforge.com
www.lehighvalleysgaugers.org
IN GOD WE TRUST

dheap, 2017-03-09:
It appears another user is having similar problems, this time with a DCS100. 
https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/136632

bobjacobsen, 2017-03-12:
There seem to be several things going wrong here.
- There was a small chance of a timing glitch in the programming process.  #3232 fixes this and improves debugging. I wasn't able to recreate this as the cause of the 1st trace in this Issue
- What I thought was a problem in LACK decoding was actually the point above, now fixed
- So we're left with JMRI starting a second read too early in the trace of the Speed Table read, without understanding the cause of that.

PR #3232 will be in JMRI 4.7.2, and _might_ fix the LokSound 4 read problem.  Tests appreciated once 4.7.2 is out.  But that isn't expected to solve the Speed Table read.  Don't know the cause of that yet.

mickmoignardOAC, 2017-03-16:
Sadly 4.7.2 does not fix the Loksound 4 read problem. Here's a Loconet trace of it: I stopped after a few CVs had failed...

[EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[EF 0E 7C 6B 00 00 00 00 00 1F 02 7F 7F 14]  Byte Write in Direct Mode on Service Track: CV32 value 2 (0x02, 00000010b).
[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
[EF 0E 7C 2B 00 00 00 00 12 00 7F 7F 7F 24]  Byte Read in Direct Mode on Service Track: CV257.
[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
[E7 0E 7C 6B 00 00 04 47 00 1E 10 7F 7F 4C]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV31 value 16 (0x10, 00010000b).
[EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[EF 0E 7C 6B 00 00 00 00 00 1F 02 7F 7F 14]  Byte Write in Direct Mode on Service Track: CV32 value 2 (0x02, 00000010b).
[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
[EF 0E 7C 2B 00 00 00 00 12 00 7F 7F 7F 24]  Byte Read in Direct Mode on Service Track: CV257.
[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
[E7 0E 7C 6B 00 00 04 47 00 1E 10 7F 7F 4C]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV31 value 16 (0x10, 00010000b).
[EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[EF 0E 7C 6B 00 00 00 00 00 1F 02 7F 7F 14]  Byte Write in Direct Mode on Service Track: CV32 value 2 (0x02, 00000010b).
[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
[EF 0E 7C 2B 00 00 00 00 12 01 7F 7F 7F 25]  Byte Read in Direct Mode on Service Track: CV258.
[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
[E7 0E 7C 6B 00 00 04 47 00 1E 10 7F 7F 4C]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV31 value 16 (0x10, 00010000b).
[EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[EF 0E 7C 6B 00 00 00 00 00 1F 02 7F 7F 14]  Byte Write in Direct Mode on Service Track: CV32 value 2 (0x02, 00000010b).
[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
[EF 0E 7C 2B 00 00 00 00 12 01 7F 7F 7F 25]  Byte Read in Direct Mode on Service Track: CV258.
[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
[E7 0E 7C 6B 00 00 04 47 00 1E 10 7F 7F 4C]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV31 value 16 (0x10, 00010000b).
[A0 24 00 7B]  Set speed of loco in slot 36 to 0.
[BB 7F 00 3B]  Request Command Station OpSwitches (or DCS210/DCS240 check for multiple command stations on LocoNet).

I've also done a Speed Table write on the same loco (coming....)

mickmoignardOAC, 2017-03-16:
Got a whole bunch of 306 and 302 errors, and some of the CVs stayed red on the pane when it was completed. 

[A0 24 00 7B]  Set speed of loco in slot 36 to 0.
[EF 0E 7C 6B 00 00 00 00 00 1C 22 7F 7F 37]  Byte Write in Direct Mode on Service Track: CV29 value 34 (0x22, 00100010b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 04 47 00 1C 22 7F 7F 7C]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV29 value 34 (0x22, 00100010b).
[EF 0E 7C 6B 00 00 00 00 00 1C 22 7F 7F 37]  Byte Write in Direct Mode on Service Track: CV29 value 34 (0x22, 00100010b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 03 47 00 1C 22 7F 7F 7B]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV29 value 34 (0x22, 00100010b).
[EF 0E 7C 6B 00 00 00 00 00 42 01 7F 7F 4A]  Byte Write in Direct Mode on Service Track: CV67 value 1 (0x01, 00000001b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 42 01 7F 7F 07]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV67 value 1 (0x01, 00000001b).
[EF 0E 7C 6B 00 00 00 00 00 43 0A 7F 7F 40]  Byte Write in Direct Mode on Service Track: CV68 value 10 (0x0A, 00001010b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 43 0A 7F 7F 0D]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV68 value 10 (0x0A, 00001010b).
[EF 0E 7C 6B 00 00 00 00 00 44 13 7F 7F 5E]  Byte Write in Direct Mode on Service Track: CV69 value 19 (0x13, 00010011b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[EF 0E 7C 6B 00 00 00 00 00 44 13 7F 7F 5E]  Byte Write in Direct Mode on Service Track: CV69 value 19 (0x13, 00010011b).
[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
[EF 0E 7C 6B 00 00 00 00 00 44 13 7F 7F 5E]  Byte Write in Direct Mode on Service Track: CV69 value 19 (0x13, 00010011b).
[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
[E7 0E 7C 6B 00 00 08 47 00 44 13 7F 7F 19]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV69 value 19 (0x13, 00010011b).
[EF 0E 7C 6B 00 00 00 00 00 45 1D 7F 7F 51]  Byte Write in Direct Mode on Service Track: CV70 value 29 (0x1D, 00011101b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 04 47 00 45 1D 7F 7F 1A]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV70 value 29 (0x1D, 00011101b).
[EF 0E 7C 6B 00 00 00 00 00 45 1D 7F 7F 51]  Byte Write in Direct Mode on Service Track: CV70 value 29 (0x1D, 00011101b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 45 1D 7F 7F 1C]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV70 value 29 (0x1D, 00011101b).
[EF 0E 7C 6B 00 00 00 00 00 46 28 7F 7F 67]  Byte Write in Direct Mode on Service Track: CV71 value 40 (0x28, 00101000b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 02 47 00 46 28 7F 7F 2A]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV71 value 40 (0x28, 00101000b).
[EF 0E 7C 6B 00 00 00 00 00 47 2F 7F 7F 61]  Byte Write in Direct Mode on Service Track: CV72 value 47 (0x2F, 00101111b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[EF 0E 7C 6B 00 00 00 00 00 47 2F 7F 7F 61]  Byte Write in Direct Mode on Service Track: CV72 value 47 (0x2F, 00101111b).
[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
[E7 0E 7C 6B 00 00 06 47 00 47 2F 7F 7F 28]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV72 value 47 (0x2F, 00101111b).
[EF 0E 7C 6B 00 00 00 00 00 47 2F 7F 7F 61]  Byte Write in Direct Mode on Service Track: CV72 value 47 (0x2F, 00101111b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 04 47 00 47 2F 7F 7F 2A]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV72 value 47 (0x2F, 00101111b).
[EF 0E 7C 6B 00 00 00 00 00 48 38 7F 7F 79]  Byte Write in Direct Mode on Service Track: CV73 value 56 (0x38, 00111000b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 03 47 00 48 38 7F 7F 35]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV73 value 56 (0x38, 00111000b).
[EF 0E 7C 6B 00 00 00 00 00 49 43 7F 7F 03]  Byte Write in Direct Mode on Service Track: CV74 value 67 (0x43, 01000011b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[A0 24 00 7B]  Set speed of loco in slot 36 to 0.
[E7 0E 7C 6B 00 00 04 47 00 49 43 7F 7F 48]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV74 value 67 (0x43, 01000011b).
[EF 0E 7C 6B 00 00 00 00 00 49 43 7F 7F 03]  Byte Write in Direct Mode on Service Track: CV74 value 67 (0x43, 01000011b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 6B 00 00 03 47 00 49 43 7F 7F 4F]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV74 value 67 (0x43, 01000011b).
[EF 0E 7C 6B 00 00 00 00 00 4A 4C 7F 7F 0F]  Byte Write in Direct Mode on Service Track: CV75 value 76 (0x4C, 01001100b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[... cut ...]
[EF 0E 7C 6B 00 00 00 00 00 5E 00 7F 7F 57]  Byte Write in Direct Mode on Service Track: CV95 value 0 (0x00, 00000000b).
[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[EF 0E 7C 6B 00 00 00 00 00 5E 00 7F 7F 57]  Byte Write in Direct Mode on Service Track: CV95 value 0 (0x00, 00000000b).
[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
[E7 0E 7C 6B 00 00 08 47 00 5E 00 7F 7F 10]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV95 value 0 (0x00, 00000000b).

bobjacobsen, 2017-03-16:
Oh, rats.  And I had such hopes.  Thanks for the traces.  Will be back shortly with some next steps.

bobjacobsen, 2017-03-16:
On that first trace, 
```
[EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07] Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
[B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
[EF 0E 7C 6B 00 00 00 00 00 1F 02 7F 7F 14] Byte Write in Direct Mode on Service Track: CV32 value 2 (0x02, 00000010b).
[B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.
[EF 0E 7C 2B 00 00 00 00 12 00 7F 7F 7F 24] Byte Read in Direct Mode on Service Track: CV257.
[B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.
```

JMRI seems to be trying to read CV 257.16.2

But I can't find that CV in the ESU_LokSoundV4_0.xml definition, or any of the ESU definitions.  Any idea what that is?

mickmoignardOAC, 2017-03-16:
Bob

It's part of the Function map pane, and is one of the CVs that deals with the first value on the pane. 

Mick

________________________________
Mick Moignard
m: +44 7774 652504
Skype: mickmoignard

Sent from my iPad, so please excuse the typos.

> On that first trace,
>
> [EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07] Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
> [B4 6F 01 25] LONG_ACK: The Slot Write command was [...]

This email and any files transmitted with it are intended only for the person or entity to which it is addressed and may contain privileged or confidential information. If you're not the intended recipient, disclosing, copying, distributing or using this information is prohibited. If you've received this email in error, please contact the sender and delete all copies. This email may contain personal views which are not the views of OAC. We monitor our email system and may record your emails. OAC PLC is registered in England | Company no.: 4663795 | Registered office: 141-142 Fenchurch Street, London, EC3M 6BL, United Kingdom

bobjacobsen, 2017-03-16:
@mickmoignardOAC  Thanks! Both for finding that CV, and for all your help.

If/when you get the chance, could you rerun the speed table test with the timestamps (and raw data) turned on?  I'm working on a hypothesis about read speed.  I just need until the 1st reject and the next two messages after that.

Thanks again.

mickmoignardOAC, 2017-03-16:
Here you go.....

17:45:01.461: [EF 0E 7C 6B 00 00 00 00 00 1C 22 7F 7F 37]  Byte Write in Direct Mode on Service Track: CV29 value 34 (0x22, 00100010b).
17:45:01.461: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
17:45:03.336: [E7 0E 7C 6B 00 00 03 47 00 1C 22 7F 7F 7B]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV29 value 34 (0x22, 00100010b).
17:45:03.414: [EF 0E 7C 6B 00 00 00 00 00 42 01 7F 7F 4A]  Byte Write in Direct Mode on Service Track: CV67 value 1 (0x01, 00000001b).
17:45:03.414: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
17:45:04.977: [BB 7F 00 3B]  Request Command Station OpSwitches (or DCS210/DCS240 check for multiple command stations on LocoNet).
17:45:06.492: [EF 0E 7C 6B 00 00 00 00 00 42 01 7F 7F 4A]  Byte Write in Direct Mode on Service Track: CV67 value 1 (0x01, 00000001b).
17:45:06.508: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
17:45:07.571: [EF 0E 7C 6B 00 00 00 00 00 42 01 7F 7F 4A]  Byte Write in Direct Mode on Service Track: CV67 value 1 (0x01, 00000001b).
17:45:07.571: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
17:45:07.696: [E7 0E 7C 6B 00 00 08 47 00 42 01 7F 7F 0D]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV67 value 1 (0x01, 00000001b).
17:45:08.633: [EF 0E 7C 6B 00 00 00 00 00 43 0A 7F 7F 40]  Byte Write in Direct Mode on Service Track: CV68 value 10 (0x0A, 00001010b).
17:45:08.633: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
17:45:11.008: [E7 0E 7C 6B 00 00 04 47 00 43 0A 7F 7F 0B]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV68 value 10 (0x0A, 00001010b).
17:45:11.711: [EF 0E 7C 6B 00 00 00 00 00 43 0A 7F 7F 40]  Byte Write in Direct Mode on Service Track: CV68 value 10 (0x0A, 00001010b).
17:45:11.727: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
17:45:13.602: [E7 0E 7C 6B 00 00 03 47 00 43 0A 7F 7F 0C]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV68 value 10 (0x0A, 00001010b).
17:45:13.664: [EF 0E 7C 6B 00 00 00 00 00 44 13 7F 7F 5E]  Byte Write in Direct Mode on Service Track: CV69 value 19 (0x13, 00010011b).
17:45:13.680: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
17:45:16.758: [EF 0E 7C 6B 00 00 00 00 00 44 13 7F 7F 5E]  Byte Write in Direct Mode on Service Track: CV69 value 19 (0x13, 00010011b).
17:45:16.774: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
17:45:17.836: [EF 0E 7C 6B 00 00 00 00 00 44 13 7F 7F 5E]  Byte Write in Direct Mode on Service Track: CV69 value 19 (0x13, 00010011b).
17:45:17.852: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
17:45:18.914: [E7 0E 7C 6B 00 00 0A 47 00 44 13 7F 7F 1B]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV69 value 19 (0x13, 00010011b).
17:45:18.914: [EF 0E 7C 6B 00 00 00 00 00 45 1D 7F 7F 51]  Byte Write in Direct Mode on Service Track: CV70 value 29 (0x1D, 00011101b).
17:45:18.930: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
17:45:22.008: [EF 0E 7C 6B 00 00 00 00 00 45 1D 7F 7F 51]  Byte Write in Direct Mode on Service Track: CV70 value 29 (0x1D, 00011101b).
17:45:22.024: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
17:45:22.242: [E7 0E 7C 6B 00 00 06 47 00 45 1D 7F 7F 18]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV70 value 29 (0x1D, 00011101b).
17:45:22.789: [A0 25 00 7A]  Set speed of loco in slot 37 to 0.
17:45:23.102: [EF 0E 7C 6B 00 00 00 00 00 45 1D 7F 7F 51]  Byte Write in Direct Mode on Service Track: CV70 value 29 (0x1D, 00011101b).
17:45:23.102: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
17:45:24.789: [A0 0C 00 53]  Set speed of loco in slot 12 to 0.
17:45:25.461: [E7 0E 7C 6B 00 00 04 47 00 45 1D 7F 7F 1A]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV70 value 29 (0x1D, 00011101b).
17:45:26.180: [EF 0E 7C 6B 00 00 00 00 00 46 28 7F 7F 67]  Byte Write in Direct Mode on Service Track: CV71 value 40 (0x28, 00101000b).
17:45:26.180: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
17:45:28.071: [E7 0E 7C 6B 00 00 03 47 00 46 28 7F 7F 2B]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV71 value 40 (0x28, 00101000b).
17:45:28.133: [EF 0E 7C 6B 00 00 00 00 00 47 2F 7F 7F 61]  Byte Write in Direct Mode on Service Track: CV72 value 47 (0x2F, 00101111b).
17:45:28.133: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
17:45:29.539: [E7 0E 7C 6B 00 00 02 47 00 47 2F 7F 7F 2C]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV72 value 47 (0x2F, 00101111b).
17:45:29.602: [EF 0E 7C 6B 00 00 00 00 00 48 38 7F 7F 79]  Byte Write in Direct Mode on Service Track: CV73 value 56 (0x38, 00111000b).
17:45:29.602: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
17:45:31.961: [E7 0E 7C 6B 00 00 04 47 00 48 38 7F 7F 32]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV73 value 56 (0x38, 00111000b).
17:45:32.680: [EF 0E 7C 6B 00 00 00 00 00 48 38 7F 7F 79]  Byte Write in Direct Mode on Service Track: CV73 value 56 (0x38, 00111000b).
17:45:32.696: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
[... cut ...]
17:47:16.113: [E7 0E 7C 6B 00 00 03 47 00 5E 00 7F 7F 1B]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV95 value 0 (0x00, 00000000b).
17:47:26.453: [BB 7B 00 3F]  Request Fast Clock information.
17:47:26.464: [E7 0E 7B 00 7F 7F 7F 47 7F 00 70 66 08 34]  Response Fast Clock is  Frozen, rate is 0:1. Day 0, 23:00. Last set by ID 0x08 0x66 (1126).
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available.
17:47:30.440: [A0 25 00 7A]  Set speed of loco in slot 37 to 0.
17:47:32.437: [A0 0C 00 53]  Set speed of loco in slot 12 to 0.

dheap, 2017-03-16:
> If/when you get the chance, could you rerun the speed table test with the timestamps (and raw data) turned on? I'm working on a hypothesis about read speed. I just need until the 1st reject and the next two messages after that.

Reads are a red herring. It's about overlapping writes because the request to write CV32 is being issued before the expected reply is received from the write CV31 request. The code is going ahead after the slot write command is issued, while the DCS240 is still busy writing to the decoder. It needs to wait for Programming Response to Write commands before issuing the next slot command and the logs show it clearly is not.

Dave

dheap, 2017-03-16:
Here we go again
`17:45:03.414: [EF 0E 7C 6B 00 00 00 00 00 42 01 7F 7F 4A] Byte Write in Direct Mode on Service Track: CV67 value 1 (0x01, 00000001b).
17:45:03.414: [B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
17:45:04.977: [BB 7F 00 3B] Request Command Station OpSwitches (or DCS210/DCS240 check for multiple command stations on LocoNet).
17:45:06.492: [EF 0E 7C 6B 00 00 00 00 00 42 01 7F 7F 4A] Byte Write in Direct Mode on Service Track: CV67 value 1 (0x01, 00000001b).
17:45:06.508: [B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.
17:45:07.571: [EF 0E 7C 6B 00 00 00 00 00 42 01 7F 7F 4A] Byte Write in Direct Mode on Service Track: CV67 value 1 (0x01, 00000001b).
17:45:07.571: [B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.
17:45:07.696: [E7 0E 7C 6B 00 00 08 47 00 42 01 7F 7F 0D] Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV67 value 1 (0x01, 00000001b).`

The code has issued multiple slot commands between the write request at 17:45:03.414 and the expected reply at 17:45:07.696. The code needs to heed the 01 in the third byte of the first LONG_ACK and wait for the Programming Response.

It appears the code is confusing the ACK to the Slot Write command with the Programming Response to the actual CV write to acknowledge that it has been done.

In the case of reads, the code has to wait for the Programming Response to get the value.

In the case of writes, the code also needs to wait for the Programming Response before sending any further commands, even though there is no value to harvest from the Programming Response.

bobjacobsen, 2017-03-16:
> > If/when you get the chance, could you rerun the speed table test with the timestamps (and raw data) turned on? I'm working on a hypothesis about read speed. I just need until the 1st reject and the next two messages after that.

> Reads are a red herring. It's about overlapping writes because the request to write CV32 is being issued before the expected reply is received from the write CV31 request. The code is going ahead [...]

All true, all understood.  The question is _why_.  It behaves differently on the failing systems(s) than it does on my test setup.  Hence the request for the timing information.

bobjacobsen, 2017-03-16:
I'm still hoping for a programming operation with timestamps (this is an Op-switch ref which might be different), but the timestamps here are a clue:
```
17:45:03.414: [B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
17:45:04.977: [BB 7F 00 3B] Request Command Station OpSwitches (or DCS210/DCS240 check for multiple command stations on LocoNet).
17:45:06.492: [EF 0E 7C 6B 00 00 00 00 00 42 01 7F 7F 4A] Byte Write in Direct Mode on Service Track: CV67 value 1 (0x01, 00000001b).
17:45:06.508: [B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.
```

If I assume there was a slot write just before the top, note that it's just over 2 seconds before the next request is made, which gets rejected.

That two seconds is the timeout that's in `SlotManager.restartEndOfProgrammingTimer()`. The value comes history, and may no longer be relevant.  But it seems this command station doing this operation is slower than past ones.

If you've got a convenient copy of the source, try changing the "2000" in lines 1041 and 1049 of jmri/jmrix/loconet/SlotManager to a larger value, e.g. 5000.

dheap, 2017-03-16:
I just extracted this from Mick's log. Here is an expanded extract:
@mickmoignardOAC 
```
17:45:01.461: [EF 0E 7C 6B 00 00 00 00 00 1C 22 7F 7F 37] Byte Write in Direct Mode on Service Track: CV29 value 34 (0x22, 00100010b).
17:45:01.461: [B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
17:45:03.336: [E7 0E 7C 6B 00 00 03 47 00 1C 22 7F 7F 7B] Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV29 value 34 (0x22, 00100010b).
17:45:03.414: [EF 0E 7C 6B 00 00 00 00 00 42 01 7F 7F 4A] Byte Write in Direct Mode on Service Track: CV67 value 1 (0x01, 00000001b).
17:45:03.414: [B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
17:45:04.977: [BB 7F 00 3B] Request Command Station OpSwitches (or DCS210/DCS240 check for multiple command stations on LocoNet).
17:45:06.492: [EF 0E 7C 6B 00 00 00 00 00 42 01 7F 7F 4A] Byte Write in Direct Mode on Service Track: CV67 value 1 (0x01, 00000001b).
17:45:06.508: [B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.
17:45:07.571: [EF 0E 7C 6B 00 00 00 00 00 42 01 7F 7F 4A] Byte Write in Direct Mode on Service Track: CV67 value 1 (0x01, 00000001b).
17:45:07.571: [B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.
17:45:07.696: [E7 0E 7C 6B 00 00 08 47 00 42 01 7F 7F 0D] Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV67 value 1 (0x01, 00000001b).
17:45:08.633: [EF 0E 7C 6B 00 00 00 00 00 43 0A 7F 7F 40] Byte Write in Direct Mode on Service Track: CV68 value 10 (0x0A, 00001010b).
17:45:08.633: [B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
17:45:11.008: [E7 0E 7C 6B 00 00 04 47 00 43 0A 7F 7F 0B] Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV68 value 10 (0x0A, 00001010b).
17:45:11.711: [EF 0E 7C 6B 00 00 00 00 00 43 0A 7F 7F 40] Byte Write in Direct Mode on Service Track: CV68 value 10 (0x0A, 00001010b).
17:45:11.727: [B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
17:45:13.602: [E7 0E 7C 6B 00 00 03 47 00 43 0A 7F 7F 0C] Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV68 value 10 (0x0A, 00001010b).
17:45:13.664: [EF 0E 7C 6B 00 00 00 00 00 44 13 7F 7F 5E] Byte Write in Direct Mode on Service Track: CV69 value 19 (0x13, 00010011b).
17:45:13.680: [B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
17:45:16.758: [EF 0E 7C 6B 00 00 00 00 00 44 13 7F 7F 5E] Byte Write in Direct Mode on Service Track: CV69 value 19 (0x13, 00010011b).
17:45:16.774: [B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.
17:45:17.836: [EF 0E 7C 6B 00 00 00 00 00 44 13 7F 7F 5E] Byte Write in Direct Mode on Service Track: CV69 value 19 (0x13, 00010011b).
17:45:17.852: [B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.
17:45:18.914: [E7 0E 7C 6B 00 00 0A 47 00 44 13 7F 7F 1B] Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV69 value 19 (0x13, 00010011b).
17:45:18.914: [EF 0E 7C 6B 00 00 00 00 00 45 1D 7F 7F 51] Byte Write in Direct Mode on Service Track: CV70 value 29 (0x1D, 00011101b).
17:45:18.930: [B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
17:45:22.008: [EF 0E 7C 6B 00 00 00 00 00 45 1D 7F 7F 51] Byte Write in Direct Mode on Service Track: CV70 value 29 (0x1D, 00011101b).
17:45:22.024: [B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.
17:45:22.242: [E7 0E 7C 6B 00 00 06 47 00 45 1D 7F 7F 18] Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV70 value 29 (0x1D, 00011101b).
```
There are examples there with no Op-Switches refs. I don't have hardware. I suspect the hardware is doing a write with read back confirm and this accounts for the delayed response.

But that still doesn't explain why we keep feeding Slot Write commands before a response. We don't do that for reads, as per Val's post (pity no timestamps):

@ValPistilli  Here are readings for Speed Table read:
```
[EF 0E 7C 2B 00 00 00 00 02 1C 7F 7F 7F 28] Programming Track Request: Read Byte in Direct Mode on Service Track variable CV29 value 255 (0xff, 11111111).
[B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 1C 23 7F 7F 3B] Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set CV29 value 35 (0x23, 100011).
[EF 0E 7C 2B 00 00 00 00 02 42 7F 7F 7F 76] Programming Track Request: Read Byte in Direct Mode on Service Track variable CV67 value 255 (0xff, 11111111).
[B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 42 02 7F 7F 44] Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set CV67 value 2 (0x2, 10).
[EF 0E 7C 2B 00 00 00 00 02 43 7F 7F 7F 77] Programming Track Request: Read Byte in Direct Mode on Service Track variable CV68 value 255 (0xff, 11111111).
[B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 43 0B 7F 7F 4C] Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set CV68 value 11 (0xb, 1011).
[EF 0E 7C 2B 00 00 00 00 02 44 7F 7F 7F 70] Programming Track Request: Read Byte in Direct Mode on Service Track variable CV69 value 255 (0xff, 11111111).
[B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 44 14 7F 7F 54] Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set CV69 value 20 (0x14, 10100).
[EF 0E 7C 2B 00 00 00 00 02 45 7F 7F 7F 71] Programming Track Request: Read Byte in Direct Mode on Service Track variable CV70 value 255 (0xff, 11111111).
[B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
[E7 0E 7C 2B 00 00 02 47 00 45 1E 7F 7F 5F] Programming Track Response: Read Byte in Direct Mode on Service Track Was Successful, set CV70 value 30 (0x1e, 11110).
[EF 0E 7C 2B 00 00 00 00 02 46 7F 7F 7F 72] Programming Track Request: Read Byte in Direct 
```

devel-bobm, 2017-03-16:
@dheap,  the message 

`[BB 7F 00 3B] Request Command Station OpSwitches (or DCS210/DCS240 check for multiple command stations on LocoNet).`

is a message where the command station is attempting to find out if any other command stations are present on LocoNet.  The DCS240 (and DCS210, I think) sends this occasionally.  When it sends the message, it does not send a reply, and it does not expect a reply.  If another command station is present, it will reply to this message, and the DCS240 which sent the request will begin beeping and stop sending track packets.

So you should be able to _ignore_ this message.

dheap, 2017-03-17:
@devel-bobm Thanks for that clarification. That confirms my feeling that I doubt it has anything to do with the current issue.

dheap, 2017-03-17:
@mickmoignardOAC A speed table read with Timestamps and Raw Data would help us see how long reads are taking.

dheap, 2017-03-17:
@bobjacobsen said:

> That two seconds is the timeout that's in SlotManager.restartEndOfProgrammingTimer(). The value comes history, and may no longer be relevant. But it seems this command station doing this operation is slower than past ones.

So if the timeout expires for any reason, the SlotManager simply tries a Slot Write again without checking the Slot Status to see if it is still busy, or even attempting to clear the slot (if that is possible)? Doesn't seem a robust way of doing things...

We now need someone with DCS240 hardware to test this...

bobjacobsen, 2017-03-17:
> > That two seconds is the timeout that's in SlotManager.restartEndOfProgrammingTimer(). The value comes history, and may no longer be relevant. But it seems this command station doing this operation is slower than past ones.

> So if the timeout expires for any reason, the SlotManager simply tries a Slot Write again without checking the Slot Status to see if it is still busy, or even attempting to clear the slot (if that is possible)? Doesn't seem a robust way of doing things...

"Robust" and "LocoNet" don't belong in the same sentence.  

The LocoNet design doesn't, in general, provide useful recovery mechanisms for timeouts, or lots of other errors for that matter.  For example, there _isn't_ a way of 'checking the Slot Status to see if it is still busy' for a programming operation other than trying a programming operation and seeing if it's rejected. So that's what the program does.

But it's worse than that, because (AFAIK) there's no _reliable_ way to tell whether the rejected write will be followed by any specific message.  There might be a way, but all the ones that have been tried have so many exceptions as to be worthless.  So you can't code "in this case, wait for the message, otherwise just proceed".  Instead, the code just tries again after a couple hundred milliseconds, waiting for whatever the problem is to clear.

If the problem is _very_ intermittent, that works.  If the probability of failure is large, as it is in this case, you end up in overlapping cycles of recovery.  Since LocoNet doesn't have a (reliable) state machine, that means things just go sideways. The only fix for that is to let things just die down, and start over after a while. Effectively, that just means a long enough timeout.

bobjacobsen, 2017-03-17:
There's now [installers with a larger timeout](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/1372/) available from [Jenkins](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/1372/).  When people have time, it would be helpful if people could test them.  I'm away from hardware until at least Sunday.

mickmoignardOAC, 2017-03-17:
Speed Table Read with 4.7.2...

09:01:53.083: [EF 0E 7C 2B 00 00 00 00 02 1C 7F 7F 7F 28]  Byte Read in Direct Mode on Service Track: CV29.
09:01:53.083: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:01:55.630: [A0 0C 00 53]  Set speed of loco in slot 12 to 0.
09:01:56.771: [E7 0E 7C 2B 00 00 05 46 00 1C 22 7F 7F 3C]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV29 value 34 (0x22, 00100010b).
09:01:56.849: [EF 0E 7C 2B 00 00 00 00 02 42 7F 7F 7F 76]  Byte Read in Direct Mode on Service Track: CV67.
09:01:56.864: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:01:59.567: [E7 0E 7C 2B 00 00 03 46 00 42 01 7F 7F 47]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV67 value 1 (0x01, 00000001b).
09:01:59.646: [EF 0E 7C 2B 00 00 00 00 02 43 7F 7F 7F 77]  Byte Read in Direct Mode on Service Track: CV68.
09:01:59.646: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:02:02.833: [E7 0E 7C 2B 00 00 04 46 00 43 0A 7F 7F 4A]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV68 value 10 (0x0A, 00001010b).
09:02:02.958: [EF 0E 7C 2B 00 00 00 00 02 44 7F 7F 7F 70]  Byte Read in Direct Mode on Service Track: CV69.
09:02:02.958: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:02:05.192: [E7 0E 7C 2B 00 00 02 46 00 44 13 7F 7F 52]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV69 value 19 (0x13, 00010011b).
09:02:05.271: [EF 0E 7C 2B 00 00 00 00 02 45 7F 7F 7F 71]  Byte Read in Direct Mode on Service Track: CV70.
09:02:05.271: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:02:07.489: [E7 0E 7C 2B 00 00 02 46 00 45 1D 7F 7F 5D]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV70 value 29 (0x1D, 00011101b).
09:02:07.567: [EF 0E 7C 2B 00 00 00 00 02 46 7F 7F 7F 72]  Byte Read in Direct Mode on Service Track: CV71.
09:02:07.567: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:02:14.583: [E7 0E 7C 2B 00 00 0C 46 00 46 28 7F 7F 65]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV71 value 40 (0x28, 00101000b).
09:02:14.661: [EF 0E 7C 2B 00 00 00 00 02 47 7F 7F 7F 73]  Byte Read in Direct Mode on Service Track: CV72.
09:02:14.661: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:02:19.271: [E7 0E 7C 2B 00 00 07 46 00 47 2F 7F 7F 68]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV72 value 47 (0x2F, 00101111b).
09:02:19.349: [EF 0E 7C 2B 00 00 00 00 02 48 7F 7F 7F 7C]  Byte Read in Direct Mode on Service Track: CV73.
09:02:19.349: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:02:22.052: [E7 0E 7C 2B 00 00 03 46 00 48 38 7F 7F 74]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV73 value 56 (0x38, 00111000b).
09:02:22.130: [EF 0E 7C 2B 00 00 00 00 02 49 7F 7F 7F 7D]  Byte Read in Direct Mode on Service Track: CV74.
09:02:22.130: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:02:24.833: [E7 0E 7C 2B 00 00 03 46 00 49 43 7F 7F 0E]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV74 value 67 (0x43, 01000011b).
09:02:24.911: [EF 0E 7C 2B 00 00 00 00 02 4A 7F 7F 7F 7E]  Byte Read in Direct Mode on Service Track: CV75.
09:02:24.911: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:02:27.552: [A0 0C 00 53]  Set speed of loco in slot 12 to 0.
09:02:27.614: [E7 0E 7C 2B 00 00 03 46 00 4A 4C 7F 7F 02]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV75 value 76 (0x4C, 01001100b).
09:02:27.692: [EF 0E 7C 2B 00 00 00 00 02 4B 7F 7F 7F 7F]  Byte Read in Direct Mode on Service Track: CV76.
09:02:27.692: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:02:31.833: [E7 0E 7C 2B 00 00 06 46 00 4B 55 7F 7F 1F]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV76 value 85 (0x55, 01010101b).
09:02:32.052: [EF 0E 7C 2B 00 00 00 00 02 4C 7F 7F 7F 78]  Byte Read in Direct Mode on Service Track: CV77.
09:02:32.052: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:02:36.660: [E7 0E 7C 2B 00 00 07 46 00 4C 61 7F 7F 2D]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV77 value 97 (0x61, 01100001b).
09:02:36.770: [EF 0E 7C 2B 00 00 00 00 02 4D 7F 7F 7F 79]  Byte Read in Direct Mode on Service Track: CV78.
09:02:36.782: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:02:39.510: [BB 7B 00 3F]  Request Fast Clock information.
09:02:39.522: [E7 0E 7B 00 7F 7F 7F 4E 7F 00 70 66 08 3D]  Response Fast Clock is  Frozen, rate is 0:1. Day 0, 23:00. Last set by ID 0x08 0x66 (1126).
	Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Busy.
09:02:42.353: [E7 0E 7C 2B 00 00 09 46 00 4D 69 7F 7F 2A]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV78 value 105 (0x69, 01101001b).
09:02:42.419: [EF 0E 7C 2B 00 00 00 00 02 4E 7F 7F 7F 7A]  Byte Read in Direct Mode on Service Track: CV79.
09:02:42.431: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:02:44.800: [BB 7F 00 3B]  Request Command Station OpSwitches (or DCS210/DCS240 check for multiple command stations on LocoNet).
[... cut ...]
09:03:47.622: [E7 0E 7C 2B 00 00 02 46 00 41 00 7F 7F 44]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV66 value 0 (0x00, 00000000b).
09:03:47.723: [EF 0E 7C 2B 00 00 00 00 02 5E 7F 7F 7F 6A]  Byte Read in Direct Mode on Service Track: CV95.
09:03:47.723: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:03:49.940: [E7 0E 7C 2B 00 00 02 46 00 5E 00 7F 7F 5B]  Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV95 value 0 (0x00, 00000000b).
09:04:03.287: [A0 0C 00 53]  Set speed of loco in slot 12 to 0.

mickmoignardOAC, 2017-03-17:
I'll install the mentioned 4.7.3ish release and see if that helps any, now.

mickmoignardOAC, 2017-03-17:
I've tried the 473ish package.  No obvious differences. See the attached file

[DCS240-DP473ish-Loksound4-run-at-20170317.txt](https://github.com/JMRI/JMRI/files/850207/DCS240-DP473ish-Loksound4-run-at-20170317.txt)

dheap, 2017-03-17:
We are seeing read response delays of around 3 seconds here:
```
09:01:56.849: [EF 0E 7C 2B 00 00 00 00 02 42 7F 7F 7F 76] Byte Read in Direct Mode on Service Track: CV67.
09:01:56.864: [B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
09:01:59.567: [E7 0E 7C 2B 00 00 03 46 00 42 01 7F 7F 47] Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV67 value 1 (0x01, 00000001b).
09:01:59.646: [EF 0E 7C 2B 00 00 00 00 02 43 7F 7F 7F 77] Byte Read in Direct Mode on Service Track: CV68.
09:01:59.646: [B4 6F 01 25] LONG_ACK: The Slot Write command was accepted.
09:02:02.833: [E7 0E 7C 2B 00 00 04 46 00 43 0A 7F 7F 4A] Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV68 value 10 (0x0A, 00001010b).

```

dheap, 2017-03-17:
> @mickmoignardOAC I've tried the 473ish package. No obvious differences. See the attached file
```
09:15:59.704: [A0 0C 00 53]  Set speed of loco in slot 12 to 0.
09:16:06.218: [EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
09:16:06.218: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:16:07.908: [E7 0E 7C 6B 00 00 03 46 00 1E 10 7F 7F 4A]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV31 value 16 (0x10, 00010000b).
09:16:07.946: [EF 0E 7C 6B 00 00 00 00 00 1F 02 7F 7F 14]  Byte Write in Direct Mode on Service Track: CV32 value 2 (0x02, 00000010b).
09:16:07.950: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:16:10.044: [EF 0E 7C 2B 00 00 00 00 12 00 7F 7F 7F 24]  Byte Read in Direct Mode on Service Track: CV257.
09:16:10.044: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
09:16:10.310: [E7 0E 7C 6B 00 00 04 46 00 1F 02 7F 7F 5E]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV32 value 2 (0x02, 00000010b).
09:16:11.467: [EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
09:16:11.467: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:16:13.510: [EF 0E 7C 6B 00 00 00 00 00 1F 02 7F 7F 14]  Byte Write in Direct Mode on Service Track: CV32 value 2 (0x02, 00000010b).
09:16:13.510: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
09:16:13.547: [EF 0E 7C 2B 00 00 00 00 12 00 7F 7F 7F 24]  Byte Read in Direct Mode on Service Track: CV257.
09:16:13.558: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
09:16:13.598: [E7 0E 7C 6B 00 00 04 46 00 1E 10 7F 7F 4D]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV31 value 16 (0x10, 00010000b).
09:16:14.864: [EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
09:16:14.865: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
09:16:16.903: [EF 0E 7C 6B 00 00 00 00 00 1F 02 7F 7F 14]  Byte Write in Direct Mode on Service Track: CV32 value 2 (0x02, 00000010b).
09:16:16.910: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
09:16:16.949: [EF 0E 7C 2B 00 00 00 00 12 01 7F 7F 7F 25]  Byte Read in Direct Mode on Service Track: CV258.
09:16:16.950: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
09:16:17.023: [E7 0E 7C 6B 00 00 04 46 00 1E 10 7F 7F 4D]  Programming Response: Write Byte in Direct Mode on Service Track Was Successful: CV31 value 16 (0x10, 00010000b).
09:16:17.348: [A0 0C 00 53]  Set speed of loco in slot 12 to 0.
```

We are seeing Slot Write commands being issued only 2 seconds apart for successive writes and only around 40 milliseconds apart when switching between Write (second indexed CV) and Read (actual CV). There could be two separate timing issues.

Exactly what JMRI version did you download? It needs to be JMRI.4.7.3ish-201703170546-jenkins-R7482fc6 or later  (Build #1372 or later) from http://jmri.tagadab.com/jenkins/job/Development/job/Packages/

dheap, 2017-03-17:
> Exactly what JMRI version did you download? It needs to be JMRI.4.7.3ish-201703170546-jenkins-R7482fc6 or later (Build #1372 or later)

 Sorry, just saw this version in your log.

bobjacobsen, 2017-03-17:
> I've tried the 473ish package. No obvious differences.

First, thank you.  I'm sorry this is taking so long, but these tests are _really_ helpful.

That trace is showing that the SlotManager is not handling the timeout condition properly.  It's still using 2 seconds for timeout in this case, even though I changed the length of the proper timer to 10 seconds.  Something inside SlotManager is confused about what it's doing.  

More to follow, though perhaps not until this evening.

bobjacobsen, 2017-03-17:
Working on this, will have a PR shortly. 

There's something going on here that I think will make it Pretty Ugly to use a DCS240 with the LokSound decoders, or really any indexed decoders.  If it takes 3-5 seconds to write the primary index, plus another couple seconds for the secondary, indexed-CV operations are going to be Really Slow.

Does somebody understand why the DCS240 is so slow? I heard mention of a "read after write": if that's what it's doing, is there a way to turn that off?  And why should direct mode (see trace) be so appallingly slow anyway?

bobjacobsen, 2017-03-17:
PR #3254 should have a fix for the issue with LokSound V4 indexed operations (Ha! Said that before!)

Will work it through the system and put a note here when downloads available.

bobjacobsen, 2017-03-17:
New installers with #3254 fixes are [available on Jenkins](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/) in [the usual place](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/), build 1374 or later.

mickmoignardOAC, 2017-03-17:
Will look at it as soon as I can but it probably won't be till Sunday at the earliest.

Mick

______________________________________
Mick Moignard
mick@mickmoignard.com
p:+44 7774 652504
skype: mickmoignard

> 
> New installers with #3254 fixes are available on Jenkins in the usual place, build 1374 or later.

This email and any files transmitted with it are intended only for the person or entity to which it is addressed and may contain privileged or confidential information. If you're not the intended recipient, disclosing, copying, distributing or using this information is prohibited. If you've received this email in error, please contact the sender and delete all copies. This email may contain personal views which are not the views of OAC. We monitor our email system and may record your emails. OAC PLC is registered in England | Company no.: 4663795 | Registered office: 141-142 Fenchurch Street, London, EC3M 6BL, United Kingdom

dheap, 2017-03-18:
> @bobjacobsen There's something going on here that I think will make it Pretty Ugly to use a DCS240 with the LokSound decoders, or really any indexed decoders. If it takes 3-5 seconds to write the primary index, plus another couple seconds for the secondary, indexed-CV operations are going to be Really Slow.

Took the time to do some LokSound V4 speed test comparisons with the three systems I have access to:
```
07:39:25.048: Power Cab Read Type from Decoder
07:39:29.420: [9E]  Enter Programming track mode
07:39:31.274: [21]  command completed successfully
07:39:32.276: [A9 00 08]  Read CV 8 in direct mode
07:39:33.176: [97 21]  Reply: 97 21
07:39:33.179: [A9 00 07]  Read CV 7 in direct mode
07:39:34.026: [FF 21]  Reply: FF 21
07:39:34.028: [A8 00 1F 00]  Program CV 31 with data < 00 > in direct mode
07:39:34.084: [21]  command completed successfully
07:39:34.086: [A8 00 20 FF]  Program CV 32 with data < FF > in direct mode
07:39:34.413: [21]  command completed successfully
07:39:34.415: [A9 01 05]  Read CV 261 in direct mode
07:39:35.584: [3D 21]  Reply: 3D 21
07:39:35.586: [A9 01 06]  Read CV 262 in direct mode
07:39:36.572: [00 21]  Reply: 00 21
07:39:36.574: [A9 01 07]  Read CV 263 in direct mode
07:39:37.560: [00 21]  Reply: 00 21
07:39:37.562: [A9 01 08]  Read CV 264 in direct mode
07:39:38.526: [02 21]  Reply: 02 21
07:39:40.389: [A9 00 1D]  Read CV 29 in direct mode
07:39:41.176: [2A 21]  Reply: 2A 21
07:39:41.176: [A9 00 01]  Read CV 1 in direct mode
07:39:42.151: [03 21]  Reply: 03 21
07:39:42.151: [A9 00 11]  Read CV 17 in direct mode
07:39:43.099: [D1 21]  Reply: D1 21
07:39:43.101: [A9 00 12]  Read CV 18 in direct mode
07:39:44.032: [FF 33]  Reply: FF 33
07:39:44.033: [A9 00 13]  Read CV 19 in direct mode
07:39:45.030: [00 21]  Reply: 00 21
07:39:45.234: [9F]  Exit Programming track mode
07:39:47.093: [21]  command completed successfully
```
Power Cab Read Type from Decoder Total 18 seconds.
```
07:40:29.390: Power Cab Read Speed Table
07:41:11.891: [9E]  Enter Programming track mode
07:41:13.707: [21]  command completed successfully
07:41:14.709: [A9 00 1D]  Read CV 29 in direct mode
07:41:15.646: [2A 21]  Reply: 2A 21
07:41:15.704: [A9 00 43]  Read CV 67 in direct mode
07:41:16.657: [03 21]  Reply: 03 21
07:41:16.679: [A9 00 44]  Read CV 68 in direct mode
07:41:17.605: [03 21]  Reply: 03 21
...
07:41:43.582: [A9 00 42]  Read CV 66 in direct mode
07:41:44.559: [00 21]  Reply: 00 21
07:41:44.570: [A9 00 5F]  Read CV 95 in direct mode
07:41:45.532: [00 21]  Reply: 00 21
07:41:45.742: [9F]  Exit Programming track mode
07:41:47.595: [21]  command completed successfully
```
Power Cab Read Speed Table Total 36 seconds.
Reads slightly under 1 second.
```
07:42:03.486: Power Cab Write Speed Table
07:42:10.569: [9E]  Enter Programming track mode
07:42:12.371: [21]  command completed successfully
07:42:13.373: [A8 00 1D 2A]  Program CV 29 with data < 2A > in direct mode
07:42:13.433: [21]  command completed successfully
07:42:13.453: [A8 00 43 01]  Program CV 67 with data < 01 > in direct mode
07:42:13.759: [21]  command completed successfully
07:42:13.773: [A8 00 44 03]  Program CV 68 with data < 03 > in direct mode
07:42:14.088: [21]  command completed successfully
...
07:42:23.285: [A8 00 42 00]  Program CV 66 with data < 00 > in direct mode
07:42:23.604: [21]  command completed successfully
[... 303 lines cut ...]
P> "
10:10:23.177: []  cmd: "C 0382"
10:10:25.819: []  rep: "= h00
P> "
10:10:25.986: []  cmd: "C 0031 016"
10:10:26.768: []  rep: "OK
P> "
10:10:26.768: []  cmd: "C 0032 004"
10:10:27.280: []  rep: "OK
P> "
10:10:27.280: []  cmd: "C 0383"
10:10:29.922: []  rep: "= h00
P> "
10:10:30.090: []  cmd: "C 0031 016"
10:10:30.865: []  rep: "OK
P> "
10:10:30.865: []  cmd: "C 0032 004"
10:10:31.377: []  rep: "OK
P> "
10:10:31.377: []  cmd: "C 0384"
10:10:34.025: []  rep: "= h00
P> "
```
SPROG Read Function Map Total 42 minutes.
Indexed Reads around 3-4 seconds.

mickmoignardOAC, 2017-03-18:
The changes in build 1374 haven't entirely worked: attached are some results using build 1376:
[DCS240-DP473ish20170318...-Loksound4.txt](https://github.com/JMRI/JMRI/files/852786/DCS240-DP473ish20170318.-Loksound4.txt)

dheap, 2017-03-18:
If you use "Choose Log File" followed by "Start Logging" in the LocoNet Monitor window rather than copying and pasting the visible buffer(it is trimmed when it gets too large) you should get the log file up to and including the critical part where it goes wrong, even if you have to crash JMRI.

From a quick glance at what you got, it appears that there is still a short timeout of 2 seconds applying with the index CV write, causing some sort of buffer overload and subsequent hang of the DCS240 (I assume a power cycle recovers the hardware). I suspect there may be still some work to do in the MultiIndexProgrammerFacade class.

> @mickmoignardOAC
> 2. Function Map pane read: Start 21:05:50, end 22:15 or so, but... 
> it went well for a time, and then the DCS240 programmer appears to have crashed. The DP trace doesn't show all the details, just the last few minutes or so, no record of when it crashed or why.
> I guess I'll have to watch it run sometime tomorrow to try and capture what goes wrong. Here's what there is of the trace: 
> What I can tell from the DP Function Map pane is that it got to row 29, the CV pane suggests that it went wrong on 16.4.454, which is shown in red as "Unknown". 16.3.453 was read successfully.
> The DCS240 programmer is crashed, hard, too. Quitting DP does not bring it back to life.  
> 
```
22:09:08.423: [EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
22:09:08.423: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
22:09:10.107: [EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
22:09:10.108: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
22:09:11.635: [EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
22:09:11.636: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
22:09:13.347: [EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
22:09:13.347: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
22:09:14.865: [EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
22:09:14.865: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
22:09:16.583: [EF 0E 7C 6B 00 00 00 00 00 1E 10 7F 7F 07]  Byte Write in Direct Mode on Service Track: CV31 value 16 (0x10, 00010000b).
22:09:16.584: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
```

ValPistilli, 2017-03-18:
Dave, if you are talking to me, I was writing a value of 16 to CV31 but the decoder doesn’t have CV31. IS there another CV I could write to?

Val Pistilli
Webmaster
www.rpmvalleyforge.com
www.lehighvalleysgaugers.org
IN GOD WE TRUST

bobjacobsen, 2017-03-18:
@mickmoignardOAC Thanks for that trace.  

The first part, before your first notes, looks ok to me. Some of the _write_ operations are taking a very long time. It wrote CVs ending with 93,94, 2, 5, 66, 95. Then traffic stops for 40 seconds. That all looks normal (except slow due to the DCS240 Really Poor Write Performance).  What were you trying to write? I'd like to check if those are the correct last CVs.

The 2nd part log is all rejected requests. The fast-clock shows that the programming track thinks its busy, but it never gets unbusy. JMRI just retires every now and then, for lack of anything better to do. 

So something went wrong at the start of that second part, but it's not clear what. 

Whatever happened, it seems to have messed up the DCS240 internally. 

@dheap what are you seeing that makes you think there's a buffer overload?  The sequencing looks right to me, but perhaps I'm missing something.

dheap, 2017-03-19:
> @bobjacobsen So something went wrong at the start of that second part, but it's not clear what.
>
> Whatever happened, it seems to have messed up the DCS240 internally.
> 
> @dheap what are you seeing that makes you think there's a buffer overload? The sequencing looks right to me, but perhaps I'm missing something.

@mickmoignardOAC says it got to row 29 ( of 40) OK. But then something has failed and JMRI is spitting out write requests for CV31 at 2 second intervals. We know that writes take at least 2.5 seconds, sometimes more so we shouldn't be retrying at such a short interval.

I said "some sort of buffer overload" because it wasn't quite the correct term.

I'm out of my depth here with Digitrax , but is the DCS240 communicating with JMRI via the USB port or via an external LocoNet connector and something like a PR3 or a LocoBuffer-USB?

Its a bit hard to see what went wrong (hope @mickmoignardOAC can catch it with log file), but my thought was that the repeated frequent Slot Write rejections were possibly actually overflowing a stack somewhere. But that bit is speculation based on what I've seen happen with some NCE firmware.

I think we need to:
- Avoid such a short retry interval with the indexed CV writes.
- Catch the log at the point things go belly up.

As a maybe-interesting aside, NCE Power Pro firmware sometimes "loses the plot" when reading decoders with many CVs such as QSI and ESU. The command station just locks up with the status light flashing. The NCE recovery procedure is:
- Don't abort JMRI, let it keep trying.
- Power cycle the command station (this is OK with the Power Pro uses an external USB-powered USB-Serial converter so the port is not lost entirely).
- Wait. JMRI eventually picks up and continues.
- When finished, use Read Changes to pick up misread CVs.

[43 more replies]

PR #3252 (2017-03-17): Length LocoNet programming timeout
Lengthen LocoNet programming timeout to try to fix #3171. Includes test from #3171 sequence.
closed 2021-04-14 by stale[bot]