JMRI/JMRI #3188: UserNames not loaded into OBlocks

petecressman, 2017-03-06:
recent changes do not load user names

petecressman, 2017-03-06:
public void propertyChange(PropertyChangeEvent e) in AbstractManager is calling
                    _tuser.get(now).setUserName(null);
after the OBlock has already called
        block.setUserName(userName);

simple test file:
<?xml-stylesheet href="/xml/XSLT/panelfile-2-9-6.xsl" type="text/xsl"?>
<layout-config xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://jmri.org/xml/schema/layout-2-9-6.xsd">
  <jmriversion>
    <major>4</major>
    <minor>7</minor>
    <test>2</test>
    <modifier>.2ish</modifier>
  </jmriversion>
  <signalmasts class="jmri.managers.configurexml.DefaultSignalMastManagerXml" />
  <signalgroups class="jmri.managers.configurexml.DefaultSignalGroupManagerXml" />
  <oblocks class="jmri.jmrit.logix.configurexml.OBlockManagerXml">
    <oblock systemName="OB1" userName="OneBlock" length="0.0" units="false" curve="0" permissive="false" speedNotch="">
      <userName>OneBlock</userName>
    </oblock>
  </oblocks>
  <warrants class="jmri.jmrit.logix.configurexml.WarrantManagerXml">
    <nxparams>
      <maxspeed>0.75</maxspeed>
      <haltstart>no</haltstart>
    </nxparams>
  </warrants>
  <signalmastlogics class="jmri.managers.configurexml.DefaultSignalMastLogicManagerXml">
    <logicDelay>500</logicDelay>
  </signalmastlogics>
  <entryexitpairs class="jmri.jmrit.signalling.configurexml.EntryExitPairsXml" />
  <paneleditor class="jmri.jmrit.display.controlPanelEditor.configurexml.ControlPanelEditorXml" name="Control Panel" x="510" y="189" height="499" width="688" editable="yes" positionable="yes" showtooltips="yes" controlling="yes" hide="no" panelmenu="yes" scrollable="both" state="0" shapeSelect="yes">
    <icons>
      <visible>
        <url>program:resources/icons/throttles/RoundRedCircle20.png</url>
      </visible>
      <path_edit>
        <url>program:resources/icons/greenSquare.gif</url>
      </path_edit>
      <hidden>
        <url>program:resources/icons/Invisible.gif</url>
      </hidden>
      <to_arrow>
        <url>program:resources/icons/track/toArrow.gif</url>
      </to_arrow>
      <from_arrow>
        <url>program:resources/icons/track/fromArrow.gif</url>
      </from_arrow>
    </icons>
  </paneleditor>
  <filehistory>
    <operation>
      <type>app</type>
      <date>Sun Mar 05 17:35:39 PST 2017</date>
      <filename>JMRI program</filename>
    </operation>
    <operation>
      <type>Store</type>
      <date>Sun Mar 05 17:37:40 PST 2017</date>
      <filename />
    </operation>
  </filehistory>
  
</layout-config>

bobjacobsen, 2017-03-06:
The test file didn't come through, sorry.  The `<` and `>` characters were probably eaten as HTML formatting.  Could you try again with them in triple-back-quote marks or attach the file? (Might have to add a .txt extension to attach it)  

I think the code you're referring to is making sure that no two user names are the same. If a user name is changed to something, anybody else with that same name is cleared to null. Is that what's not working?

petecressman, 2017-03-06:
The test file, anOBlock.***, is a CPE panel with but one additional object, sysName=OB1, uName=OneBlock.  no white space characters.
What happens at AbstractNamedBean.setUserName is (see line 194)
       From                                       old name         new name
OBlockManagerXml.createNewOBlock       null               nullOBlockManagerXml.setUserName            null               OneBlockAbtractManager.setUserName              OneBlock          null

I believe the last call may be in response to the PropertyChangeEvent fired at the first call registering the OBlock with AbstractManager.  Don't know why a name check should change former behavior.

 

   
<?xml version="1.0" encoding="UTF-8"?>

bobjacobsen, 2017-03-07:
#3195 should have a fix for this.

bobjacobsen, 2017-03-07:
#3195 fix is failing tests because of Warrants being written differently when headless. See #3197. Is there a reason for this?

bobjacobsen, 2017-03-08:
This specific program fixed by PR #3199, now merged.

PR #3195 uncovered problems with NxFrame persistence when running headless, which I'll open as a new Issue.

closed 2017-03-08 by bobjacobsen