JMRI/JMRI #3194: Build 1334 Issue - Java Exception : java.util.Objects.requireNonNull

dsand47, 2017-03-07:
This error occurs for a number of my profiles.  If I create a new profile, the error does not occur.

I was able to get around the error by removing the following lines from the user-interface.xml file.

```
        <preferences xmlns="" class="jmri.jmrit.beantable.LogixTableAction">
            <reminderPrompts>
                <reminder>remindSaveLogix</reminder>
            </reminderPrompts>
        </preferences>

```

The session log is attached.
[session.log.txt](https://github.com/JMRI/JMRI/files/823364/session.log.txt)

Dave Sand

rhwood, 2017-03-07:
@dsand47 can you confirm this is fixed in builds from master?

dsand47, 2017-03-07:
Yes, it is fixed.

Dave Sand

PR #3196 (2017-03-07): Use a UserPreferencesManager only if available
Fixes #3194
commit 00b193dd (2017-03-07): Use a UserPreferencesManager only if available (#3196)

* Use a UserPreferencesManager only if available

Fixes #3194

* Potentially fix broken test
closed 2017-03-07 by dsand47