JMRI/JMRI #3201: NxFrame doesn't persist consistently when running headless: Bad MVC in NxFrame

bobjacobsen, 2017-03-08:
(See #3188, #3195 for some background)

The OBlock, Warrant and NxFrame persistence writes and reads the .xml file differently depending on whether running headless or not.  
- In part, this is because there are explicit calls to `GraphicsEnvironment.isHeadless()` in `java/src/jmri/jmrit/logix/configurexml/WarrantManagerXml.java`
- But removing them doesn't fix it, because then there are explicit Swing calls during the load/store that cause errors when headless.  (See below for a typical traceback)

This means we can't include checks of persistence for these classes in our Unit test suites, because they'd operate differently in different environments.  And it's not a good thing to do, because people using these features who run JMRI headless occasionally might lose the content of their files should they store them while running headless.

The fundamental problem seems to be that NxFrame doesn't have a model-view-controller (MVC) structure, where the data (model) is kept separate from the screen-based presentation objects (view).  

The attached [`oblock.xml.txt`](https://github.com/JMRI/JMRI/files/827795/oblock.xml.txt) file provides an easy demo of those.  Rename it to `oblock.xml` and put it in `java/test/jmri/configurexml/load/`, then run the CI tests in headless mode (e.g. as would be run on `Travis Linux headless=YES`)

Typical traceback with `GraphicsEnvironment.isHeadless()` removed:
```
    [junit] ERROR - Unexpected error (Exception) while load(/home/travis/build/JMRI/JMRI/java/test/jmri/configurexml/load/oblock.xml) in adaptor of type jmri.jmrit.logix.configurexml.WarrantManagerXml Exception: java.lang.NullPointerException
    [junit] See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [main] jmri.configurexml.ErrorHandler.?()
    [junit] java.lang.NullPointerException
    [junit] 	at jmri.jmrit.logix.configurexml.WarrantManagerXml.loadNXParams(WarrantManagerXml.java:392)
    [junit] 	at jmri.jmrit.logix.configurexml.WarrantManagerXml.load(WarrantManagerXml.java:204)
    [junit] 	at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:705)
    [junit] 	at jmri.configurexml.ConfigXmlManager.lambda$load$0(ConfigXmlManager.java:628)
    [junit] 	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:301)
    [junit] 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
    [junit] 	at java.awt.EventQueue.access$500(EventQueue.java:97)
    [junit] 	at java.awt.EventQueue$3.run(EventQueue.java:709)
    [junit] 	at java.awt.EventQueue$3.run(EventQueue.java:703)
    [junit] 	at java.security.AccessController.doPrivileged(Native Method)
    [junit] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
    [junit] 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
    [junit] 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
    [junit] 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
    [junit] 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
    [junit] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
    [junit] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
    [junit] 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
    [junit] ERROR - Unexpected error (Exception) while load(/home/travis/build/JMRI/JMRI/java/test/jmri/configurexml/load/oblock.xml) in adaptor of type jmri.jmrit.logix.configurexml.WarrantManagerXml Exception: java.lang.NullPointerException
    [junit] See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [main] jmri.configurexml.ErrorHandler.?()
    [junit] java.lang.NullPointerException
    [junit] 	at jmri.jmrit.logix.configurexml.WarrantManagerXml.loadNXParams(WarrantManagerXml.java:392)
    [junit] 	at jmri.jmrit.logix.configurexml.WarrantManagerXml.load(WarrantManagerXml.java:204)
    [junit] 	at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:705)
    [junit] 	at jmri.configurexml.ConfigXmlManager.lambda$load$0(ConfigXmlManager.java:628)
    [junit] 	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:301)
    [junit] 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
    [junit] 	at java.awt.EventQueue.access$500(EventQueue.java:97)
    [junit] 	at java.awt.EventQueue$3.run(EventQueue.java:709)
    [junit] 	at java.awt.EventQueue$3.run(EventQueue.java:703)
    [junit] 	at java.security.AccessController.doPrivileged(Native Method)
    [junit] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
    [junit] 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
    [junit] 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
    [junit] 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
    [junit] 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
    [junit] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
    [junit] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
    [junit] 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
    [junit] ERROR - Exception thrown while storing to file in addConfigStore Exception: java.lang.NullPointerException
    [junit] See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [main] jmri.configurexml.ErrorHandler.?()
    [junit] java.lang.NullPointerException
    [junit] 	at jmri.jmrit.logix.configurexml.WarrantManagerXml.storeNXParams(WarrantManagerXml.java:184)
    [junit] 	at jmri.jmrit.logix.configurexml.WarrantManagerXml.store(WarrantManagerXml.java:49)
    [junit] 	at jmri.configurexml.AbstractXmlAdapter.store(AbstractXmlAdapter.java:83)
    [junit] 	at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:543)
    [junit] 	at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:523)
    [junit] 	at jmri.configurexml.ConfigXmlManager.addConfigStore(ConfigXmlManager.java:302)
    [junit] 	at jmri.configurexml.ConfigXmlManager.storeConfig(ConfigXmlManager.java:475)
    [junit] 	at jmri.implementation.JmriConfigurationManager.storeConfig(JmriConfigurationManager.java:147)
    [junit] 	at jmri.configurexml.LoadAndStoreTestBase.loadLoadStoreFileCheck(LoadAndStoreTestBase.java:171)
    [junit] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
    [junit] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[... 15 lines cut ...]
    [junit] 	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
    [junit] 	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
    [junit] 	at org.junit.runners.Suite.runChild(Suite.java:128)
    [junit] 	at org.junit.runners.Suite.runChild(Suite.java:27)
    [junit] 	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
    [junit] 	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
    [junit] 	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
    [junit] 	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
    [junit] 	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
    [junit] 	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
    [junit] 	at junit.framework.JUnit4TestAdapter.run(JUnit4TestAdapter.java:38)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at org.apache.tools.ant.taskdefs.optional.junit.JUnitTestRunner.run(JUnitTestRunner.java:535)
    [junit] 	at org.apache.tools.ant.taskdefs.optional.junit.JUnitTestRunner.launch(JUnitTestRunner.java:1182)
    [junit] 	at org.apache.tools.ant.taskdefs.optional.junit.JUnitTestRunner.main(JUnitTestRunner.java:1033)
    [junit] ERROR - match failed in testLoadStoreCurrent line 17 [main] jmri.configurexml.LoadAndStoreTest.?()
    [junit] ERROR -    inLine = "  <warrants class="jmri.jmrit.logix.configurexml.WarrantManagerXml">" [main] jmri.configurexml.LoadAndStoreTest.?()
    [junit] ERROR -   outLine = "  <signalmastlogics class="jmri.managers.configurexml.DefaultSignalMastLogicManagerXml">" [main] jmri.configurexml.LoadAndStoreTest.?()
    [junit] ERROR -      comparing "oblock.xml" and "oblock.xml" [main] jmri.configurexml.LoadAndStoreTest.?()
```

petecressman, 2017-03-08:
The 'anOBlock' xml is inappropriate for a load and store test in a headless environment.  (As are most of the  xml files in test...logix/configurexml/valid/).  

The loadNXParams() and storeNXParams() are convenience methods to restore some user entries between 'displayed' sessions.  Should a user load and store an xml file in a headless environment and then load it at the next displayed session, he/she would have the defaults for these settings, not what might have been there before.  Recently, persistence of several of these settings is now made only in Preferences.  Thus the use of  loadNXParams() and storeNXParams() to provide this convenience per xml file is less of a feature.  

I don't consider this (#3201) a defect, but if anyone thinks otherwise, it is OK by me to remove both loadNXParams() and storeNXParams() and have the user accept default entries after all loads.

Now that #3199 has fixed #3188, I will write a test to verify #3199 in any environment.

petecressman, 2017-03-18:
As of release 4.7.2 NXFrame no longer stores any persistent members.

closed 2017-07-03 by bobjacobsen