JMRI/JMRI #3218: import in PreferencesExamples.py is wrong

rhwood, 2017-03-10:
The import statement in PreferencesExamples.py is wrong.

There are two possible fixes:

1. Change ```import preferences``` to ```from preferences import *```

2. Change all uses of  ```getPreferences(...)``` to ```preferences.getPreferences(...)```

Which is preferable?

rhwood, 2017-03-11:
#3219 is an implementation of possible fix 2.

geowar1, 2017-03-11:
I would change the import; It's a single change per file (that uses preferences) vs. having to find and change every line of code that uses getPreferences().

millerdl, 2017-03-11:
In Python `from library import *` is considered poor form because it pollutes the namespace with all names from `library` and can result in name clashes.  The preferred approach is `import library` and calling functions as `library.function()`.  If the library name is long it can be shortened by using the form `import library as lib` and then calling as `lib.function()`. Also, if only one or a few classes/methods from `library` are required, an alternate form is `from library import function1` and then calling as `function1()`. These are all really only style issues so any of the approaches are acceptable and I see you have changed the calls so it follows the preferred approach.  (Reference https://www.python.org/dev/peps/pep-0008/#imports for Python style).  Of course we don't fully follow that style guide since we are interacting with Java and it makes sense to follow Java conventions for a lot of things.

rhwood, 2017-03-11:
@millerdl Thank for the guidance.