JMRI/JMRI #3245: where is "jmri.jmrix.loconet.locobuffer.LocoBufferStatsFrame()"

KenC57, 2017-03-15:
It was found by a user that the script SampleLnStats.py does not work. It is referencing a class that doesn't exist. I found a near match, jmri.jmrix.loconet.locostats.LocoStatsPanel() which says it was a move from the old name. But there seems to have been a number of changes beyond just a name change. Here is the trace from trying to run the script:

2017-03-15 08:35:07,299 automat.AbstractAutomaton             WARN  - Exception ends AbstractAutomaton thread: Traceback (most recent call last):
  File "<script>", line 23, in init
AttributeError: 'javapackage' object has no attribute 'LocoBufferStatsFrame'
 [Sample LocoNet Statistics]
Traceback (most recent call last):
  File "<script>", line 23, in init
AttributeError: 'javapackage' object has no attribute 'LocoBufferStatsFrame'

	at org.python.core.Py.AttributeError(Py.java:205)
	at org.python.core.PyObject.noAttributeError(PyObject.java:1013)
	at org.python.core.PyObject.__getattr__(PyObject.java:1008)
	at org.python.pycode._pyx1.init$2(<script>:25)
	at org.python.pycode._pyx1.call_function(<script>)
	at org.python.core.PyTableCode.call(PyTableCode.java:167)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:307)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:198)
	at org.python.core.PyFunction.__call__(PyFunction.java:482)
	at org.python.core.PyMethod.instancemethod___call__(PyMethod.java:237)
	at org.python.core.PyMethod.__call__(PyMethod.java:228)
	at org.python.core.PyMethod.__call__(PyMethod.java:218)
	at org.python.core.PyMethod.__call__(PyMethod.java:213)
	at org.python.core.PyObject._jcallexc(PyObject.java:3626)
	at org.python.core.PyObject._jcall(PyObject.java:3658)
	at org.python.proxies.__builtin__$SampleLnStats$1.init(Unknown Source)
	at jmri.jmrit.automat.AbstractAutomaton.run(AbstractAutomaton.java:140)
	at java.lang.Thread.run(Thread.java:745)
2017-03-15 08:41:18,396 jython.RunJythonScript                INFO  - No file selected [AWT-EventQueue-0]

pabender, 2017-03-15:
Sent from my iPad

> On Mar 15, 2017, at 8:42 AM, Ken Cameron <notifications@github.com> wrote:
> 
> It was found by a user that the script SampleLnStats.py does not work. It is referencing a class that doesn't exist. I found a near match, jmri.jmrix.loconet.locostats.LocoStatsPanel() which says it was a move from the old name. But there seems to have been a number of changes beyond just a name change. Here is the trace from trying to run the script:
> 
Since this is now a panel, you have to add it to a frame to be able to display it.

Paul

KenC57, 2017-03-15:
For fun I tried adding getter/setter for breaks and errors to the LocoStatsPanel. But it is expecting a memo object for the connection. It looks like LnPanel is expecting that via the initComponents, but that doesn't seem to happen. So memo is null and it fails while trying to build the refresh request msg. The method of getting a memo seems to be the same as a number of other systems. Comment is about

'make "memo" object available as convenience'

Anyone have clues here??

bobjacobsen, 2017-03-15:
The following is from my jmriusers post on this (do we have to discuss everything in three different places?)

> *) JMRI is moving away from having a JPanel, JFrame and Action class for everything it needs to display. Instead, things are now just being created as descendants of JmriPanel, which can then be created & loaded into windows by name.  If you want to show a jmri.jmrix.loconet.locostats.LocoStatsPanel, do (haven’t checked Python syntax, this is a quick copy&paste from Java)
>
> jmri.util.swing.JmriNamedPaneAction(“Loco Stats", "jmri.jmrix.loconet.locostats.LocoStatsPanel”).actionPerformed(None)
>
> (note the quotes around the jmri.jmrix.loconet.locostats.LocoStatsPanel class name; it really is a String). That creates an AbstractAction object, followed by immediately invoking it. (Those usually sit in a button, menu item, etc waiting to be invoked, but here we do it right away)
>
> For more on this, see the page in the developer section:  http://jmri.org/help/en/html/doc/Technical/Swing.shtml
>
> *) You can add tests of scripts to our usual test runs. There aren’t a lot of them. It would be nice to see more.  The PyUnit syntax is similar to JUnit and pretty self-explanatory:
> 
> http://jmri.org/jython/test/PowerSensorTest.py
>
> There’s more in, yes, you guessed it, a developer page:  http://localhost/help/en/html/doc/Technical/JUnit.shtml#testScriptCode

bobjacobsen, 2017-03-15:
For this specific case (using some tools that jmriusers might not get), here's how I'd debug this:
```
% grep -r LocoStatsPanel java/src/jmri/jmrix
```
finds a line in java/src/jmri/jmrix/loconet/swing/LocoNetMenu.java. Aha! That's where the LocoNet code is constructing it's menu.  Reading that, I see that `LnNamedPaneAction` is the LocoNet-specific class that handles initialization for LocoNet panels.  Then look at the JavaDoc and try
```
memo = jmri.InstanceManager.getDefault(jmri.jmrix.loconet.LocoNetSystemConnectionMemo)
jmri.jmrix.loconet.LnNamedPaneAction(“Loco Stats", jmri.util.swing.sdi.JmriJFrameInterface(), memo, "jmri.jmrix.loconet.locostats.LocoStatsPanel”).actionPerformed(None)
```
(the 2nd argument is because this doesn't have a convenience constructor without it, so we have to ask specifically for a new window)

Haven't tried it, but that's where I'd start.

KenC57, 2017-03-15:
Looking at the script, it seems it wants to create the pane, but not display
it. Then it uses the pane just because it knows how to get updates of the
buffer status.

 

I've been able to change things around a bit. Adding getter/setter pairs to
LocoStatsPanel() was the first step. Then having the script get the memo
(only works for the default connection) and pass it to LocoStatsPanel as an
argument during its instantiation (another change). 

 

At this point, it seems to be working for most purposes. But still a few
problems. Like when it tries to setVisible, I get this dump pointing back to
LocoStatsPanel. My dirty idea would be adding some flag/option to turn off
those lines. I'm guessing the setVisible fails because this isn't opened
inside the right type of frame, well, no frame.

-ken c

rhwood, 2017-03-15:
Those statistical getters should be in a truely headless object, not in a panel or frame.

KenC57, 2017-03-16:
I am suspecting that is the root of the problem. The LocoStatsPanel is both
the logic and the display. So if I was feeling up to it, splitting it into a
logic part and a display part would be the right action. Not sure if I feel
up for that right now.

-ken c

bobjacobsen, 2017-03-16:
This actually works:
```
import jmri
memo = jmri.InstanceManager.getDefault(jmri.jmrix.loconet.LocoNetSystemConnectionMemo)
jmri.jmrix.loconet.swing.LnNamedPaneAction("Loco Stats", jmri.util.swing.sdi.JmriJFrameInterface(), "jmri.jmrix.loconet.locostats.LocoStatsPanel", memo).actionPerformed(None)
```

(I had the arguments in the wrong order in the prior post)

KenC57, 2017-03-16:
Bob,

 

While that does launch the monitor, and that in turn covers the issues with
the setVisible, it doesn't return a reference to the monitor created. The
script needs that to talk to it to get values and refresh them. So launching
that way, how do I ask for that instance??

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

devel-bobm, 2017-03-16:
I will take up the task of dis-entangling the functionality from the gui.

pabender, 2017-03-16:
Ken,

Sent from my iPhone
> On Mar 16, 2017, at 10:23 AM, Ken Cameron <notifications@github.com> wrote:
> While that does launch the monitor, and that in turn covers the issues with
> the setVisible, it doesn't return a reference to the monitor created. The
> script needs that to talk to it to get values and refresh them. So launching
> that way, how do I ask for that instance??

If the script is just interested in the messages, then using the monitor for this is probably the wrong approach.

The script should really register as a listener and translate the messages itself.  

I don't know if the loconet messages have this, but many message types have a toMonitorString method to do the translation.  If the Loconet monitor is doing the translation, it might be a good idea to move that translation to a toMonitorString method in the LocoNet message class.

Paul

KenC57, 2017-03-16:
Group,

 

The monitor is doing a poll via a specific command and then parsing the
return, which varies depending on the hardware attached. It seems they all
use the same command, but the reply has different data depending on the
interface device. So for some interfaces, this script doesn't get everything
it would like. Those columns of course never change. It also talks to the
traffic controller to get byte and message count data.

 

This does bring up the general issue that a number of our scripts might not
work because of changes in the codebase. Looking at the history, I'm
thinking this might have stopped working quite some time ago.

 

I'm going to put up a branch so you can see what I've hacked into the code
so far.

-ken c

KenC57, 2017-03-16:
Posted pull #3249 as working but poor example of a fix. Just so others can review what's up.

devel-bobm, 2017-04-05:
Pull request #3335 attempts to dis-entangle the GUI from the LocoNet functionality.  Includes a very simple jython script showing how to use it.  Is marked Work-in-progress pending further testing.

bobjacobsen, 2017-04-09:
I think the script in #3335 resolves this. Can we close it?

closed 2017-06-05 by bobjacobsen