JMRI/JMRI #3248: Southern Railway Signal Definitions

gpmccartney, 2017-03-16:
[SOU-1981-xml.zip](https://github.com/JMRI/JMRI/files/848703/SOU-1981-xml.zip)

KenC57, 2017-03-16:
Looks like one folder of the xml parts and the other is the icon part, correct??

gpmccartney, 2017-03-16:
Yes Ken that is correct.   I am not sure how to zip those up to get them in the right place for submission.  I added xml to the folder name to distinguish the two.

Thanks
Greg

From: Ken Cameron
Sent: Thursday, March 16, 2017 3:13 PM
To: JMRI/JMRI
Cc: gpmccartney; Author
Subject: Re: [JMRI/JMRI] Southern Railway Signal Definitions (#3248)

Looks like one folder of the xml parts and the other is the icon part, correct??

KenC57, 2017-03-18:
These should now be in the daily builds and be part of 4.7.3 next month.

closed 2017-03-18 by KenC57