JMRI/JMRI #3251: Copying Logix with 4.7.2  "Uncaught Exception"

Walter71342JMRI, 2017-03-17:
Had been working OK, Had copied other logix's with this version with out a problem.
I was trying to copy IX:AUTO:0018	with user name of #234 BLK09. AT the bottom is "Zip" of the panel file.	

Here is the System Console:
2017-03-16 23:00:49,761 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-03-16 23:00:49,768 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Walt\JMRI\log\messages.log [main]
2017-03-16 23:00:49,769 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Walt\JMRI\log\session.log [main]
2017-03-16 23:00:49,772 apps.Apps                             INFO  - PanelPro version 4.7.2-R741b21f starts under Java 1.8.0_121 on Windows 10 x86 v10.0 at Thu Mar 16 23:00:49 CDT 2017 [main]
2017-03-16 23:00:52,204 apps.Apps                             INFO  - Starting with profile HOn3.3ec971a3 [main]
2017-03-16 23:00:52,266 node.NodeIdentity                     INFO  - Using jmri-64006A824CD2-3ec971a3 as the JMRI Node identity [main]
2017-03-16 23:00:52,500 locobuffer.LocoBufferAdapter          INFO  - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true  CD: false [main]
2017-03-16 23:00:53,323 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0_121 [main]
2017-03-16 23:00:53,324 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.0 [main]
2017-03-16 23:00:53,343 logix.WarrantPreferences              INFO  - Found Warrant preferences file: C:\Users\Walt\Dropbox\jmri\HOn3\signal\WarrantPreferences.xml [main]
2017-03-16 23:00:53,457 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2017-03-16 23:00:53,457 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Walt\Dropbox\jmri\HOn3\ [main]
2017-03-16 23:00:53,457 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Walt\JMRI\HOn3\ [main]
2017-03-16 23:00:53,457 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Walt\JMRI\ [main]
2017-03-16 23:00:53,458 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Walt\ [main]
2017-03-16 23:00:53,458 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Walt\Dropbox\jmri\py scripts\ [main]
2017-03-16 23:00:53,562 util.prefs                            WARN  - Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5. [init prefs]
2017-03-16 23:00:59,890 igurexml.DefaultConditionalManagerXml WARN  - No state variables found for conditional IX:AUTO:0025C1 [AWT-EventQueue-0]
2017-03-16 23:01:00,253 layoutEditor.LayoutEditor             INFO  - Frame size: {w:1900, h:960} [AWT-EventQueue-0]
2017-03-16 23:01:00,654 layoutEditor.LayoutEditor             INFO  - Frame size: {w:1920, h:105} [AWT-EventQueue-0]
2017-03-16 23:01:00,658 layoutEditor.LayoutEditor             INFO  - Frame size: {w:1920, h:105} [AWT-EventQueue-0]
2017-03-16 23:01:00,658 layoutEditor.LayoutEditor             INFO  - Frame size: {w:1920, h:105} [AWT-EventQueue-0]
2017-03-16 23:01:02,258 layoutEditor.LayoutEditor             INFO  - Font size: 11.0 [AWT-EventQueue-0]
2017-03-16 23:01:02,272 layoutEditor.LayoutEditor             INFO  - jmri.jmrit.display.layoutEditor.LayoutEditor:B.toolBarThin is bottom [AWT-EventQueue-0]
2017-03-16 23:01:02,273 layoutEditor.LayoutEditor             INFO  - prefsMgr.prefsWindowLocation(jmri.jmrit.display.layoutEditor.LayoutEditor:B) is java.awt.Point[x=364,y=39] [AWT-EventQueue-0]
2017-03-16 23:01:02,273 layoutEditor.LayoutEditor             INFO  - prefsMgr.prefsWindowSize is(jmri.jmrit.display.layoutEditor.LayoutEditor:B) java.awt.Dimension[width=923,height=714] [AWT-EventQueue-0]
2017-03-16 23:01:02,273 layoutEditor.LayoutEditor             INFO  - prefsMgr.getProperty(jmri.jmrit.display.layoutEditor.LayoutEditor:B, "windowRectangle2D") is null [AWT-EventQueue-0]
2017-03-16 23:01:02,274 layoutEditor.LayoutEditor             INFO  - Frame size: {w:1920, h:105} [AWT-EventQueue-0]
2017-03-16 23:01:15,610 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [Listed Table Generation]
2017-03-16 23:01:47,221 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrit.beantable.LogixTableAction.copyConditionalToLogix(LogixTableAction.java:948)
	at jmri.jmrit.beantable.LogixTableAction.copyLogixPressed(LogixTableAction.java:922)
	at jmri.jmrit.beantable.LogixTableAction$17$1.actionPerformed(LogixTableAction.java:841)
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

This is the panel file:

[NarrowGaugeNew.zip](https://github.com/JMRI/JMRI/files/849636/NarrowGaugeNew.zip)

Walt

Walter71342JMRI, 2017-03-17:
Now its working again. I think i was doing something wrong. mayb not changing the user name of the sub Logix's.

But sure seems like a trap I created for my self.

Walt

Walter71342JMRI, 2017-03-18:
The panel for changing the name if the Sub logixs was under another screen, and I had not entered any info into it
I can reproduce this if needed.

geowar1, 2017-11-01:
Please assign to author (Walter71342JMRI) to regress and close if fixed or assign to Dave Sand (did the last Logix rewrite) to investigate.

closed 2017-11-08 by Walter71342JMRI