JMRI/JMRI #3258: DCC4PC fails node discovery

KenC57, 2017-03-18:
User michael.mills1503@hotmail.co.uk reports:
Very disappointing that we still have this problem after all this time
First card ok
Second card generates approx 600 sensor entries
Third card is not detected at all
If there is any testing etc that I can do then please please let me know.
-----
I've attached a zip of submitted log files.
[Dcc4pc.zip](https://github.com/JMRI/JMRI/files/852423/Dcc4pc.zip)

bobjacobsen, 2017-07-05:
I took a look at this and haven't been able to figure out what the code does, nor what's supposed to happen on the line.  

Log (with detailed info) of a successful start up with one card:  
[session-dcc4pc-1-card.log.txt](https://github.com/JMRI/JMRI/files/1125699/session-dcc4pc-1-card.log.txt)

Log of startup with three cards, didn't go well: 
[session-dcc4pc-3-cards.log.txt](https://github.com/JMRI/JMRI/files/1125700/session-dcc4pc-3-cards.log.txt)

The documentation, such as it is, is here: http://www.dcc4pc.co.uk/downloads.html

bobjacobsen, 2017-07-05:
Some error information, configuration, logs, etc
https://sourceforge.net/p/jmri/mailman/message/35899868/
https://sourceforge.net/p/jmri/mailman/message/35873747/
https://sourceforge.net/p/jmri/mailman/message/35671915/
https://sourceforge.net/p/jmri/mailman/message/35671928/
https://sourceforge.net/p/jmri/mailman/search/?q=dcc4pc&limit=25&page=0&sort=posted_date%20desc

closed 2017-07-20 by kevin-dickerson

rhwood, 2017-07-20:
#3797 is presumed to fix this.