JMRI/JMRI #3278: xml file start for Bachmann N scale EM1 (Made by Lenz, not yet supported by JMRI)

kmcsjr, 2017-03-26:
[Bachmann_Lenz_EM1.zip](https://github.com/JMRI/JMRI/files/870496/Bachmann_Lenz_EM1.zip)

My thanks to Peter Ulvestad. He found the manual, for this decoder. I took the Lenz_Silver_Mini file and made the changes I could. Where I could find the CV, in the file, I made the appropriate changes. When CVs were addressed, by referencing other files, I checked to see if the CV settings from the Bmann and the Mini versions were identical. I was able to use the file and adjust what was there, so my loco runs. I summarized the remaining issues in the xml file. I will also list them here. I am adding clarification here, as well.

To clarify The Lenz Mini manual only lists Function Outputs A and B, the Bmann version has an Output C. I think it's lighting, and..... this thing has a front and rear N scale light, so... I'm sure it's cool, it's beyond me.

  
To clarify, the CVs are listed in the silver mini manual, but I couldn't find a reference, in the xml file. The default is 255, when I look at my CVs in DecoderPro.

  
I was able to set CV 57, but that won't help someone trusting the file.

  
Beyond my abilities. At least, for a while.
  
I think the value populated properly, when I hose motor 5, but I'd appreciate more experienced eyes.

I'm happy to work on this, but I'd take any help, to get this one done quickly and I'll try to contribute more.

dheap, 2017-03-26:
GitHub uses Markdown for styling written comments. There is a small button below the Comment box that brings up Markdown help. There are also easy buttons in the bar above the Comment box to insert Markdown.

For example, to include code, highlight the pasted code block and press the "<>" icon. The result will look like this:
```
  <decoder>
    <family name="NMRA standard CV definitions" mfg="NMRA">
      <model model="NMRA standard CV definitions"/>
    </family>
    <programming direct="yes" paged="yes" register="yes" ops="yes"/>
    <variables>
      <xi:include href="http://jmri.org/xml/decoders/nmra/shortAndLongAddress.xml"/>
      <xi:include href="http://jmri.org/xml/decoders/nmra/accelDecel_255.xml"/>
      <xi:include href="http://jmri.org/xml/decoders/nmra/vStartHighMid.xml"/>
      <xi:include href="http://jmri.org/xml/decoders/nmra/mfgVersionId.xml"/>
    <variable item="Total PWM Period" CV="9">
```

You can also edit your comment after posting, but that won't trigger email updates to those of us who have set up email alerts.

kmcsjr, 2017-03-28:
This is the Bachmann decoder Manual

http://www.bachmanntrains.com/home-usa/dwg/dwgs/NEM-1LenzDecoderCVTable.pdf

From my notes above, I don't think it will take much to get it to a final state. I'll try to poke at it, this weekend.

closed 2021-04-14 by stale[bot]