JMRI/JMRI #3286: Discussion of Write-Verify-Read and Cache-Index-CV-Write

dheap, 2017-03-27:
I've reviewed the code to date; #3280 and #3282.

In Preferences->Roster->Programmer we see two new preferences:
- Allow caching when writing/reading index CVs
- When possible, confirm index CV writes

The term "index CV" is generally understood to refer to CVs 31 & 32 in the NMRA S9.2.2 scheme (or equivalents in other schemes). Apart for the fact that we almost never _read_ index CVs, the above two actions seem not unreasonable, and _may_ offer benefits.

In MultiIndexProgrammerFacade we have this documentation (excerpt):

 cvFirst is false:
- 11.12.123 Writes 11 to the first index CV, then 12 to the second index CV, then does write/read/confirm to 123

Note that this mentions no provision for write/read/confirmation of either index CV. It only mentions write/read/confirmation of "123", which is what we generally refer to as the _indexed_  CV. (Indexed CVs are 257 to 512 in the NMRA S9.2.2 scheme or equivalents in other schemes).

There seems to be a conflict between the apparent intentions of the Preferences
 and the MultiIndexProgrammerFacade. I am aware that the infrastructure is not yet fully in place for these changes.

I will post this now for information only, but please don't jump in with responses and changes at this stage. Please wait until I have posted the rest of my comments and suggestions.

bobjacobsen, 2017-03-27:
See also #3285

bobjacobsen, 2017-03-27:
I understand that you're not done, but I want to correct a clear misunderstanding: "write/read/confirm" in the MultiIndexProgrammerFacade Javadoc refers to the fact that the Programmer interface allows operations `write(..)`, `read(..)` and `confirm(..)`. It does _not_ refer to a verification read after write, which is not done in that class. For that, see [VerifyWriteProgrammerFacade](http://jmri.sourceforge.net/JavaDoc/doc/jmri/implementation/VerifyWriteProgrammerFacade.html).

dheap, 2017-03-27:
@bobjacobsen I was just quoting your JavaDoc verbatim. I was fully aware of [VerifyWriteProgrammerFacade](http://jmri.sourceforge.net/JavaDoc/doc/jmri/implementation/VerifyWriteProgrammerFacade.html) and its role in a verification read after write.

I am relieved to hear that the statement was not an intention to omit verifying (with [VerifyWriteProgrammerFacade](http://jmri.sourceforge.net/JavaDoc/doc/jmri/implementation/VerifyWriteProgrammerFacade.html)) the values written to the Index CVs by the MultiIndexProgrammerFacade.

Could we please leave it at that until I have made my comments and suggestions?

bobjacobsen, 2017-03-27:
@dheap That's pretty weak. What you said was this:

> Note that this mentions no provision for write/read/confirmation of either index CV. It only mentions write/read/confirmation of "123", which is what we generally refer to as the indexed CV. (Indexed CVs are 257 to 512 in the NMRA S9.2.2 scheme or equivalents in other schemes).

That makes no sense as all if all you're doing is  "I was just quoting your JavaDoc verbatim".

Enough bike-shedding. I've unsubscribed from this Issue.  If you want to add more capability in this area, perhaps you or somebody else will write it.

dheap, 2017-03-27:
@bobjacobsen I am quite saddened by your response to my request, but will immediately cease attempts to discuss this topic.

closed 2017-03-27 by dheap