JMRI/JMRI #3289: Long delay at end of 'ant ci-test' while stopping jMDNS

bobjacobsen, 2017-03-27:
When running 'ant ci-test', I get a long (several minute) delay at the end. During that time, the `main` thread is doing jMDNS termination:

```
  [junit] "main" #1 prio=5 os_prio=31 tid=0x00007fb2d4003000 nid=0x1003 waiting on condition [0x0000700000182000]
    [junit]    java.lang.Thread.State: TIMED_WAITING (parking)
    [junit] 	at sun.misc.Unsafe.park(Native Method)
    [junit] 	- parking to wait for  <0x000000079700b718> (a java.util.concurrent.Semaphore$FairSync)
    [junit] 	at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
    [junit] 	at java.util.concurrent.locks.AbstractQueuedSynchronizer.doAcquireSharedNanos(AbstractQueuedSynchronizer.java:1037)
    [junit] 	at java.util.concurrent.locks.AbstractQueuedSynchronizer.tryAcquireSharedNanos(AbstractQueuedSynchronizer.java:1328)
    [junit] 	at java.util.concurrent.Semaphore.tryAcquire(Semaphore.java:409)
    [junit] 	at javax.jmdns.impl.DNSStatefulObject$DNSStatefulObjectSemaphore.waitForEvent(DNSStatefulObject.java:66)
    [junit] 	at javax.jmdns.impl.DNSStatefulObject$DefaultImplementation.waitForCanceled(DNSStatefulObject.java:404)
    [junit] 	at javax.jmdns.impl.ServiceInfoImpl.waitForCanceled(ServiceInfoImpl.java:1110)
    [junit] 	at javax.jmdns.impl.JmDNSImpl.unregisterService(JmDNSImpl.java:1045)
    [junit] 	at jmri.util.zeroconf.ZeroConfService.lambda$stop$469(ZeroConfService.java:323)
    [junit] 	at jmri.util.zeroconf.ZeroConfService$$Lambda$622/494850799.accept(Unknown Source)
    [junit] 	at java.util.HashMap$ValueSpliterator.forEachRemaining(HashMap.java:1612)
    [junit] 	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
    [junit] 	at jmri.util.zeroconf.ZeroConfService.stop(ZeroConfService.java:319)
    [junit] 	at jmri.util.zeroconf.ZeroConfServiceTest.testStop(ZeroConfServiceTest.java:165)
    [junit] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
    [junit] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
    [junit] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
    [junit] 	at java.lang.reflect.Method.invoke(Method.java:483)
    [junit] 	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
    [junit] 	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
    [junit] 	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
    [junit] 	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:33)
    [junit] 	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
    [junit] 	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
    [junit] 	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
    [junit] 	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
    [junit] 	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
    [junit] 	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
```

This seems to descend from ` jmri.util.zeroconf.ZeroConfServiceTest`.

There's a little bit of work going on, as the traceback changes occasionally:
```
    [junit] "main" #1 prio=5 os_prio=31 tid=0x00007fb2d4003000 nid=0x1003 waiting on condition [0x0000700000181000]
    [junit]    java.lang.Thread.State: TIMED_WAITING (parking)
    [junit] 	at sun.misc.Unsafe.park(Native Method)
    [junit] 	- parking to wait for  <0x000000079a884818> (a java.util.concurrent.Semaphore$FairSync)
    [junit] 	at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
    [junit] 	at java.util.concurrent.locks.AbstractQueuedSynchronizer.doAcquireSharedNanos(AbstractQueuedSynchronizer.java:1037)
    [junit] 	at java.util.concurrent.locks.AbstractQueuedSynchronizer.tryAcquireSharedNanos(AbstractQueuedSynchronizer.java:1328)
    [junit] 	at java.util.concurrent.Semaphore.tryAcquire(Semaphore.java:409)
    [junit] 	at javax.jmdns.impl.DNSStatefulObject$DNSStatefulObjectSemaphore.waitForEvent(DNSStatefulObject.java:66)
    [junit] 	at javax.jmdns.impl.DNSStatefulObject$DefaultImplementation.waitForCanceled(DNSStatefulObject.java:404)
    [junit] 	at javax.jmdns.impl.ServiceInfoImpl.waitForCanceled(ServiceInfoImpl.java:1110)
    [junit] 	at javax.jmdns.impl.JmDNSImpl.unregisterService(JmDNSImpl.java:1045)
    [junit] 	at jmri.util.zeroconf.ZeroConfService.lambda$stop$469(ZeroConfService.java:323)
```

```
    [junit] "main" #1 prio=5 os_prio=31 tid=0x00007fb2d4003000 nid=0x1003 waiting on condition [0x0000700000181000]
    [junit]    java.lang.Thread.State: TIMED_WAITING (parking)
    [junit] 	at sun.misc.Unsafe.park(Native Method)
    [junit] 	- parking to wait for  <0x0000000799d78b98> (a java.util.concurrent.Semaphore$FairSync)
    [junit] 	at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
    [junit] 	at java.util.concurrent.locks.AbstractQueuedSynchronizer.doAcquireSharedNanos(AbstractQueuedSynchronizer.java:1037)
    [junit] 	at java.util.concurrent.locks.AbstractQueuedSynchronizer.tryAcquireSharedNanos(AbstractQueuedSynchronizer.java:1328)
    [junit] 	at java.util.concurrent.Semaphore.tryAcquire(Semaphore.java:409)
    [junit] 	at javax.jmdns.impl.DNSStatefulObject$DNSStatefulObjectSemaphore.waitForEvent(DNSStatefulObject.java:66)
    [junit] 	at javax.jmdns.impl.DNSStatefulObject$DefaultImplementation.waitForCanceled(DNSStatefulObject.java:404)
    [junit] 	at javax.jmdns.impl.ServiceInfoImpl.waitForCanceled(ServiceInfoImpl.java:1110)
    [junit] 	at javax.jmdns.impl.JmDNSImpl.unregisterService(JmDNSImpl.java:1045)
    [junit] 	at jmri.util.zeroconf.ZeroConfService.lambda$stop$469(ZeroConfService.java:323)
[... 35 lines cut ...]
    [junit] 	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:233)
    [junit] 	at java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:418)
    [junit] 	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:583)
    [junit] 	at jmri.util.zeroconf.ZeroConfService.stopAll(ZeroConfService.java:349)
    [junit] 	at jmri.util.zeroconf.ZeroConfService.stopAll(ZeroConfService.java:343)
    [junit] 	at jmri.server.json.util.JsonUtilHttpServiceTest.tearDown(JsonUtilHttpServiceTest.java:58)
    [junit] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
    [junit] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
    [junit] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
    [junit] 	at java.lang.reflect.Method.invoke(Method.java:483)
    [junit] 	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
    [junit] 	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
    [junit] 	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
    [junit] 	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:33)
    [junit] 	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
    [junit] 	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
    [junit] 	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
    [junit] 	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
    [junit] 	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
    [junit] 	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
 ```

That's running from `jmri.server.json.util.JsonUtilHttpServiceTest.tearDown`, but seems to have the same "take a real long time" behavior.

Anybody know how to get jMDNS to quit faster?

bobjacobsen, 2017-03-27:
A little targeted timing:
```
% time ./runtest.csh jmri.util.zeroconf.ZeroConfServiceTest
JUnit version 4.12
.Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
............
Time: 172.564
OK (13 tests)
5.032u 1.741s 2:53.83 3.8%	0+0k 80+28io 6793pf+0w
```

```
% time ./runtest.csh jmri.server.json.util.JsonUtilHttpServiceTest
JUnit version 4.12
Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
...........
Time: 129.753
OK (11 tests)
4.444u 1.092s 2:10.82 4.2%	0+0k 3+16io 1pf+0w
```
Just those two those two tests are taking a total of five minutes.

rhwood, 2017-03-27:
The only was to make jMDNS to quit faster is to mock it out of the equation. jMDNS needs to advise all systems in the local network that the service it is advertising is no longer available, and waits to handle any responses (iirc the RFC allows this to take awhile).

bobjacobsen, 2017-03-27:
Except that stopping jMDNS under e.g. PanelPro is much faster:
```
     [java] 2017-03-27 09:16:04,586 server.WebServer                      INFO  - Starting Web Server on port 12080 [WebServer]
     [java] 2017-03-27 09:16:04,705 server.WebServer                      INFO  - Starting ZeroConfService _http._tcp.local for Web Server with properties {path=/, json=4.0} [WebServer]

(Select Quit around 9:16:13)

     [java] 2017-03-27 09:16:13,250 server.WebServer                      INFO  - Stopping Web Server [Thread-37]
     [java] 2017-03-27 09:16:37,247 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
```

Is the test perhaps doing a bunch of starts and stops?

bobjacobsen, 2017-03-27:
The `unregisterService(..)` [javadoc](http://jmdns.sourceforge.net/apidocs/javax/jmdns/JmDNS.html#unregisterService(javax.jmdns.ServiceInfo)) talks about 
```
Unregistered services will not disappear form the list of services immediately. According to the specification, when unregistering services we send goodbye packets and then wait 1s before purging the cache.
This is support for shared records that can be rescued by some other cooperation DNS.
```

rhwood, 2017-03-27:
The shutdown handler parallelizes a number of things to speed up the shutdown (it used to appear to hang on macOS), so PanelPro is faster; we had problems with tests being very unreliable if we did not deliberately wait for jMDNS to complete one task before moving into the next; the tests would find themselves in inconsistent states if they did not wait.

pabender, 2017-03-27:
Sent from my iPad

> On Mar 27, 2017, at 12:14 PM, Randall Wood <notifications@github.com> wrote:
> The only was to make jMDNS to quit faster is to mock it out of the equation. jMDNS needs to advise all systems in the local network that the service it is advertising is no longer available, and waits to handle any responses (iirc the RFC allows this to take awhile).

We probably should mock jMDNS.  

It really doesn't make sense for us to produce use an actual jMDNS instance for any of the tests.

I can work on that tonight.  I don't think it will be overly difficult to get it up and running.

Paul

pabender, 2017-03-31:
So I've been looking at this off and on all week.

I keep getting the same 17 test errors on alltest no matter how I mock this.

I'm going to set this aside for a few days and see If what's happening is any clearer once I come back to it.

rhwood, 2018-12-09:
Has this concern been resolved? If not, I have a branch where `jmri.util.zeroconf.ZeroConfService` has been broken into a service object and a manager object and the manager mocked.

pabender, 2018-12-09:
@rhwood no, it hasn’t, so if you have something that fixes it, that would be good.

rhwood, 2018-12-10:
I have found the cause. #5138 installs a version of JmDNS known to *always* timeout on any closing action for 5 seconds (jmdns/jmdns#132).

pabender, 2018-12-11:
This issue actually pre-dates PR #5138.  Are you sure that is the cause? Was JmDNS 3.5.1 installed when this issue was first raised?

rhwood, 2018-12-12:
You're right, I was off by a year (I was looking at dates in GitHub and the current year is not shown). I've managed to figure out that this long pause is present in PanelPro (at least on macOS) as well, so its not unique to the CI tests.

On macOS, the windows all close rapidly, but the process does not exist for about 15 seconds, waiting on the JmDNS services to close. My profile is running a WiThrottle and Web Server. The WiThrottle is for some reason creating unique services per IP address (one IPv4 and one IPv6), while the Web Server is creating a single service across both IP addresses, and JmDNS itself is timing out after 5 seconds per service closing them.

closed 2021-04-14 by stale[bot]