JMRI/JMRI #3294: JMRI PanelPro - Programming Support for Märklin CS2 / Doehler&Haass Programmer

BR365, 2017-03-27:
Hello,
is there planned to include Programming Support for Märklin CS2 and the Doehler&Haass Programmer?

I tested JMRI PanelPro with my Banana Pi and the trackbox (Gleisbox) and this works fine! I can switch on/off the track current and also monitor the CAN commands.

But I can't use the Marklin-CS2 connection for programming:

[img]http://up.picr.de/28720630th.jpg[/img]

Thanks for any support and ideas,
Moritz

bobjacobsen, 2017-03-28:
(some highlights on Programming Support for Märklin CS2 from a JMRIusers discussion)

There's some documentation available in German:
CAN-Protokoll (PDF, 7 MB) - Märklin https://www.maerklin.de/fileadmin/media/produkte/CS2_can-protokoll_1-0.pdf 

CAN CS2 Protokoll - 2.0.doc 
https://www.maerklin.de/fileadmin/media/service/software-updates/cs2CAN-Protokoll-2_0.pdf 

A brief look shows info on CV programming, mixed in with lots of info about MFX and others.

BR365, 2017-03-28:
Thanks, Bob!

BR365, 2017-04-21:
Hello Bob, hello all,
are there any news about CS2 support of JMRI PanelPro and Doehler&Haass Programmer support?

Thanks,
Moritz

bobjacobsen, 2017-04-23:
CS2 is just a matter of finding time.  Or perhaps somebody else will step in.

So far, I haven't seen any documentation for D&H Programer.  But I might have missed a post somewhere. Is a reference available?

BR365, 2017-04-23:
Hi Bob,
thanks for the update.

I asked D&H for documentation and they answered, that the developer of JMRI should please contact them directly:

**Doehler & Haass GmbH & Co. KG**
Friedrich Rehak
http://www.doehler-haass.de   
doehler-haass@t-online.de

If necessary, I will talk to them again and explain, that I will be the contact person between D&H and JMRI. But You can contact them directly, too. I guess, that's what they prefer.

regards,
Moritz

closed 2021-05-30 by stale[bot]