JMRI/JMRI #3298: LayoutEditor Test Error

dsand47, 2017-03-27:
While testing my Logix changes, an error occurred in the LayoutEditor test.  I used a shorter test sequence to get the attached report.

Dave Sand

[test-error.txt](https://github.com/JMRI/JMRI/files/874003/test-error.txt)

bobjacobsen, 2017-03-27:
```
	at jmri.InstanceManager.getNullableDefault(InstanceManager.java:210)
	at jmri.InstanceManager.getOptionalDefault(InstanceManager.java:263)
	at jmri.jmrit.display.layoutEditor.LayoutEditor.lambda$new$6(LayoutEditor.java:1077)
```

That's a mystery!  Line 210 of the InstanceManager is dereferencing a List reference obtained from getList(..).  But getList(..) should never return a null reference (see line 125).

dsand47, 2017-03-28:
I was running another test sequence tonight and the LayoutEditor error did not occur.

closed 2017-05-07 by bobjacobsen