JMRI/JMRI #3300: Econami setup is missing F11 brake Function Mapping and Selections of Light types

mickmoignardOAC, 2017-03-28:
I thing this is a regression from 4.4, where these are working correctly. In 4.6 and 4.7.x, the issues are

- Function map pane does not include the F11 Brake. The CV pane does include the CVs for it, though (1.275 and 1.403)
- Lights pane does not include the light effect CVs for whatever lights are available in the relevant decoder version.  This is part of CV49 - 54. 
These are all OK on the TSU-2 decoders.

dheap, 2017-03-28:
I've tried New Loco with random Econami models in 4.7.1, 4.7.2 and current master on Mac. Can't see either problem. Please give some specific model examples and also check whether New Loco or just an existing roster entry.

The presence of the CVs in an existing roster entry would be expected if a correct definition was used at some time in the past.

I can check Windows when I get some more specifics.

dheap, 2017-03-28:
If an existing roster entry, please upload it here.

mickmoignardOAC, 2017-03-29:
I've seen this with both new loco and existing locos, ECO-100 and -200, and when using a UK econami for which there are no defs yet (separate issue).  Deffo seen it in 4.7.2 and a recent build of 4.7.3, and I think it was in 4.6 as well.  I have 4.4 on another computer where the issue doesn't arise. Will attach some roster entries shortly,

mickmoignardOAC, 2017-03-29:
All on Windows, btw.  Both computers using Windows 10, and seen it with Sprog and Loconet USB connections.

mickmoignardOAC, 2017-03-29:
Solved: When looking for the roster entries, I did a little extra digging on that computer and found some old pre-release Econami def files from way back.   Removed them and all is now well.

closed 2017-03-29 by mickmoignardOAC

mickmoignardOAC, 2017-03-29:
They weren't on the computer which is running DP 4.4, which is why that was fine!