JMRI/JMRI #3302: possible additions to JMRI web server

mstevetodd, 2017-03-28:
Here are some additional pages I'm thinking of adding to the JMRI web server:

- Panel
  - Upload Panel Files
- Roster
  - Upload Roster Files
  - Rebuild Roster
- Help
  - About
  - Locations
  - Context
  - System Console
- Loconet 
  - Monitor Loconet
  - Monitor Slots
  - Monitor Stats
- Tools
  - Config Profile Selector
  - Restart JMRI
  - Reboot Computer

I'd appreciate any feedback concerning these items, naming, grouping, and/or the coding thereof.
Note that I did not include any of the Preference/Config items, as I think @rhwood may be working on those.
BACKGROUND:
I recently installed the Falcon Pixel Player for RaspberryPi (https://github.com/FalconChristmas/fpp), and was very impressed that ALL the configuration could be done via the web interface. This included network settings, an ssh command line, etc. Very nice!

pabender, 2017-03-28:
I would tend to make system specific additions the lowest priority (The Loconet items you mention).  Those won't function on every machine that JMRI is installed on.

Cliff-Anderson, 2017-03-30:
For those users who know nothing about running PanelPro on a computer that does not have a monitor attached, the reason behind this list of other complexities not directly needed by most of the Ops Crew seems strange.  
It finally dawned that my use of the monitor is not consistent with the RaspberryPi described in the BACKGROUND paragraph.  

My first reaction to the @pabender comment above was to search out the source code for 
public class LocoMonPane extends jmri.jmrix.AbstractMonPane implements LocoNetListener, LnPanelInterface{}
and to try to understand if AbstractMonPane could do the same work, but having no real experience with Java, that got me entangled in stuff that seemed even worse.  

As an afterthought, and then rereading the list of pages several times, it now seems as if the intent is to provide some **safe** subset of the menu items that could be found if the PanelPro App GUI were simply made available.  Simple, but **potentially dangerous** to provide that access as a page?

mstevetodd, 2017-03-30:
@Cliff-Anderson , 
Yes, the target user of this enhancement is **not** using the traditional GUI desktop with monitor, keyboard and mouse. They're accessing JMRI functionality from a remote device, via a web browser. Unfortunately, there is no way to "simply" make a Java Swing desktop application (like JMRI) available as a web app. Which means that any functionality needed must be rewritten as a web app of some sort. 
Hence the need for discussion regarding which ones are needed, priority, security implications, etc.

mstevetodd, 2017-04-06:
Update:
I've added an About page, but did not add Locations, Context or System Console functions. Randall was concerned that, in the event that JMRI web server was visible on the Internet, showing the user name (part of the path of most installations) could be a security issue.
I have added an Upload capability to the existing Roster table web page. The roster is updated as entries are added/updated, so I no longer see the need for the Rebuild Roster option.

mstevetodd, 2017-04-19:
Working on a /tables page, for displaying the contents of various JMRI tables (turnouts, sensors, memories, etc.) While open, these pages will update to reflect changes, adds and removals for the tables in JMRI. Complete functionality will need to wait for issue #3411

mstevetodd, 2017-04-20:
@rhwood has suggested using the name "Layout Elements" or "Layout" instead of "Tables". So path would now be /layout/turnouts, for example.

danielb987, 2018-08-04:
I have used a Raspberry Pi a couple of years ago and I had problems with it if I removed power without shutting down linux first. The memory card got messed up if I didn't closed the file system by shutting down Linux. Therefore, if you implement "Reboot computer", it would be nice to also have "Shutdown computer".

I think these two should be disabled by default so that the user needs to turn them on if he wants to use them, so that this feature is not enabled by misstake.

Maybe it would be good to also be able to set a password for this feature, for example if used by a club, so that not everyone with access to the web page could shutdown the computer.

rhwood, 2018-08-05:
If your raspberry pi OS provides packages for Cockpit ( https://cockpit-project.org ), I would use that web service for managing the Raspberry Pi itself, and not attempting to build authentication/authorization controls into JMRI itself.

mstevetodd, 2018-08-07:
I agree the OS-specific features don't belong in JMRI, but I was initially stumped on how to integrate links to another web server into the JMRI web menus. Then I recalled that all web-related files can be "overridden" by placing same-named files in the preferences folder structure. I'll go that route.

mstevetodd, 2018-08-13:
Update:
I've added the Tables menu item which works for a number of JMRI object types. You can see it in action at http://jmri.mstevetodd.com:1080
I've also overridden the navbar on my latest RPi-JMRI image to include links to the session log and to buttons for restarting or shutting down the RaspberryPi.
I installed Cockpit, but either that or the dependent updates broke the wifi access point, so I had to revert that change. I'll try it again in the future.

mstevetodd, 2019-11-25:
Closing. I'd still like a Config Profile Selector, but I will create a more specific issue for that.

closed 2019-11-25 by mstevetodd