JMRI/JMRI #3320: Some thoughts around improving how Conditionals "names" work

bobjacobsen, 2017-04-02:
This issues captures a few ideas from jmriusers and jmri-developers discussions around how Conditionals (the calculation part of Logix) handle user and system names.  Several people are thinking about this (around 1st April 2017), so if you're going to work on the code in this area, please assign yourself to this issue.

- It may be time to rethink the Logix/Conditional interface and the need for names.

Users don't really need to know anything about Conditional names, either system or user, and the right way to fix the issues that arose recently (about whether user names need to be unique or not, and problems either way) is to remove them from the interface. We’re putting way too much effort into thinking about “names” and what patten they should have, when in fact the user doesn’t have any particular need to know names:  They just want the program to know the one they’re talking about. So that’s what we should show them:

- [ ]  When looking at the Conditionals within a Logix, they should just be a list, with perhaps a user-entered “comment” or “description”, and a brief summary of what they contain

- [ ] When you have to select/specify one, e.g. to refer to it from another Conditional, pirovide a GUI widget that lets you pick first the Logix, then pick the Conditional within that Logix (e.g. a pair of comboboxes) by referring to that description and summary.

- [ ] Success would be defined by getting the phrases “system name” and “user name” for Conditionals out of the GUI, Javadoc and help files.

This will probably take some advance work on the Conditional and Logix code. The Logix code, both for run time and for add/edit, needs a bunch of work.  In particular:

- [ ] Both Logix and Conditionals need tests that ensure their various references to NamedBeans are properly initializing and updating, particularly when references are changed by e.g. renaming.
- [ ] There are some tests of the user interface for Add/Edit, but they're brittle and tend to leave windows open (often due to problems with trying to open more than one window as the tests run in parallel: The dreaded "You can't edit more than one Logix" message)
 
For implementing the change, @dsand47 suggested: 

- [ ] `As described, it seems to me that this is a schema change.  A Logix becomes a container similar to a Layout Editor panel.  Within that container would be 1 to n Conditional containers each with 1 to n Variable and 0 to n Action objects.`

It certainly could be an XML schema change, or not if that's easier.  But improving the internal structure to have Conditionals kept inside individual Logix, without having them be free-floating NamedBeans in a global manager, seems like a very good idea.

JLang1951, 2017-04-02:
As I posted in JMRI usergroups I installed the build that Bob posted and my conditional usernames are back but there are still a lot of errors in the system console. Here is the system console:

[Sys_Console 04-02-17.txt](https://github.com/JMRI/JMRI/files/888545/Sys_Console.04-02-17.txt)

Here is the panel file that I load:

[J_Lang_LE_panel.zip](https://github.com/JMRI/JMRI/files/888546/J_Lang_LE_panel.zip)

John Lang

dsand47, 2017-04-02:
While a Java rookie, I do have some thoughts about how we could demote Conditionals.

The XML schema does not change.  Backward compatibly is important.  DefaultLogixManagerXML and DefaultConditionalMangerXML would be adapted.

A HashMap would be added to DefaultLogix.  The key would be the Conditional system name and the contents would be a Conditional object.  The _conditionalSystemNames ArrayList still provides the conditional sequence information.

The Conditional GUI components would not display the system name, just a name/description supplied by the user (internally still a userName for XML compatibility).  I would not bother with comments.  I suspect that very few people would use them.

Phase 1 is basically converting the data structures and adapting the GUI.  Phase 2 could look at alternate GUI ideas such as tree structures.

Dave Sand

pabender, 2017-04-02:
Sent from my iPad

> On Apr 2, 2017, at 7:30 PM, Dave Sand <notifications@github.com> wrote:
> 
> While a Java rookie, I do have some thoughts about how we could demote Conditionals.
> 
> The XML schema does not change. Backward compatibly is important. DefaultLogixManagerXML and DefaultConditionalMangerXML would be adapted.
> 
Backwards compatibility does not mean the XML schema cannot change.  What it means is that new versions of the software will continue to be able to read old files.  

We never guarantee that you can take a file written with a newer version of the software and use it on an older version.  If we did that, adding new features would be difficult.

At this point, we are really just talking about making the XML files allow the structure the GUI has always enforced.

Paul

JLang1951, 2017-04-03:
Just an FYI the panel file that I put up here was created over 2 years ago without any changes since then. I can see where a file created with a newer version would not be backward compatible with an older version of JMRI but hopefully newer versions of JMRI will allow older panel files to still function.

John

bobjacobsen, 2017-04-28:
See #3430

closed 2020-01-13 by bobjacobsen