JMRI/JMRI #3324: Duplicate CMRI Tabs in the Turnout and Sensor Tables

dsand47, 2017-04-02:
Based on the Yahoo posting by John Lang, I have been doing some research into this problem.

I believe that I may have found the root of the problem, but I don't know enough about the **jmrix** components to be sure.  I also have no way to test CMRI live, only simulation.  It is possible that the duplicate tabs only exist when using simulation.

I have attached two files.  The first file is a **git diff** output that identifies where I put in diagnostic code, as well as a brute force fix for the duplicate table tabs.  The second file is a session log.

[CMRI-Diff.txt](https://github.com/JMRI/JMRI/files/888717/CMRI-Diff.txt)
[session.log.txt](https://github.com/JMRI/JMRI/files/888719/session.log.txt)

Dave Sand

KenC57, 2017-04-02:
I've seen a few times where the preference editor may try to create a second
connection when you really are just changing a real to simulator or the
other way around. You need to watch the tab to make sure you end up editing
or deleting the correct tabs to leave you with the same connections you
started with. Part of the issue is if you don't get it right you would get
things like a N2 instead of N for the connection prefix.

dsand47, 2017-04-02:
I have also seen that.  Sometimes I end up deleting all of the connections and add them back in one at a time after restarting.

For the testing of this issue, the profile.xml was visually inspected and not changed during the testing activities.

Dave Sand

devel-bobm, 2017-04-03:
Reference: #2953

JLang1951, 2017-04-03:
I just had a chance to fire up the RR. and load this panel live, I still get the duplicate CMRI tabs in the sensors & turnout tables. I never encountered this problem with this panel file or any others that I have loaded with JMRI 4.57 or earlier versions. Everything started to go south with v. 4.6 and later. I thought that I would mention this if it helps.

John

spassmo, 2017-04-06:
Duplicate of #2828

bobjacobsen, 2017-04-16:
I believe this is now fixed, would appreciate feedback.

JLang1951, 2017-04-16:
I tried the build  JMRI.4.7.3ish+jenkins+20170416T0404Z+Ra7628aa.exe from Jenkins on a number of my panel files and the duplicate tab is gone in the tables. I also went into turnout settings and changed and saved changes and all seems to be working. No errors in the Sys_Console. Looks good so far.
Thanks,
John

closed 2017-04-22 by bobjacobsen