JMRI/JMRI #3346: Switchboard support in Webserver

silverailscolo, 2017-04-06:
Clicking an open Switchboard panel in WebServer currently will display a loading error. As @rhwood stated in #3309 , a new servlet has to be written to support this type of panel as an operating display.
Besides creating ``JMRI/java/src/jmri/web/servlet/panel/SwitchboardServlet.java`` and some extra methods in ``SwitchboardEditor.java``, could an expert tell me if this will also require additions to ``JMRI/web/js/panel.js`` or elsewhere?
The handling of states and text/icon properties in the BeanSwitch class comes quite close to the way existing panel servlets handle this. Placing the switches in an html frame could be done as simple as popping them in as one string of elements wrapping to the next line, or I could try css positioning.

mstevetodd, 2017-04-06:
@silverailscolo, can you post an example panel file so I can see what the new xml looks like?

silverailscolo, 2017-04-07:
@mstevetodd attached is a small panel file with a switchboard (near the end of the file).
Switches are not ``<positionable>`` items, as they have no x, y of their own. In fact, they aren't as separate items in xml, but created from the (address) range min-max, the bean type and connection.
But it would be easy to supply an ArrayList of type ``<BeanSwitch>`` to the servlet upon request.
[switchboarddemo.xml.txt](https://github.com/JMRI/JMRI/files/904621/switchboarddemo.xml.txt)

mstevetodd, 2017-04-07:
Thanks, @silverailscolo, but I cannot open this file with the latest HEAD. Looks like some images are missing:
`switchboardEditor.SwitchboardEditor   ERROR - error reading image from resources/icons/misc/switchboard/ -off-s.png-resources/icons/misc/switchboard/ -on-s.png`
Can you commit the missing images?

mstevetodd, 2017-04-07:
I looked at the new switchboardeditor element in your file. Here are my thoughts on supporting this as a client-side panel:
1) First get the basic xml returning for the path `/panel/Switchboard/Schakelpaneel?format=xml`. This xml should mirror the switchboardeditor element. 
2) You may want to add include additional child elements if that makes things work better. The existing panels use this to "explode" details such as paths to image files for aspects, etc. Remember that panel.js doesn't have access to the JMRI runtime, except as communicated thru this xml and json messages.
3a) In panel.js, line 115, the code loops through all child elements of the current panels, creating widgets for each. If you add the children in 2), this can stay the same. 
3b) If not, you'll need a separate code path for switchboardeditor, which generates the desired widgets based on the panel attributes (rangemin, rangemax, type, etc.). This seems like it would require duplicating a lot of the assumptions you make in your jPanel code, so I prefer 2) adding the children in the servlet so they are returned in the XML.
4) It looks like you're using the same system items (turnouts, sensors, etc.), so no new json should be needed. Just make sure your new widgets define the expected states, and the system items are hooked up properly: `jmri.getTurnout(turnoutname)`, etc.
5) You may be able to use the `$drawIcon(widget)` code to draw and redraw your new widgets on creation and json change. If not, you'll need new $drawXXX() routines. This will likely lead you to include the icon image paths in the xml for each widget.
6) Since all of your widgets appear to be two-state, $HandleClick() should work fine.

silverailscolo, 2017-04-07:
The path for the images should read ``resources/icons/misc/switchboard/T-on-s.png``, so tonight I will have to check where the missing T (or S or L) has gone.

mstevetodd, 2017-04-07:
A bit more debugging and I found that I had an error in my Loconet Simulator connection prefix, I had a trailing space, so none of the system items (LT1, etc.) were loading. Once I removed that, the error went away, and your jPanel seems to be working fine now. Sorry about that.

KenC57, 2017-04-07:
SteveT,

 

So the connection entry for the prefix in the Edit->Preferences->Connections
had the extra space?? I guess that's another place we need to watch for.

-ken c

mstevetodd, 2017-04-07:
@KenC57 Yes, the Connection Prefix keeps trailing spaces. Eventually I noticed an error that included 
`LT1 needed L T`

silverailscolo, 2017-04-07:
Thanks, and I'll add a check (and replacement default) for a space at position 2 in the label name.

silverailscolo, 2017-04-08:
Cannot explain why a ``/panel/Switchboard/`` is cast to PanelEditor.
```
2017-04-08 21:23:14,779 servlet.ServletHandler                WARN  - /panel/Switchboard/Schakelpaneel [WebServer-69]
java.lang.ClassCastException: jmri.jmrit.display.switchboardEditor.SwitchboardEditor cannot be cast to jmri.jmrit.display.panelEditor.PanelEditor
	at jmri.web.servlet.panel.PanelServlet.getXmlPanel(PanelServlet.java:44)
	at jmri.web.servlet.panel.AbstractPanelServlet.getPanelText(AbstractPanelServlet.java:177)
	at jmri.web.servlet.panel.AbstractPanelServlet.doGet(AbstractPanelServlet.java:120)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:687)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:845)
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:583)
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1174)
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:511)
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1106)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)
	at org.eclipse.jetty.server.handler.ContextHandlerCollection.handle(ContextHandlerCollection.java:213)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:134)
	at org.eclipse.jetty.server.Server.handle(Server.java:524)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:319)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:253)
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:273)
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:95)
	at org.eclipse.jetty.io.SelectChannelEndPoint$2.run(SelectChannelEndPoint.java:93)
	at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.executeProduceConsume(ExecuteProduceConsume.java:303)
	at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.produceConsume(ExecuteProduceConsume.java:148)
	at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.run(ExecuteProduceConsume.java:136)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:671)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:589)
	at java.lang.Thread.run(Thread.java:745)
```

bobjacobsen, 2017-04-08:
> Cannot explain why a /panel/Switchboard/ is cast to PanelEditor.

The `getXmlPanel(name)` call into `jmri.web.servlet.panel.PanelServlet`end up at code that only knows about PanelEditor panels. 

The first thing to try might be to change line 44 to Editor instead of PanelEditor.  That’s a common base for PanelEditor and Although I only took a quick look, much of what’s being asked for is present in Editor.

silverailscolo, 2017-04-08:
@bobjacobsen thanks a lot. Though I will keep this change local until sure it didn't break other webserver stuff, it worked and got me to step 2 of step 2 as @mstevetodd  suggested. Classic Panel still working too. First lines of Switchboard XML:
```
<?xml version="1.0" encoding="UTF-8"?>
<panel name="Schakelpaneel" height="373" width="607" panelheight="354" panelwidth="588" showtooltips="yes" controlling="yes">
  <backgroundColor red="0" green="0" blue="0" />
</panel>
```

silverailscolo, 2017-04-09:
The above change looked promising but further experimenting showed it still interpreted the new panel type as a classic Panel, ignoring the instructions in the SwitchBoardServlet. My IDE's 'Find in path...' search did not tell where in the code a choice for a particular Servlet is made.  Using LayoutEditor panels as an example did reveal that any new servlet should be registered as a service in ``jmri/src/META-INF/services/javax.servlet.HttpServlet``. With that fixed, and reverting the changes in ``Panels.js``, a Switchboard will now load its own properties in the browser:
```
<?xml version="1.0" encoding="UTF-8"?>
<panel name="Schakelpaneel" height="373" width="607" panelheight="354" panelwidth="588" showtooltips="yes" controlling="yes" hideunconnected="no" rangemin="1" rangemax="32" type="T" connection="L" shape="button" columns="8" defaulttextcolor="white">
  <backgroundColor red="0" green="0" blue="0" />
</panel>
```
Notice stuff like 'rangemin', unique to Switchboards. Not sure where to add what I learned in the technical help, but this completes step 1.

silverailscolo, 2017-04-11:
I've made it to step 5 (for a basic 'button' style Switchboard using only float: left). Clicking drives the layout via existing json objects. Next up is providing some sort of feedback. 
[SWB branch](https://github.com/silverailscolo/JMRI/tree/eb-switchboard-cpe)
Scary bug: every time the server refreshes the web panel, it adds a new set of switches instead of replacing them. This might be caused by the way the switchboard is rebuilt in java.

[image]

silverailscolo, 2017-04-12:
PR #3385 contains an update with the above bug fixed and operating switch buttons. Graphic icon styles (step 6) are not yet available in web server. However simple, it looks rather OK on mobile devices.
BTW I was very impressed by the code present (.js, .xml, .css, JSON!). After grasping how all that is supposed to work, porting the Switchboard GUI was a matter of adding small bits, step by step. Having the menu option available to view the XML in the browser was indispensable!

closed 2017-04-12 by silverailscolo