JMRI/JMRI #3352: Roster Speed Profile arbitrarily deducts 2 seconds from time

mattharris, 2017-04-07:
The Roster Speed Profile speed calculation routine arbitrarily deducts 2 seconds from the recorded time.

In many circumstances, this leads to inaccurate speed calculations.

For more background [see thread on jmriusers](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/137379).

mattharris, 2017-04-07:
As there may well be circumstances that require adjustments, investigate allowing the 'speed calculation adjustment' to be modified in the UI

Pugwash1, 2017-04-08:
The 2 seconds is a big red herring.

The Speed Profile is not about accurate speed. It is about building a set of lookup tables about how a loco reacts to throttle settings. Most of the places that it was used, such as AutoEngineer and Warrents, have been ruthlessly destroyed by changes made by people unaware of the underlying logic.
The concept is that you create a set of: this loco does this velocity for this throttle setting. 
You select your gold standard curve.
When you set any speed, either from a JMRI throttle or AutoEngineer,  it looks at the gold standard curve, gets the velocity,  uses that velocity to get a throttle setting for the current loco and bingo, the same speed results. No speed matching required.

Its not about speed matching or speed measuring, its about consistent velocity for given throttle setting.

Now if you just want the old measure speed stuff, then to make it work you need to:
Not send change direction or speed faster than the command station can react (thread sleep blah)
Make sure that all the sensors are on the same sensing system, e.g. don't have one sensor on a bd4 and another on a bdl168, or a bd20, or a different bdl168, or any other manufacturers occupancy detectors.
The current screen needs to be consistent with the current screen for all  default values, this will prevent user push back to change.
It should save the current screen values on exit or have the ability to do so, this will save user hassle.
Have the ability to set fwd start sensor, fwd end sensor, rev start sensor, rev end sensor. This will allow the tests to cover multiple blocks, currently only one block is allowed.
It must still save profiles as before to ensure backward compatibility. (build users one at a time, lose them all by one small change)

If my diatribe is a little curt, I'm sorry. I just spent 2 days looking at the code and counting sleepers and finding a great idea going down the toilet.

Steve G

Shortd70, 2017-04-08:
I have written some jython scripts to control my trains. Warrants and Signal Controlled Warrants don't do quite what I want, so I have used Warrants to define the routes and sensors; I have used Oblocks to specify block speeds and use the speed profiles to run the trains at the speed for the block. However whilst profiling my latest loco, I realised that the results were incorrect!
I agree with your comments about sensor types; being able to use more than one block would be useful.
However the rest of the function is fine - but why subtract 2 seconds?

Dave

Pugwash1, 2017-04-08:
OK.  I've started to do this. I've discovered a few  problems where these profiles are used and will review them. Can some assign this to me. Tanks.

mattharris, 2017-04-09:
@Pugwash1 Can't assign you right now as you're not (currently) a member of the JMRI organisation - I've sent an invite which should rectify this.

Pugwash1, 2017-04-10:
Ahhhhh. OK changed exception type.

bobjacobsen, 2017-04-16:
Does #3369 resolve this? If so, can we close it? Thanks

Shortd70, 2017-04-16:
Hi Bob,

I looked at the code changes, and it looks ok, so I downloaded  the latest development version of JMRI and installed it, but It won't run - I get the JMRI logo screen and the LocoNet Simulator dialog but nothing else.

Any ideas?

regards,

Dave

The message does not contain any threats
AVG for MS Exchange Server (2013.0.3556 - 4769/14327)

Pugwash1, 2017-04-16:
Hi
It should resolve the basic problem.
In working on it I discovered a number of other problems and some needed improvements, but before I can finish with them I need to know where I should be taking the "scale" from. See previous email question.
Steve G.

On April 15, 2017 9:41:16 PM EDT, Bob Jacobsen <notifications@github.com> wrote:
>Does #3369 resolve this? If so, can we close it? Thanks

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

Shortd70, 2017-04-19:
Hi Bob,

I've downloaded and installed 4.7.3 - it runs, and the Speed Profile now works as I expected.

Thanks very much

regards,

Dave Short

AVG for MS Exchange Server (2013.0.3556 - 4769/14346)

closed 2017-04-19 by bobjacobsen