JMRI/JMRI #3359: Empty xmlns element when persisting connections

bobjacobsen, 2017-04-08:
Current code writes out connections (e.g. for C/MRI serial, which is this example) like this:
```
<?xml version="1.0" encoding="UTF-8"?>
<auxiliary-configuration xmlns="http://www.netbeans.org/ns/auxiliary-configuration/1">
    <connections xmlns="http://jmri.org/xml/schema/auxiliary-configuration/connections-2-9-6.xsd">
        <connection xmlns="" class="jmri.jmrix.cmri.serial.serialdriver.configurexml.ConnectionConfigXml" disabled="no" manufacturer="C/MRI" port="/dev/tty.usbserial-FTNOECFG" speed="9,600 baud" systemPrefix="C" userName="C/MRI">
            <options/>
        </connection>
    </connections>
    <startup xmlns="http://jmri.org/xml/schema/auxiliary-configuration/startup-4-3-5.xsd"/>
</auxiliary-configuration>
```

Note the `xmlns=""` attribute on the `<connection>` element.  That should probably either have a valid content (see line directly above it) or be omitted.

bobjacobsen, 2020-01-13:
The source of this is (apparently) the namespace in the "connections" enclosing element.  That's defined by this line:
```
java/src//jmri/jmrix/ConnectionConfigManager.java:        Element element = new Element(CONNECTIONS, NAMESPACE);
```

When the "connection" element is defined without a namespace in the constructor call, the element has to be written with an empty namespace.

It's not clear that JMRI should be writing profile information into that namespace. Figuring that out would be the next step on this.  Our test files don't use "xmlns=" _except_ for profiles and docbook in three files (profiles removed):

```
% grep -r xmlns= java/test/jmri/
java/test/jmri//swing/JmriJTablePersistenceManagerTest-user-interface.xml:        <table xmlns="" name="Test1">
java/test/jmri//swing/JmriJTablePersistenceManagerTest-user-interface.xml:        <table xmlns="" name="Test2">

java/test/jmri//implementation/testAspects.xml:    <copyright xmlns="http://docbook.org/ns/docbook">
java/test/jmri//implementation/testAspects.xml:    <authorgroup xmlns="http://docbook.org/ns/docbook" >
java/test/jmri//implementation/testAspects.xml:    <revhistory xmlns="http://docbook.org/ns/docbook">

java/test/jmri//implementation/test-appearance-one-searchlight.xml:    <copyright xmlns="http://docbook.org/ns/docbook">
java/test/jmri//implementation/test-appearance-one-searchlight.xml:    <authorgroup xmlns="http://docbook.org/ns/docbook">
java/test/jmri//implementation/test-appearance-one-searchlight.xml:    <revhistory xmlns="http://docbook.org/ns/docbook">
```

A scan of the xml/ directory is similar:  Just profiles and docbook

danielb987, 2020-01-13:
I have a `user-interface.xml` in my profile in which `preferences`, `comboBoxLastValue` and `window` have empty `xmlns`.

`<preferences xmlns="" class="jmri.jmrit.beantable.SignalMastTableAction"/>`
`<comboBox xmlns="" lastSelected="LocoNet LocoBuffer-USB" name="Digitrax"/>`
`<window xmlns="" class="jmri.jmrit.roster.Roster"/>`

closed 2021-04-14 by stale[bot]