JMRI/JMRI #3370: Missing resource

rhwood, 2017-04-09:
MenuFile is missing from resources called for by jmri.jmrit.catalog.ImageIndexEditor

Stacktrace is
```
2017-04-08 23:28:09,647 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.util.MissingResourceException: Can't find resource for bundle java.util.PropertyResourceBundle, key MenuFile
at java.util.ResourceBundle.getObject(Unknown Source)
at java.util.ResourceBundle.getString(Unknown Source)
at jmri.jmrit.catalog.ImageIndexEditor.init(ImageIndexEditor.java:76)
at jmri.jmrit.catalog.ImageIndexEditor.instance(ImageIndexEditor.java:69)
at jmri.jmrit.display.panelEditor.PanelEditor$3.actionPerformed(PanelEditor.java:180)
at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
at javax.swing.AbstractButton.doClick(Unknown Source)
at javax.swing.plaf.basic.BasicMenuItemUI.doClick(Unknown Source)
at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(Unknown Source)
at java.awt.Component.processMouseEvent(Unknown Source)
at javax.swing.JComponent.processMouseEvent(Unknown Source)
at java.awt.Component.processEvent(Unknown Source)
at java.awt.Container.processEvent(Unknown Source)
at java.awt.Component.dispatchEventImpl(Unknown Source)
at java.awt.Container.dispatchEventImpl(Unknown Source)
at java.awt.Component.dispatchEvent(Unknown Source)
at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
at java.awt.Container.dispatchEventImpl(Unknown Source)
at java.awt.Window.dispatchEventImpl(Unknown Source)
at java.awt.Component.dispatchEvent(Unknown Source)
at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
at java.awt.EventQueue.access$500(Unknown Source)
at java.awt.EventQueue$3.run(Unknown Source)
at java.awt.EventQueue$3.run(Unknown Source)
at java.security.AccessController.doPrivileged(Native Method)
at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
at java.awt.EventQueue$4.run(Unknown Source)
at java.awt.EventQueue$4.run(Unknown Source)
at java.security.AccessController.doPrivileged(Native Method)
at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
at java.awt.EventQueue.dispatchEvent(Unknown Source)
at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
at java.awt.EventDispatchThread.run(Unknown Source)
```

From [the user's group](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/137509;_ylc=X3oDMTM4MW4za2xkBF9TAzk3MzU5NzE0BGdycElkAzY2MDI3NTgEZ3Jwc3BJZAMxNzA1MDYzNTc2BG1zZ0lkAzEzNzUxMgRzZWMDZnRyBHNsawN2dHBjBHN0aW1lAzE0OTE3MjI0NDMEdHBjSWQDMTM3NTA5)

rhwood, 2017-04-10:
This appears to have been an issue only in 4.5.? to 4.6 and fixed in 90f5f7d which was included in 4.7.1 and newer releases. I've asked the original reporter to confirm this is the case for him.

Walter71342JMRI, 2017-04-10:
randall: I think there is still a bug om 4.7.2. If I left click on it I get a blank screen, Right click work fine.
Walt

rhwood, 2017-04-10:
Original reporter confirms that this error is not in 4.7.2.

closed 2017-04-10 by rhwood