JMRI/JMRI #3386: JMRI Startup hangs:  java.lang.NullPointerException

dsand47, 2017-04-12:
After creating or changing an LRoute in 4.6 or 4.7.x and selecting **_skip message_** the following is added to user-interface.xml:

        <preferences xmlns="" class="jmri.jmrit.beantable.LRouteTableAction">
            <reminderPrompts>
                <reminder>remindSaveRoute</reminder>
            </reminderPrompts>
        </preferences>

Subsequent attempts to start JMRI hang with the Java Error.  The attached session log file provides the details.

Removing the xml fragment allows JMRI to start.  If the **skip** message option is not checked, there is no issue.

[session.log.txt](https://github.com/JMRI/JMRI/files/918104/session.log.txt)

Dave Sand

bobjacobsen, 2017-04-14:
I don't understand enough to actually fix this, but I'll shortly provide a PR that prevents the hang.

rhwood, 2017-04-14:
The user preferences managers have long expected to be able to use a default constructor for any class that stores preferences with them to be able to get a user-presentable description of the class. That is what's going wrong here (a default constructor is failing to create a class); but I haven't looked to see why it's failing. I don't know if that is the best design to deal with the problem or not, but it is the one that has been in place.

bobjacobsen, 2017-04-14:
Could you try a build with #3390 to see if the hang is fixed?  You'll still get an exception logged, and somebody will have to figure out what's going on there.  Thanks.

dsand47, 2017-04-15:
The hang no longer occurs.  There is an error message displayed in the system log:

2017-04-14 23:56:31,969 managers.JmriUserPreferencesManager   ERROR - setClassDescription(jmri.jmrit.beantable.LRouteTableAction) failed in newInstance [main]

The entire session log is attached.

[session.log.txt](https://github.com/JMRI/JMRI/files/923346/session.log.txt)

Dave Sand

closed 2018-02-23 by dsand47