JMRI/JMRI #3387: PanelPro intermittent script NPE

Cliff-Anderson, 2017-04-13:
Big nuisance: 

A PanelPro Start Up Null Pointer Exception that seems to be loosely related to executing some script and that only occurs sometimes but also seems to cause no real harm is becoming as irritating as a hangnail.  

Basic facts:    64 bit Windows 10 Home system  (Forcefully updated from Windows 8.1 without my permission)
        Intel(R) Core(TM) i7-4510U CPU @ 2.00 GHz  2.60 GHz
        12.0 GB RAM
        904 GB Disk        Free Space 762 GB
        JMRI 4.7.2 and 4.6
        JRE:  Java 1.8.0_121  64 bit
        JDK:  8u112  64 bit

PanelPro Profile:  LocoNet Simulator
    Open:  apps.SystemConsoleAction
    Execute Script:  C:/Program Files (x86)/JMRI/jython/AddButton.py

Usually Notepad++ is also running, and sometimes FireFox is too.  Lots of background Windows stuff churning beyond my control.  

If I start PanelPro as described above, repeatedly (consistently either with 4.6 or else with 4.7.2 but not mixing them) and quit, wait more than thirty seconds, and then repeat, something like a dozen to twenty times in a row, at least once the result is a Null Pointer Exception report in the log.  No harm seems to be done either and the program appears to run as if nothing serious had happened.  Perhaps it is just a false alarm?  I also have Eclipse Java Neon installed and can Pull source files from GitHub.  The build seems OK, but recently, I am unable to Run the built product.  I only mention that, because it is possible that I have messed something up with Eclipse that is contributing to the issue described above.  Otherwise, I would have run this exercise with the latest Pull version too.  

It seems possible that this intermittent Null Pointer Exception is the result of something wrong with my computer.  

The Null Pointer Exception problem was discovered while debugging and testing a Jython script that I am working on, and considerable time was wasted trying to figure out what was wrong with my script.  It had the symptoms of an intermittent problem internal to my script, since most of my script development is done by trial and error and try again.  

Eventually, I tried just the LocoNet Simulation and the System Console and repeated that at least one hundred times in a row with no problems.  Finally, I selected a script from the distribution folder, and it happened to be AddButton.py because I am very familiar with it.  

With the same kind of repeatability, if that is the correct word in these circumstances, the problem that my script seemed to be causing, showed up with a well-worn and respectable product.  The message and the traceback are identical with different scripts.

One of the session.log files from this morning follows.  Hopefully someone can make more sense out of it than I can.

A zipped copy of that session.log file is attached.

Cliff Anderson
[NPE V4-7-2 session.zip](https://github.com/JMRI/JMRI/files/918476/NPE.V4-7-2.session.zip)

bobjacobsen, 2017-04-14:
I don't quite understand why you're trying to run AddButton.py.  Do you have a modified version?  The one that's distributed with JMRI is marked as obsolete, just there for people who are already using it.  The recommended way to add a button now is use the Startup Items Preferences.

Cliff-Anderson, 2017-04-14:
The message got muddled, as my intent was to articulate the fact that even a simple tried and true distribution script can cause an NPE when requested from the Start Up menu, not to claim that there was anything wrong with any particular distribution script.  

This intermittent problem seems more likely to be Java or a Win10 issue than a JMRI issue, ore maybe a timing issue, but I am way out of my safe turf.

Today I tested with several other distribution scripts.  Using "->" to replace four consecutive blanks and ampersand+"lt"+semicolon to replace each less-than character, the following text imitates the profile.xml file requesting an unchanged copy of "InitTurnouts.py" in the startup:

&lt;?xml version="1.0" encoding="UTF-8"?>
&lt;auxiliary-configuration xmlns="http://www.netbeans.org/ns/auxiliary-configuration/1">
->&lt;connections xmlns="http://jmri.org/xml/schema/auxiliary-configuration/connections-2-9-6.xsd">
->->&lt;connection xmlns="" class="jmri.jmrix.loconet.hexfile.configurexml.ConnectionConfigXml" disabled="no" manufacturer="Digitrax" systemPrefix="L" userName="LocoNet">
->->->&lt;options>
->->->->&lt;option>
->->->->->&lt;name>SensorDefaultState&lt;/name>
->->->->->&lt;value>Unknown&lt;/value>
->->->->&lt;/option>
->->->&lt;/options>
->->&lt;/connection>
->&lt;/connections>
->&lt;startup xmlns="http://jmri.org/xml/schema/auxiliary-configuration/startup-4-3-5.xsd">
->->&lt;perform xmlns="" class="apps.configurexml.PerformActionModelXml" name="apps.SystemConsoleAction" type="Action">
->->->&lt;property name="systemPrefix" value=""/>
->->&lt;/perform>
->->&lt;perform xmlns="" class="apps.configurexml.PerformScriptModelXml" name="C:/Program Files (x86)/JMRI/jython/InitTurnouts.py" type="ScriptFile"/>
->&lt;/startup>
&lt;/auxiliary-configuration>

The first time I opened PanelPro, the same intermittent nonsense appeared.  Usually it takes many tries to trigger the problem when trying to trace it.  It always seems more frequent when I am working on some other puzzle.  In the following copy of "session.log" the symbols "->" have replaced all of the tab characters.   From the line starting with "2017-04-14 11:27:37,875 " and many on down, the message contents are identical to what I have found and reported earlier.  

2017-04-14 11:27:17,479 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-04-14 11:27:17,541 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\cliffaa-rr\JMRI\log\messages.log [main]
2017-04-14 11:27:17,541 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\cliffaa-rr\JMRI\log\session.log [main]
2017-04-14 11:27:17,619 apps.Apps                             INFO  - PanelPro version 4.7.2-R741b21f starts under Java 1.8.0_121 on Windows 10 amd64 v10.0 at Fri Apr 14 11:27:17 PDT 2017 [main]
2017-04-14 11:27:21,796 apps.Apps                             INFO  - Starting with profile LocoNet_sim_naked.3e75bb94 [main]
2017-04-14 11:27:22,390 node.NodeIdentity                     INFO  - Using jmri-F81654D51042-3e75bb94 as the JMRI Node identity [main]
2017-04-14 11:27:24,593 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-8]
2017-04-14 11:27:25,452 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: BB 00 00 44 [AWT-EventQueue-0]
2017-04-14 11:27:25,452 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 78 27 10 [AWT-EventQueue-0]
2017-04-14 11:27:25,468 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 79 27 11 [AWT-EventQueue-0]
2017-04-14 11:27:26,140 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 7A 27 12 [AWT-EventQueue-0]
2017-04-14 11:27:26,171 apps.PerformScriptModel               INFO  - Running script C:\Program Files (x86)\JMRI\jython\InitTurnouts.py [main]
2017-04-14 11:27:26,202 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.0 [main]
2017-04-14 11:27:26,234 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0_121 [main]
2017-04-14 11:27:27,078 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 7B 27 13 [AWT-EventQueue-0]
2017-04-14 11:27:27,078 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 78 07 30 [AWT-EventQueue-0]
2017-04-14 11:27:27,218 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 79 07 31 [AWT-EventQueue-0]
2017-04-14 11:27:27,718 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 7A 07 32 [AWT-EventQueue-0]
2017-04-14 11:27:28,218 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 7B 07 33 [AWT-EventQueue-0]
2017-04-14 11:27:37,843 logix.WarrantPreferences              INFO  - Found Warrant preferences file: C:\Users\cliffaa-rr\JMRI\Config Profiles\LocoNet_sim_naked\signal\WarrantPreferences.xml [main]
2017-04-14 11:27:37,875 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
->at javax.swing.text.WrappedPlainView$WrappedLine.paint(Unknown Source)
->at javax.swing.text.BoxView.paintChild(Unknown Source)
->at javax.swing.text.BoxView.paint(Unknown Source)
[... cut ...]
2017-04-14 11:27:38,703 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\cliffaa-rr\JMRI\Config Profiles\LocoNet_sim_naked\ [main]
2017-04-14 11:27:38,718 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\cliffaa-rr\JMRI\ [main]
2017-04-14 11:27:38,718 util.FileUtilSupport                  INFO  - File path home: is C:\Users\cliffaa-rr\ [main]
2017-04-14 11:27:38,734 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\cliffaa-rr\JMRI\jython\ [main]
2017-04-14 11:29:31,225 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]

rhwood, 2017-04-14:
Instead of modifying content you want to share literally by adding non-space characters, use three backticks on the lines preceding and following the content.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

Cliff-Anderson, 2017-05-29:
Sorry to be a pest, but this silly thing is still repeatable.

What follows is copied from the Eclipse Console Log window and based on a Pull that was done only a few minutes ago. The same almost empty Config Profile as described before was used and is also copied here.  

If only I had a better grasp on the use of Eclipse, maybe I could provide some useful info.

My real concern is that what is causing this may mask some other issues.  Thanks, Randall for the editing clue.

Cliff

```
17:18:08.868 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
17:18:08.942 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\cliffaa-rr\JMRI\log\messages.log [main]
17:18:08.946 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\cliffaa-rr\JMRI\log\session.log [main]
17:18:08.972 apps.Apps                             INFO  - DecoderPro version 4.7.4ish+cliffaa-rr+20170524T0320Z starts under Java 1.8.0_112 on Windows 10 amd64 v10.0 at Sun May 28 17:18:08 PDT 2017 [main]
17:18:13.273 apps.Apps                             INFO  - Starting with profile LocoNet_sim_naked.3e75bb94 [main]
17:18:14.041 node.NodeIdentity                     INFO  - Using jmri-F81654D51042-3e75bb94 as the JMRI Node identity [main]
17:18:16.255 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-9]
17:18:17.173 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: BB 00 00 44 [AWT-EventQueue-0]
17:18:17.175 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 78 27 10 [AWT-EventQueue-0]
17:18:17.184 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 79 27 11 [AWT-EventQueue-0]
17:18:17.590 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 7A 27 12 [AWT-EventQueue-0]
17:18:18.427 apps.PerformScriptModel               INFO  - Running script C:\Program Files (x86)\JMRI\jython\InitTurnouts.py [main]
17:18:18.529 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.0 [main]
17:18:18.539 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0_112 [main]
17:18:18.575 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 7B 27 13 [AWT-EventQueue-0]
17:18:19.511 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 78 07 30 [AWT-EventQueue-0]
17:18:19.513 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 79 07 31 [AWT-EventQueue-0]
17:18:19.832 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 7A 07 32 [AWT-EventQueue-0]
17:18:20.332 loconet.LnTrafficController           DEBUG - notify of incoming LocoNet packet: B0 7B 07 33 [AWT-EventQueue-0]
17:18:30.995 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at javax.swing.text.WrappedPlainView$WrappedLine.paint(WrappedPlainView.java:584)
	at javax.swing.text.BoxView.paintChild(BoxView.java:161)
	at javax.swing.text.BoxView.paint(BoxView.java:433)
	at javax.swing.text.WrappedPlainView.paint(WrappedPlainView.java:369)
	at javax.swing.plaf.basic.BasicTextUI$RootView.paint(BasicTextUI.java:1434)
	at javax.swing.plaf.basic.BasicTextUI.paintSafely(BasicTextUI.java:737)
	at javax.swing.plaf.basic.BasicTextUI.paint(BasicTextUI.java:881)
	at javax.swing.plaf.basic.BasicTextUI.update(BasicTextUI.java:860)
	at javax.swing.JComponent.paintComponent(JComponent.java:780)
	at javax.swing.JComponent.paint(JComponent.java:1056)
	at javax.swing.JComponent.paintToOffscreen(JComponent.java:5210)
	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1579)
	at javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1502)
	at javax.swing.RepaintManager.paint(RepaintManager.java:1272)
	at javax.swing.JComponent._paintImmediately(JComponent.java:5158)
	at javax.swing.JComponent.paintImmediately(JComponent.java:4969)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:831)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:814)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:814)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:789)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:738)
	at javax.swing.RepaintManager.access$1200(RepaintManager.java:64)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1732)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
[... cut ...]
    </connections>
    <startup xmlns="http://jmri.org/xml/schema/auxiliary-configuration/startup-4-3-5.xsd">
        <perform xmlns="" class="apps.configurexml.PerformActionModelXml" name="apps.SystemConsoleAction" type="Action">
            <property name="systemPrefix" value=""/>
        </perform>
        <perform xmlns="" class="apps.configurexml.PerformScriptModelXml" name="C:/Program Files (x86)/JMRI/jython/InitTurnouts.py" type="ScriptFile"/>
    </startup>
</auxiliary-configuration>
```

closed 2021-04-14 by stale[bot]