JMRI/JMRI #3394: Did Roster sorting change?

bobjacobsen, 2017-04-15:
From a user comment:

> In the 'Roster Group Screen', the DCC address column does numeric sort but the next column over, 'Road Number' does not. Also, the 'Roster Group Association' doe not alpha sort the last column on the right, which used to sort by alpha, the group names. All of these used to sort in Version 3.3.  Something got changed.

geowar1, 2017-04-15:
FYI: I searched the JMRI sources for an AlphaNumeric comparator but didn't find one so I added one at: <JMRI/java/src/jmri/util/AlphanumComparator.java>. All the LayoutEditor combobox dropdown lists are now (4.7.2) sorted alphanumerically.

geowar1, 2017-04-15:
Actually I added this line:
        java.util.Arrays.sort(displayList, new AlphanumComparator());
to <JmriBeanComboBox.java>. So all JmriBeanComboBox's drop down lists are alphanumerically sorted. It would be awesome if all tables were sorted alphanumerically…

rhwood, 2017-04-15:
If a table column's class is a String it is sorted using the Java RuleBasedCollator (which does Locale aware alphabetic sorting). See the algorithm used by the Java TableRowSorter for how other classes are handled. Since the road number is a String, this may mean that 10 sorts before 2, which would need to be fixed.

On my phone,
Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

rhwood, 2017-04-16:
I'v just confirmed in JMRI/master, that, in the Rosters table, all columns except the RoadNumber are correctly sorting. The RoadNumber is *incorrectly* sorting such that it sorts:
```
1006
1021
12
1386
1409
152
1590
```
This is complicated by some numbering schemes that also used letters (pre-TOPS BR) and spaces in the road number (post-TOPS BR).

At least in macOS, the Roster Group Association table does not allow any sorting at all, and the last column is boolean (depicted as a check box), so I'm lost at what the concern with that table is.

Recommend we use the [alphanumeric-comparator library](https://github.com/sawano/alphanumeric-comparator) instead of rolling our own.

geowar1, 2017-04-16:
The one I added for JmriBeanComboBox'es I got from the web at: <http://www.davekoelle.com/alphanum.html>.
Looking at the sources, the [alphanumeric-comparator library](https://github.com/sawano/alphanumeric-comparator) ignores leading zeros such that "X007" is equal to "X07" or "X7"… OTOH it does appear to allow the use of a Collator (for international string comparison). (Just saying… I'm fine with ether…)

pabender, 2017-04-16:
> The one I added for JmriBeanComboBox'es I got from the web at: http://www.davekoelle.com/alphanum.html.
> Looking at the sources, the alphanumeric-comparator library ignores leading zeros such that [...]

Whether or not X007 X07 and X7 are different is dependent on the system.

If we are talking about system names in named beans, or DCC decoder addresses, then most of the time X007 X07 and X7 are going to be the same object.  ( in the Lenz case, for example XT7 is the same as XT07 and XT007. ).  

In the DCC decoder address case on NCE systems, 07 and 7 are NOT the same, and we can make an argument for addresses 1-127 ( short addresses ) should precede 01-0127 (long addresses ) ( someone please correct me if I have the long vs short designations reversed... ). To sort that correctly requires a custom comparator object, because NO comparator that isn't familiar with the application domain would get that right.

User names ( and roster names for the current discussion ) on the other hand I am not sure how we logically should sort X007 X07 and X7.  Regional differences don't really tell us how to sort that because each railroad 

Paul

dheap, 2017-04-17:
While 07 is used on NCE throttles to indicate you are wanting to select long address 7, this Is not reflected in the DCC Address column or in any decoders.

The only possible case of interest was if a user always used the loco address as the ID field, but in this case I think the user could cope with 7 and 07 sorting adjacent to each other.

Best practice with an NCE system is to never assign a short address to a loco because the command station assumes short addresses 1-127 are available for consist addresses and the Create Consist button on a throttle automatically assigns a consist address, starting at 127 and working down, using the highest free address in the command station's internal consist table.

Dave

> @pabender In the DCC decoder address case on NCE systems, 07 and 7 are NOT the same, and we can make an argument for addresses 1-127 ( short addresses ) should precede 01-0127 (long addresses ) ( someone please correct me if I have the long vs short designations reversed... ). To sort that correctly requires a custom comparator object, because NO comparator that isn't familiar with the application domain would get that right.

rhwood, 2017-05-30:
I have been experimenting with fixing this, but have not managed to get a comparator that, if one address is not a number, treats all other numbers correctly (that is, given the "numbers" 101, 10, V12, I have gotten a comparator to sort this as 10, 101, V12).

geowar1, 2017-05-30:
On May 30, 2017, at 6:35 AM, Randall Wood <notifications@github.com> wrote:
> I have been experimenting with fixing this, but have not managed to get a comparator that, if one address is not a number, treats all other numbers correctly (that is, given the "numbers" 101, 10, V12, I have gotten a comparator to sort this as 10, 101, V12).

I added AlphanumComparator to jmri.util and setup the jmri combobox dropdown lists to use it (in 7.4.3).

For what you want I’d write a pre-comparator that moves any non-digit prefix to the end before doing an alphanumeric compare.

-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

dheap, 2017-05-30:
@rhwood I have gotten a comparator to sort this as 10, 101, V12

I would see sorting numeric-only correctly and at the top, followed by all containing alphanumerics as acceptable. This seems to be what you suggest above.

I would see sorting all trailing numeric characters in mixed fields correctly as even better. "10, 101, A1, A2, V1, V2, V10"

closed 2021-04-14 by stale[bot]