JMRI/JMRI #3401: LayoutEditor has become verbose

bobjacobsen, 2017-04-17:
Some extra-verbose messages during startup were reported with the v4.7.3 builds.  Should these be debug level?

```
2017-04-16 17:35:36,752 layoutEditor.LayoutEditor             INFO  - Frame size: {w:1900, h:960} [AWT-EventQueue-0]
2017-04-16 17:35:37,273 layoutEditor.LayoutEditor             INFO  - Frame size: {w:1920, h:105} [AWT-EventQueue-0]
2017-04-16 17:35:37,276 layoutEditor.LayoutEditor             INFO  - Frame size: {w:1920, h:105} [AWT-EventQueue-0]
2017-04-16 17:35:37,277 layoutEditor.LayoutEditor             INFO  - Frame size: {w:1920, h:105} [AWT-EventQueue-0]
...
2017-04-16 17:35:37,515 layoutEditor.LayoutEditor             INFO  - Font size: 11.0 [AWT-EventQueue-0]
2017-04-16 17:35:37,551 layoutEditor.LayoutEditor             INFO  - jmri.jmrit.display.layoutEditor.LayoutEditor:B.toolBarThin is bottom [AWT-EventQueue-0]
2017-04-16 17:35:37,552 layoutEditor.LayoutEditor             INFO  - prefsMgr.prefsWindowLocation(jmri.jmrit.display.layoutEditor.LayoutEditor:B) is java.awt.Point[x=364,y=39] [AWT-EventQueue-0]
2017-04-16 17:35:37,552 layoutEditor.LayoutEditor             INFO  - prefsMgr.prefsWindowSize is(jmri.jmrit.display.layoutEditor.LayoutEditor:B) java.awt.Dimension[width=923,height=714] [AWT-EventQueue-0]
2017-04-16 17:35:37,552 layoutEditor.LayoutEditor             INFO  - prefsMgr.getProperty(jmri.jmrit.display.layoutEditor.LayoutEditor:B, "windowRectangle2D") is null [AWT-EventQueue-0]
2017-04-16 17:35:37,552 layoutEditor.LayoutEditor             INFO  - Frame size: {w:1920, h:105} [AWT-EventQueue-0]
```

```
2017-04-16 18:20:13,265 layoutEditor.LayoutBlock              INFO  - From
71, no routes returned for getRouteByNeighbour(74) [main]
2017-04-16 18:20:13,265 layoutEditor.LayoutBlock              INFO  - From
71, no routes returned for getRouteByNeighbour(74) [main]
2017-04-16 18:20:13,359 layoutEditor.LayoutBlock              INFO  - From
71, no routes returned for getRouteByNeighbour(74) [main]
2017-04-16 18:20:13,360 layoutEditor.LayoutBlock              INFO  - From
71, no routes returned for getRouteByNeighbour(74) [main]
2017-04-16 18:20:13,362 layoutEditor.LayoutBlock              INFO  - From
77, no routes returned for getRouteByNeighbour(87) [main]
```
(those also seem redundant with each other, at least the first ones)

```
2017-04-16 13:11:38,007 layoutEditor.LayoutEditor             INFO  - jmri.jmrit.display.layoutEditor.LayoutEditor:Monashee Pacific Signal Panel.toolBarThin is null [AWT-EventQueue-0]
2017-04-16 13:11:38,888 layoutEditor.LayoutEditor             INFO  - prefsMgr.prefsWindowLocation(jmri.jmrit.display.layoutEditor.LayoutEditor:Monashee Pacific Signal Panel) is java.awt.Point[x=0,y=1] [AWT-EventQueue-0]
2017-04-16 13:11:38,890 layoutEditor.LayoutEditor             INFO  - prefsMgr.prefsWindowSize is(jmri.jmrit.display.layoutEditor.LayoutEditor:Monashee Pacific Signal Panel) java.awt.Dimension[width=0,height=0] [AWT-EventQueue-0]
2017-04-16 13:11:38,891 layoutEditor.LayoutEditor             INFO  - prefsMgr.getProperty(jmri.jmrit.display.layoutEditor.LayoutEditor:Monashee Pacific Signal Panel, "windowRectangle2D") is null [AWT-EventQueue-0]
```

JLang1951, 2017-04-17:
I get the same console messages when I open a LE panel. I've been ignoring the "no routes returned for getRouteByNeighbour" messages because I can go to one of the blocks that it is complaining about and the block routing info is correct hence I have no problem with signalmast logic.
My system console:
[JDL System Console 04-17-17.txt](https://github.com/JMRI/JMRI/files/926479/JDL.System.Console.04-17-17.txt)
It would be nice to not have them there if there isn't a problem.

John Lang

geowar1, 2017-04-17:
You may assign this one to me…

geowar1, 2017-04-25:
https://github.com/JMRI/JMRI/pull/3431

geowar1, 2017-04-27:
https://github.com/JMRI/JMRI/pull/3431 has been merged into JMRI/master…
Coming soon to a build near you!

closed 2017-05-07 by bobjacobsen