JMRI/JMRI #3407: Fresh install of JMRI generates LoadLibrary error 126

AndrewInEssex, 2017-04-18:
Hi, I've just installed JMRI and if I try and run DecoderPro or PanelPro I receive a LoadLibrary failed with error 126 message. I have successfully installed JMRI on another PC, but I can't get it to run on this one. I think the error is to do with a missing DLL or module within a DLL (so it could be an out of date DLL), of course the problem is, I've no idea what DLL it's having a problem with. Any help greatly appreciated.

bobjacobsen, 2017-04-18:
A bit more information, please. 
- Which JMRI version?
- Which Windows version?
- When does it fail? E.g. During install, during startup, while setting presences, some other time?

Thanks.

AndrewInEssex, 2017-04-18:
Hi Bob, thanks for the reply; it's version 4.6, I'm running Windows 10. It fails during start-up.

closed 2017-04-18 by AndrewInEssex