JMRI/JMRI #3410: Sprog Nano does't do program track, but we try.

KenC57, 2017-04-19:
So If JMRI knows it is a nano, we shouldn't offer the programming track method.
-ken c
From the Soundtraxx list:

Got in touch with Bill at Decoder Pro. Apparently the Sprog Nano does not do service mode, so it can't read the decoders, hence the "no locomotive detected" error message. Somehow I missed that detail in the product literature. He helped me configure the Nano to correct the issue of not being able to control the motor and get it set up for programming in ops mode, so I can now program the decoders and run the loco to test the settings. I can't read what's already on them, but I can at least program them. Looks like I'll be upgrading to the Sprog 2 so I can read what's already on the decoders I've already got installed. (I haven't a clue what values I have for most of the CVs on them.)

bobjacobsen, 2017-04-19:
There are two connection types listed for the "SPROG DCC" manufacturer type:

- SPROG Nano Command Station (jmri.jmrix.sprog.sprognano package)

- Pi-SPROG Nano Command Station (jmri.jmrix.sprog.pi.pisprognano package)

Should both of them be set to not have a service mode programmer? If just one, which one?

pabender, 2017-04-19:
Sent from my iPad

> On Apr 19, 2017, at 12:58 AM, Bob Jacobsen <notifications@github.com> wrote:
> 
> There are two connection types listed for the "SPROG DCC" manufacturer type:
> 
> SPROG Nano Command Station (jmri.jmrix.sprog.sprognano package)
> 
> Pi-SPROG Nano Command Station (jmri.jmrix.sprog.pi.pisprognano package)
> 
> Should both of them be set to not have a service mode programmer? If just one, which one?
> 

It should be both of them.

Paul

KenC57, 2017-04-22:
It looks like it is not as clear on setting which interfaces would do ops vs service mode as I've seen on other systems. Given the history of Sprog, I'm hoping Andrew or somebody can pick up on this.

KenC57, 2017-04-22:
I was comparing this with a NCE SB5 system, which also only does OPS mode, and saw a difference I don't understand. The Sprog defaults shows the service mode tied to 'internal' while the SB5 show it tied to the NCE. Opening the single CV programmer in the Sprog shows OPS and Service modes, but NCE correctly only shows OPS. So it seems confused.

closed 2021-04-14 by stale[bot]