JMRI/JMRI #3411: Managers provide listeners insufficient information

rhwood, 2017-04-19:
Managers that allow the addition or deletion of managed objects (mostly NamedBeans) need to include added or deleted object in notifications to IndexedPropertyChangeListeners. Doing this will allow monitoring objects (scripts or web servers) to act on those addition or deletion events without having to maintain their own copy of the list of managed objects.

bobjacobsen, 2017-04-20:
For background on this, see the tutorial discussions on [indexed properties](https://docs.oracle.com/javase/tutorial/javabeans/writing/properties.html) and on [indexed property change reporting](https://www.java-tips.org/java-se-tips-100019/28-java-beans/1769-reporting-indexed-property-changes-in-beans.html).

The canonical example of an indexed property get and set for the indexed array as a whole, and for individual elements:
```
public int[] getTestGrades() {...}
public void setTestGrades(int[] tg) {...}
public int getTestGrades(int index) {...}
public void setTestGrades(int index, int grade) {...}
```

Our `Manager` classes don't provide those.  Instead of access by index, they provide access via user name and system name lookup, with `[]` and `List` access to those names, and `List` access to the `NamedBeans` themselves.
```
    public NamedBean getBeanBySystemName(@Nonnull String systemName);
    public NamedBean getBeanByUserName(@Nonnull String userName);

    public String[] getSystemNameArray();
    public List<String> getSystemNameList();

    public List<NamedBean> getNamedBeanList();
```

To fully implement an indexed parameter, we'd add
```
public NamedBean[] getNamedBeans();
public void setNamedBeans(NamedBean[] nb);
public NamedBean getNamedBeans(int index);
public void setNamedBeans(int index, NamedBean nb);
```

Because of the way we inherit into specific classes, we'd also have to (or at least should) provide specific implementations. They'd look like:
```
public Turnout[] getTurnouts();
public void setTurnouts(Turnout[] nb);
public Turnout getTurnouts(int index);
public void setTurnout(int index, Turnout nb);
```

If this was just a typing issue, it could be handled via generics, hence centrally implemented.  But this also involves changes to the method names, which requires that it be done for each type; that's a pain, but manageable.

Should we do that as part of this?

Aside from the work, it raises a question of semantics.  What does this do?
```
manager.setNamedBeans(new NamedBean[] {fooBean, barBean});
```

or this

```
NamedBean[] n = manager.getNamedBeans();
n[0] = someBean;
```
(which could also be written in terms of our existing `getNamedBeanList()` call, of course)

Java doesn't have a `const` concept.

bobjacobsen, 2017-04-20:
> Managers that allow the addition or deletion of managed objects (mostly NamedBeans) need to include added or deleted object in notifications to IndexedPropertyChangeListeners.

Separately from the architectural issues in the earlier comment above (which aren't really required for the specific request, they're just good to consider doing at the same time), there's a question of how to do notification around addition and deletion.  I haven't seen how to do that for [`IndexedPropertyChangeEvent`](https://docs.oracle.com/javase/8/docs/api/java/beans/IndexedPropertyChangeEvent.html) notifications. Is there an example of that somewhere? 

Thanks

rhwood, 2018-03-16:
OP is no longer concerned.

PR #7076 (2019-07-01): Use NamedBeans, not Strings, in NamedBean selectors
This replaces `jmri.swing.util.JmriBeanComboBox` with `jmri.swing.NamedBeanComboBox`. The difference between the two is that JmriBeanComboBox uses various display formats of the names of NamedBeans as the items in its model, while NamedBeanComboBox uses specific types of NamedBeans as the items in its model. As a result, objects using a NamedBeanComboBox can directly get directly get the correct type of NamedBean from the NamedBeanComboBox selection.

- [x] Add a comparator for NamedBeans based on User Name (for sorting User Names and dealing [... cut]
commit 5bc15fc8 (2019-07-01): feat: Fire indexed property change when adding/removing beans

This does the same thing as the datalisteners API, but in a standard JavaBeans-compliant manner.
Fixes #3411
closed 2019-07-01 by bobjacobsen