JMRI/JMRI #3424: ESU Ecos systems cannot be used in headless mode

pabender, 2017-04-22:
This was discovered while writing tests.

The ESU Ecos System's EcosLocoAddressManager produces several JDialog boxes during execution.  These dialog boxes cannot be displayed on a headless system.

The EcosSystemConnectionMemo creates the EcosLocoAddressManager.

The EcosLocoAddressManager creates 3 JDialogs to confirm if:
1) The user wants to remove a locomotive from the Ecos in response to a property change event from the JMRI roster
2) The user wants to add a locomotive to the Ecos in response to a property change event from the JMRI roster
3) The user wants to remove a locomotive from the JMRI roster if a corresponding entry is not found in the Ecos loco database.

A remove dialog is also shown during shutdown.

Because any changes to the roster may invoke a dialog box if an Ecos system is configured, Ecos systems on a headless system will trigger a headless exception.

As a temporary solution to getting the tests running, the tests that use the EcosLocoAddressManager have been disabled in headless mode.  In the longer term, a redesign of the EcosLocoAddressManager, to remove the direct creation of the JDialogs.

KenC57, 2017-04-22:
I'm curious about possible solution methods for this issue. Would it be
reasonable to have a setting that is saved that would give a fixed answer to
those questions?? So when headless it already has the yes/no response to the
questions. Would that work?? I'm asking just to get the idea of how things
like this could/should be handled. Now when running with a head, it could
use that as the default selection so you only had to hit entry to the dialog
box.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2017-04-22:
To be a fixed answer, they could be made options (or advanced options) in the preferences for the Ecos connection. 

But they sound like they're roster-change by roster-change selections. In that case, the answer might be not a rethink of how to get answer to the questions, but whether those are the right questions.  

For example, you could add a parameter to the roster entry that's "Lives in eCos or lives in JMRI", and have a specialization of Roster (or RosterEntry) that just handles the behavior you want for that locomotive without having to ask each time.  Then give the ecOs menu  a selection for an "ecoS Roster setup" tool for managing the roster(s) in a more detailed way.

pabender, 2017-04-22:
Right, there are other solutions, and I'm not sure what the right one is.

This is code that @kevin-dickerson wrote.  I think the intention is to keep the JMRI roster and the Ecos roster synchronized.  

What doesn't really make sense to me is why would I want to delete something from the JMRI roster just because it disappeared from the Ecos, which seems to be the intention of one of the questions.

bobjacobsen, 2017-04-23:
> What doesn't really make sense to me is why would I want to delete something from the JMRI roster just because it disappeared from the Ecos, which seems to be the intention of one of the questions.

IIRC, originally one roster (JMRI?) was considered a "master", and was kept pushed to the other device (eCos?).  But then there was a long discussion about having the other (eCos?) be the master, hence the questions being added.

Some pointers: #423 #2905 and [jmriusers topic 134949](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/134949)

pabender, 2017-04-23:
Thanks @bobjacobsen.

Perhaps @kjlisby can take a look at what it will take to make this function in a headless environment" since he is familiar with the code.

kjlisby, 2017-04-23:
I am not too familiar with this particular part of the ECOS code. But I have been correcting an error therein a while ago (something with those questions keep coming in a circle, as I remember it).

But I do own an ECOS and can therefore try things out. 

As far as I can see, the ECOS preference pane already makes it possible to configure it in a way that no questions should be asked. So is the problem that these configuration options do not work as they should? I have not been spending time on testing it yet.

pabender, 2017-04-23:
Karl,

Sent from my iPhone

> On Apr 23, 2017, at 5:32 AM, kjlisby <notifications@github.com> wrote: 
> As far as I can see, the ECOS preference pane already makes it possible to configure it in a way that no questions should be asked. So is the problem that these configuration options do not work as they should? I have not been spending time on testing it yet.
> 
 I am even less familiar with this part of the code than you are, but that gives me clue I need to make the tests work in headless mode.

It looks like the problem is really that the preferences that cause the EcosLocoAddressManager to display the dialog messages default to allowing the questions to be asked.  

In a headless environment, the dialogs cannot be displayed ( because there is no GUI to display them on ).

Paul

pabender, 2017-04-26:
So it turns out that:

1) One of the JDialog objects created by EcosLocoAddressManager wasn't being protected by a preference check.  (one of the others was protected by catching a headless exception.)

2) The constructor for EcosLocoAddressManager creates a jmri.jmrit.beantable.ListedTableFrame object if one doesn't exist already.  I've added a try/catch for a headless exception around this bit of code.  The purpose of this particular bit of code is to register a new table with the GUI.  I suspect this registration can probably be moved into the jmri.jmrix.ecos.swing package tree.

I'm not sure these are the only things that have been causing headless issues here...

closed 2021-04-14 by stale[bot]