JMRI/JMRI #3425: AppVeyor not detecting ERROR-level messages?

bobjacobsen, 2017-04-22:
From a recent AppVeyor run:
```
    [junit] Testsuite: apps.tests.AllTest
    [junit] AL lib: (EE) MMDevApiOpenPlayback: Device init failed: 0x80070490
    [junit] Tests run: 10698, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 652.226 sec
    [junit] Tests run: 10698, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 652.226 sec
    [junit] 
    [junit] ------------- Standard Output ---------------
    [junit] Using org.netbeans.jemmy.drivers.DefaultDriverInstaller driver installer
    [junit] 
    [junit] 
    [junit] ------------- ---------------- ---------------
    [junit] ------------- Standard Error -----------------
    [junit] ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main] jmri.util.exceptionhandler.UncaughtExceptionHandler.?()
    [junit] java.lang.NullPointerException
    [junit] 	at jmri.jmrix.dccpp.dccppovertcp.ClientRxHandler.run(ClientRxHandler.java:127)
    [junit] WARN  - line not supported: interface SourceDataLine supporting format PCM_UNSIGNED 11200.0 Hz, 8 bit, mono, 1 bytes/frame,  [main] jmri.jmrit.Sound.playSoundBuffer()
    [junit] WARN  - sendMessage: Exception: port warn java.io.IOException: Read end dead [main] jmri.jmrix.AbstractMRTrafficController.?()
    [junit] WARN  - Timeout on reply to message: 21 24 05 consecutive timeouts = 0 [main] jmri.jmrix.AbstractMRTrafficController.?()
    [junit] WARN  - Found unknown Windows version: Windows Server 2012 R2 [main] java.games.input.DefaultControllerEnvironment.?()
    [junit] Warning: org.jdom2.transform.JDOMSource$DocumentReader: http://javax.xml.XMLConstants/property/accessExternalDTD
    [junit] Warning: org.jdom2.transform.JDOMSource$DocumentReader: http://www.oracle.com/xml/jaxp/properties/entityExpansionLimit
    [junit] WARN  - Initialised Null audio system - no sounds will be available. [main] jmri.jmrit.audio.NullAudioFactory.?()
    [junit] ------------- ---------------- ---------------
    [junit] Exception while removing reference.
[INFO] Executed tasks
```

Note there was an ERROR-level message, but the run shows only passed tests.  Normally, ERROR should be caught and result in a JUnit failure.

bobjacobsen, 2017-04-22:
This might not be an AppVeyor thing, but it shows up most clearly there.

The corresponding Travis run had:
```
-------------------------------------------------------
 T E S T S
-------------------------------------------------------
Running jmri.jmrit.progsupport.ProgDeferredServiceModePaneTest
Using org.netbeans.jemmy.drivers.DefaultDriverInstaller driver installer
Tests run: 1, Failures: 0, Errors: 0, Skipped: 1, Time elapsed: 0.616 sec - in jmri.jmrit.progsupport.ProgDeferredServiceModePaneTest
Running jmri.jmrit.operations.locations.schedules.ScheduleOptionsActionTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 1, Time elapsed: 0 sec - in jmri.jmrit.operations.locations.schedules.ScheduleOptionsActionTest
Running jmri.jmrix.dccpp.DCCppTurnoutReplyCacheTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.148 sec - in jmri.jmrix.dccpp.DCCppTurnoutReplyCacheTest
```
(followed by over 5000 lines of similar output, which is not such a good thing, then)
```
Running jmri.jmrix.pi.RaspberryPiAdapterTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 sec - in jmri.jmrix.pi.RaspberryPiAdapterTest
ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [z21.Z21XPressNetTunnel sourceThread] jmri.util.exceptionhandler.UncaughtExceptionHandler.uncaughtException()
java.lang.NullPointerException
	at jmri.jmrix.roco.z21.Z21Message.<init>(Z21Message.java:40)
	at jmri.jmrix.roco.z21.Z21XPressNetTunnel.message(Z21XPressNetTunnel.java:211)
	at jmri.jmrix.roco.z21.Z21XPressNetTunnel.run(Z21XPressNetTunnel.java:79)
	at java.lang.Thread.run(Thread.java:745)
Results :
Tests run: 24517, Failures: 0, Errors: 0, Skipped: 898
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 11:25 min
[INFO] Finished at: 2017-04-21T04:54:34+00:00
```

So an ERROR there wasn't seen either.  

Are we somehow running tests with separate Log4J setups now?

bobjacobsen, 2017-04-22:
Finally, the corresponding Travis graphical run looked like the AppVeyor run:
```

[jacoco:coverage] Enhancing junit with coverage
    [junit] Running apps.tests.AllTest
    [junit] Testsuite: apps.tests.AllTest
    [junit] Tests run: 10698, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 767.317 sec
    [junit] Tests run: 10698, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 767.317 sec
    [junit] 
    [junit] ------------- Standard Output ---------------
    [junit] Using org.netbeans.jemmy.drivers.DefaultDriverInstaller driver installer
    [junit] 
    [junit] 
    [junit] ------------- ---------------- ---------------
    [junit] ------------- Standard Error -----------------
    [junit] ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main] jmri.util.exceptionhandler.UncaughtExceptionHandler.?()
    [junit] java.lang.NullPointerException
    [junit] 	at jmri.jmrix.dccpp.dccppovertcp.ClientRxHandler.run(ClientRxHandler.java:127)
    [junit] WARN  - line not supported: interface SourceDataLine supporting format PCM_UNSIGNED 11200.0 Hz, 8 bit, mono, 1 bytes/frame,  [main] jmri.jmrit.Sound.playSoundBuffer()
    [junit] Warning: org.jdom2.transform.JDOMSource$DocumentReader: http://javax.xml.XMLConstants/property/accessExternalDTD
    [junit] Warning: org.jdom2.transform.JDOMSource$DocumentReader: http://www.oracle.com/xml/jaxp/properties/entityExpansionLimit
    [junit] WARN  - Initialised Null audio system - no sounds will be available. [main] jmri.jmrit.audio.NullAudioFactory.?()
    [junit] ------------- ---------------- ---------------
ci-test-travis-coveragecheck:
[jacoco:report] Loading execution data file /home/travis/build/JMRI/JMRI/jacoco.exec
[jacoco:report] Writing bundle 'AntTestReporting' with 6706 classes
[INFO] Executed tasks
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
```

rhwood, 2017-04-22:
We are trying to catch the ERROR messages at each Log4JFixture.tearDown call, as well as at the end of the tests.

Note that logging each test as it happens is valuable--its caught which test is not completing in headless mode in #3340, which is output not previously available (and still not available in the GUI failures, or else we would not be encountering unexplainable test timeouts in that mode).

pabender, 2017-04-22:
@bobjacobsen Part of the reason we're not catching some of those error messages is due to threading.  I'm not sure how to force the test to wait until all threads created by the test have terminated.  In many cases, the test doesn't even know it actually created a thread.

As for the "5000 lines of nearly identical text", I agree with @rhwood that there is some value there, but I also agree with @bobjacobsen that the output is somewhat problematic.  The real issue I have is that the output makes it more difficult to find which test actually failed (though I noticed in the recent builds there is a summary at the bottom... but I still have to scroll through all of the package output to find that summary....)

bobjacobsen, 2017-05-08:
I understand the value in catching "unexplainable test timeouts". Are we still seeing those? If not, can we drop back to the prior (much less verbose) method?

If we can't go back to running these individually (and I'm not convinced of the relative value of that yet), can we at least get rid of all the lines like:
```
Running jmri.jmrit.operations.locations.schedules.ScheduleOptionsActionTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 1, Time elapsed: 0 sec - in jmri.jmrit.operations.locations.schedules.ScheduleOptionsActionTest
```
that fill the log?

rhwood, 2017-05-08:
Yes, we are still seeing them. From the comment in #3471 

> Travis linux graphics timed out three times; runs OK on my desktop, so not sure where problem lies.

these need to be extended to the GUI tests to really figure these problems out.

rhwood, 2020-07-20:
This is OBE as we no longer use AppVeyor and the replacement CI is different enough this does not occur in it.

closed 2020-07-20 by rhwood