JMRI/JMRI #3435: Broken LocoNet Stats window starting in 4.7.3

bobjacobsen, 2017-04-26:
(From a [jmri-developers post by Dick Bronson](https://sourceforge.net/p/jmri/mailman/message/35808598/) this date) 

Someplace between version 4.7.2 and 4.7.3 something has broken the 'Monitor LocoNet Stats' functionality. (Hopefully this list accepts attached images) When used with a LocoBuffer-USB the normal operation is to display the "Version: Breaks: Errors:" immediately. After the problem was introduced, it now shows raw data unless you click on the 'Update' button, and then it gives bogus information for the 'Version:' entry.

The Window title has also changed, so something major has changed and been broken. It is still broken in the current 4.7.4ish build.

[image]

[image]

devel-bobm, 2017-04-26:
I expect this is an unintended behavior related to #3335.

devel-bobm, 2017-04-27:
Changed reporting of LocoBuffer version number to hexadecimal.  Now launching the initial query of status on the Layout thread.  Finished the partially-completed I18N work.

devel-bobm, 2017-04-28:
See also #3441 .

devel-bobm, 2017-05-01:
Dick reports that build artifacts from #3439 work fine on Win10, but incurs a 2-second delay on Linux - message is sent to LocoNet 2 seconds after tool is started, and message is sent 2 seconds after "update" button is pushed.

devel-bobm, 2017-05-02:
I've tried the build on a Ubuntu Linux (32-bit) machine, and did not notice a 2-second delay there.

devel-bobm, 2017-05-05:
Debugging Dick's "delayed by 2 seconds" issue is stalled account cannot reproduce.  Ideas?

bobjacobsen, 2017-05-06:
If that 2-seconds wait is long enough, try to get a thread dump with ^-\

That might give a clue if something is blocked or not started.

devel-bobm, 2017-05-06:
Is that a three-key combination of <ctrl>, <minus>, and <backslash>, or a two-key combination of <ctrl> and <Backslash>?

bobjacobsen, 2017-05-06:
> Is that a three-key combination of , , and , or a two-key combination of and ?

On most Linux shells ctrl-backslash does a thread dump of any child Java process. 

Bob

bobjacobsen, 2017-05-25:
@devel-bobm: Do we still have a mysterious 2-second delay on Linux for this? Or can I close it?

devel-bobm, 2017-05-25:
My understanding is that Dick still has the problem.  I cannot reproduce.
 
I'll remind Dick about the thread dump request...

devel-bobm, 2017-05-26:
Dick reports that he will not be able to follow up until next week some time.

devel-bobm, 2017-09-21:
@RR-CirKits - any update on this?  Do you still have a mysterious 2-second delay?  With the most-recent test release (4.9.4)?

RR-CirKits, 2017-09-21:
Bob,
The response seems pretty immediate now. I don't remember the exact 
conditions where I saw it for comparison.

Dick :)

devel-bobm, 2017-09-23:
Closing the issue.  If the problem recurrs, please reopen this issue or a new one.

closed 2017-09-23 by devel-bobm