JMRI/JMRI #3438: Double-slip blank on Layout web page

geowar1, 2017-04-27:
Double- (and single-?) slips (and their control circles) don't appear on my layout when displayed via the web server.

To reproduce: Create a new Layout; add double-slip, 4 end bumpers & connect with track segments; start web-server… 
Expected behavior: I expect to see the 4 end bumpers, 4 track segments and the double slip.
Actual behavior: The double-slip and track segments don't appear.

:(

mstevetodd, 2017-04-27:
None of the multi-turnout widgets (like the slips) are implemented (yet) in the panel.js client-side panel code. Workarounds include using multiple turnouts, or using the older /frame page instead of /panel.

geowar1, 2017-04-28:
I added single/double slip drawing code to the panel.js file's $drawTurnout($widget) function… didn't work… turns out the widgets aren't even being created for slips… For a non-slip turnout I see these: <div id="widget-206" class="layoutturnout drawn rotatable LayoutTurnoutXmlturnout clickable " title="LT37:Main 3/Yard,27...30" alt="LT37:Main 3/Yard,27...30" style="position: absolute; left: 186px; top: 426px; z-index: 3; width: 12px; height: 12px;"></div> but nothing for slips… Little help?

geowar1, 2017-05-04:
FYI: Progress made… see: <https://github.com/JMRI/JMRI/pull/3449#pullrequestreview-35654865>

geowar1, 2017-05-05:
Ok, the PR for this has been merged… so, coming soon to a build near you… (or not).

geowar1, 2017-05-05:
https://github.com/JMRI/JMRI/pull/3449

bobjacobsen, 2017-05-07:
Can we close this?

geowar1, 2017-05-07:
Yes, I believe so… (Working solid on my system… YMMV ;-)

closed 2017-05-07 by bobjacobsen