JMRI/JMRI #3443: Missing/wrong case bean name, hard to find

KenC57, 2017-04-29:
I have a panel that had mixed case for the signal head names. Latest code fails to load the panel due to
NamedBeanHandleManager:getNamedBeanHandle() calling Objects.requireNonNull() for the bean and name. But error message gives no clue which name or bean was the failing item. I only fixed my panel by adding a log.info() line in a few likely suspects. It then gave me a name that was all upper case while the rest of the panel was using camel case for the name. I then fixed it. Normal users would not be able to do this. Also the 'Check XML' worked fine.

Looking for clues how to improve this. Would a try/catch around the requireNonNull() work to allow a better error message than a stack dump?? The attached file has a signal name SyrJctSouth2 that was all upper case in a watchedsignal element.

[SMRC-SouthLayout69_xml.txt](https://github.com/JMRI/JMRI/files/966114/SMRC-SouthLayout69_xml.txt)

-ken c

rhwood, 2017-04-29:
Wrapping RequireNonNull defeats the intended purpose of forcing a failure if null is passed as an argument (i.e. null arguments to the method become acceptable). You can wrap the call to the method that calls RequireNonNull instead.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

bobjacobsen, 2017-04-29:
Here's the traceback I see when loading that file:
```
     [java] 2017-04-29 07:44:28,923 configurexml.ErrorHandler             ERROR - Unexpected error (Exception) while load(/Users/jake/Downloads/SMRC-SouthLayout69_xml.txt) in adaptor of type jmri.jmrit.blockboss.configurexml.BlockBossLogicXml Exception: java.lang.NullPointerException: bean must be nonnull
     [java] See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
     [java] java.lang.NullPointerException: bean must be nonnull
     [java] 	at java.util.Objects.requireNonNull(Objects.java:228)
     [java] 	at jmri.NamedBeanHandleManager.getNamedBeanHandle(NamedBeanHandleManager.java:57)
     [java] 	at jmri.jmrit.blockboss.BlockBossLogic.setWatchedSignal1(BlockBossLogic.java:326)
     [java] 	at jmri.jmrit.blockboss.configurexml.BlockBossLogicXml.load(BlockBossLogicXml.java:264)
```

Reading up, it's trying to load BlockBossLogic (the historical name for SSL). When it tries to handle setting the "watched signal", i.e. the next signal in advance, it can't resolve a NamedBean reference.

There are a couple things here that can be improved:

- [ ] The right way to make sure this can't happen in normal use is to have the SSL GUI use some better widgets for entering bean names.  AFAIK, right now they're just a text field, and it might not match up. Better would be to use `jmri.util.swing.JmriBeanComboBox` or the "pick list" support from `jmri.jmrit.picker.*`, see [the programmer info on the Names page](http://jmri.sourceforge.net/help/en/html/doc/Technical/Names.shtml).

The right way to catch it in existing files, which will also help with scripts, etc, is to armor the `BlockBossLogic` call (including what it invokes) so that they at least log something useful. It looks like there was an attempt to do something like that, as the code is:
```
        if (name == null || name.equals("")) {
            watchedSignal1 = null;
            return;
        }
        watchedSignal1 = nbhm.getNamedBeanHandle(name, InstanceManager.getDefault(jmri.SignalHeadManager.class).getSignalHead(name));
        if (watchedSignal1.getBean() == null) {
            log.warn(rb.getString("Signal_") + name + rb.getString("_was_not_found!"));
        }
```
but the `if (watchedSignal1.getBean() == null)` isn't invoked because of an exception thrown inside [NamedBeanHandleManager#getNamedBeanHandle](http://jmri.org/JavaDoc/doc/jmri/NamedBeanHandleManager.html#getNamedBeanHandle-java.lang.String-T-) due to being called with a null bean reference.

By the time it gets to passing a bean reference, there's no way to back up to the name, so this is the right behavior for `NamedBeanHandleManager#getNamedBeanHandle`.  Nor will putting a try-catch around that behavior do anything useful, as there's no information by that point.

The real problem is the embedded call to `getSignalHead(..)` in:
```
        watchedSignal1 = nbhm.getNamedBeanHandle(name, InstanceManager.getDefault(jmri.SignalHeadManager.class).getSignalHead(name));
```
Should that fail to find the signal and return null, there's nothing dealing with that.  So the right solution for the local problem is

- [ ] Armor the code (multiple places) in BlockBossLogic against not finding a reference with e.g.
```
        watchedSignal1 = nbhm.getNamedBeanHandle(name, InstanceManager.getDefault(jmri.SignalHeadManager.class).getSignalHead(name));
```
being replaced by
```
        SignalHead sh = InstanceManager.getDefault(jmri.SignalHeadManager.class).getSignalHead(name);
        if (sh == null) {
                log.warn("Invalid SignalHead name \"{}\" in for Watched Signal 1", sh);
                return;
        }
        watchedSignal1 = nbhm.getNamedBeanHandle(name, sh);
```

bobjacobsen, 2017-04-29:
One more thing on this:  In theory, it's straight-forward to have the XML Schema check at the start of reading the file catch this error.  Schema validation can check cross-references.  The error message is a bit obscure, but it does give the line in the file where the problem reference is.

But that only works if the entire contents of the layout information is in a single file: the cross-reference check has to find the definition of the NamedBean in that same XML document.  Lots of people, including me, have divided the layout configuration information into multiple files.

Instead of building it in the Schema check, another choice would be to 
- [ ] have common methods for reading and checking a system name or user name from the XML file.  That then centralizes the warnings into one place, which helps raise their quality.

I'm happy to work on that, as time is available.

KenC57, 2017-04-29:
I'm going to go after armoring the code. So tracking places that call getSignalHead() and protect things there.

bobjacobsen, 2017-04-29:
> I'm going to go after armoring the code. So tracking places that call getSignalHead() and protect things there.

That's fine, but it won't fix this problem.  The problem was in the ConfigureXml code that will eventually _set_ the values, before it gets to that point.

KenC57, 2017-04-29:
Bob J, you mean while the xml file is being read, protect it there?? So while reading the SSL (bossblock)?? Granted I'm thinking providing meaningful error when it finds it later too is the belt-suspenders route.

bobjacobsen, 2017-05-01:
I'm working on some common code in AbstractNamedBeanManagerConfigXML to make problems during read easier (e.g. automatic if the common routines are used) to fix.  See my `xml-name-support` branch.  @KenC57: If you haven't started work on the SSL code, I can use that as a first example.

KenC57, 2017-05-01:
Nope, didn't start yet. Have final papers this week, then finals. So hands
have become busy again for two weeks.

-ken c

closed 2021-04-14 by stale[bot]