JMRI/JMRI #3454: FileUtilTest may fail if ProfileUtilTest is run before it

rhwood, 2017-05-03:
The FileUtilTest tests appear to have an unexpected interaction with the ProfileUtilsTest such that four tests in FileUtilTest fail if run after the ProfileUtilsTest (but only sometimes).

It seems that when Travis CI fails with FileUtilTest failures, it is failing with four tests looking for a temporary directory with a name beginning with "dest", which is created by a ProfileUtilsTest.

I am not sure what the fix is.

rhwood, 2017-05-03:
Sample failure:
```
Failed tests: 
  FileUtilTest.testFindExternalFilename:327 expected:<file:/tmp/dest1620818003431338534/267281db-43fd-4f68-9f34-952594ca7d56> but was:<null>
  FileUtilTest.testFindURIPath:285 expected:<file:/tmp/dest1620818003431338534/05ad0dc2-871e-4314-8cce-d775214608a1> but was:<null>
  FileUtilTest.testFindURIPathLocation:299 expected:<file:/tmp/dest1620818003431338534/836d6374-dc84-4893-b0f8-aa66f6ccb4b5> but was:<null>
  FileUtilTest.testFindURIPathSearchPaths:313 expected:<file:/tmp/dest1620818003431338534/d84d7bcd-3f34-42a5-8484-a9b0ed0c9db8> but was:<null>
```

rhwood, 2017-05-03:
A possible fix for this has been merged in #3448, monitoring for now.

geowar1, 2017-05-07:
I'm still getting those four FileUtilTest failures occasionally. See: <https://github.com/JMRI/JMRI/pull/3467>

rhwood, 2017-05-24:
Haven't seen this recently, so closing. Reopen if observed again.

closed 2017-05-24 by rhwood