JMRI/JMRI #3459: SE8c looping behavior

bobjacobsen, 2017-05-04:
This may be a side effect of PR #3363

From a [jmriusers thread](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/138146):

>We are using stock 4.7.3 without the added PR3363. The problem did not exist on 4.6.0 and earlier. This may not be the same problem as the original post. In the op session yesterday we had 7 computers all Window 10 and connected to Loconet. The problem was only reveled in a multi computer environment. We edited all preferences to make the Loconet interface have only a single message (no retry) and disabled all Signal Repeaters (in Mast Table Tool) with no effect.
>
>Problem: SE8c Turnout messages will sometimes repeat indefinitely Even after unplugging the Loconet cable from the source computer, the message repeat will sometimes start up again and can only be stopped by unplugging the USB cable from the LocoBuffer. The runaway message repeating continues even with the Digitrax command station disconnected. Computers with 4.6.0 were sometimes the message originator, but never the repeater of the messages. Computers with 4.7.3 will repeat the SE8c message sequence even though they may not have that SE8c driver Logix enabled and there were no Signal Repeaters. The repeater computers do have the Mast Table definitions for the bogus repeat messages, but no Logix or functions to cause the generation of the relevant Signal Mast Turnout messages. 
>
>In conclusion, the original signal driver sequence can come from another computer and the endless-repeater computer may only repeat. The offending SE8c messages can originate from any computer on any version of JMRI, but the runaway repeating was always a computer with 4.7.3. During the runaway the Loconet is effectively shut down by saturation from the repeat messages.

devel-bobm, 2017-05-06:
@bobjacobsen, see the comment I have added to #3363.  

If my understanding is correct, I would think that the logic change in #3363 would (unintentionally) reduce the number of times that the head is updated because of a LocoNet message.  If this logic change is responsible for the "LocoNet storm", then this logic would actually _reduce_ the amount of LocoNet traffic, not increase it.

I wonder if instead the thread change from "normal" to "daemon" is affecting the problem.

And I wonder if the problem is buried in hardware-agnostic signal logic code somewhere.  In this case, I would think that the problem would appear with other signal systems.

silverailscolo, 2017-05-07:
Not sure if this is related, but might help to locate source of the issue:
Recognized this looping in a simple, stand alone setup with LocoNet Simulator, 2 shsm signal masts, 2 Se8c signal heads, 4 turnouts **and** a panel. As soon as the panel file is loaded, LT111 and head drv5A1 toggle between 2 states as fast as JMRI will allow; opening extra tables slows it down a bit. Note that signal mast drv4B and the second turnout remain steady.
Panel file with looping LT111:
[Testpanel_se8c_storm473newpanelsmlempty.xml.txt](https://github.com/JMRI/JMRI/files/981378/Testpanel_se8c_storm473newpanelsmlempty.xml.txt)

Either turning Feedback for the Se8c signal head turnout LT111 **Off** (and restarting) or choosing **NoFeedback** (immediate effect) in the turnout table Edit config feedback pane will stop turnout LT111 (and head drv5A1) from toggling. Side note: not sure what the difference is between the Off and NoFeedback choices in the Turnout Automation combobox:
[image]
Further cleaning out the panel file, I noted that removing the panel stops the looping. The contents of the panel don't matter, even a completely empty panel (of any type, even a Switchboard) will cause looping of LT111 and the Se8c head upon reloading.
Panel file without a control panel, no looping:
[Testpanel_se8c_storm473nopanel.xml.txt](https://github.com/JMRI/JMRI/files/981380/Testpanel_se8c_storm473nopanel.xml.txt)
Closer inspection of the Signal Mast Logic revealed that source mast drv4B sporting the looping signal head looks at mast drv5A but has **NO** included items to follow, so effectively there is no SML attached to that mast. This might create some undefined state, causing the looping but a mast without SML is normal as initially any new mast does not have SML until that's discovered or manually defined. Adding an item in the Edit Logic pane on mast drv5A (and restarting/reloading) stops the looping.
This looping does not occur with the same files under 4.6 and earlier, as mentioned by user.

Summarizing:
- Reloading example panel file -smlempty: looping turnout LT111 and signal head drv5A1
- No Panel: no looping
- SML with at least 1 item on the mast: no looping
- Set Turnout Feedback to Off/NoFeedback: no looping
- 4.6: no looping

bobjacobsen, 2017-05-07:
@silverailscolo Thank you!  That was really helpful.  I couldn't exactly recreate, but I think I've come close.  Could you try either the bobjacobsen/se8c-storm branch (close to current master) or the attached jmri.jar file (which should work with either 4.7.3 or the head of master) and let me know?  

Thanks again!

[jmri.jar.zip](https://github.com/JMRI/JMRI/files/981770/jmri.jar.zip)

silverailscolo, 2017-05-07:
It worked; no more looping on my _simple_ LocoNet Simulator SE8C test config! Thanks for the fix @bobjacobsen

bobjacobsen, 2017-05-08:
PR #3471 is now in the most recent [Jenkins Package](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/1508/), 1508 or later.

bobjacobsen, 2017-06-07:
See PR #3614 for an update.  See also discussion on jmri-developers.

closed 2017-07-03 by bobjacobsen