JMRI/JMRI #3465: PR #3430 Status

dsand47, 2017-05-06:
I would like PR #3430  to be included in 4.7.4.  

Since I don't seem to have the authority to do so, the WIP label can be removed.

The last Travis run timed out.  I don't know how to re-run it.

It now has a conflict.  I selected the **Resolve Conflicts** button and reviewed the listing.  The tooltip for **Mark as resolved** says to remove the conflict markers.  In this case, the section between <<< and === is empty, which makes sense, because I am deleting the method that is between === and >>>.  My guess , for this case, is that I should just delete everything from <<< to >>> and then select **Mark as resolved**.  Correct?

Dave Sand

silverailscolo, 2017-05-06:
It could might look different in other cases, but you're correct to remove all <<<, === and >>> lines and choose which of the 2 blocks should stay in.

bobjacobsen, 2017-05-06:
I went ahead and resolved the conflicts, so the tests are running now.  See comment in #3430.

bobjacobsen, 2017-05-06:
#3430 passed tests, so I've merged it and will close this shortly.

closed 2017-05-06 by bobjacobsen