JMRI/JMRI #3491: ClientRxHandlerTest does not catch an Error condition

rhwood, 2017-05-12:
The ClientRxHandlerTest appears to completing before all threads timeout, with the following result:

```
[INFO] Running jmri.jmrix.dccpp.dccppovertcp.ClientRxHandlerTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.004 s - in jmri.jmrix.dccpp.dccppovertcp.ClientRxHandlerTest
[INFO] Running jmri.jmrit.display.configurexml.IndicatorTurnoutIconXmlTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 s - in jmri.jmrit.display.configurexml.IndicatorTurnoutIconXmlTest
[INFO] Running jmri.jmrit.operations.setup.OperationsBackupGuiTest
ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main] jmri.util.exceptionhandler.UncaughtExceptionHandler.?()
java.lang.NullPointerException
	at jmri.jmrix.dccpp.dccppovertcp.ClientRxHandler.run(ClientRxHandler.java:127)
```

bobjacobsen, 2017-05-14:
Several additional cases, see in #3481:

```
    [junit] ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [ClientRxHandler:127.0.0.1] jmri.util.exceptionhandler.UncaughtExceptionHandler.uncaughtException()
    [junit] java.lang.NullPointerException
    [junit] 	at jmri.jmrix.loconet.loconetovertcp.ClientRxHandler.run(ClientRxHandler.java:129)
    [junit] WARN  - Exception: java.lang.NullPointerException [main] jmrix.loconet.locormi.LnMessageClientPollThread.?()
```

```
ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main] jmri.util.exceptionhandler.UncaughtExceptionHandler.?()
java.lang.NullPointerException
	at jmri.jmrix.can.cbus.CbusThrottle.keepAlive(CbusThrottle.java:551)
	at jmri.jmrix.can.cbus.CbusThrottle$1.actionPerformed(CbusThrottle.java:540)
	at javax.swing.Timer.fireActionPerformed(Timer.java:313)
	at javax.swing.Timer$DoPostEvent.run(Timer.java:245)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

```
ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [z21.Z21XPressNetTunnel sourceThread] jmri.util.exceptionhandler.UncaughtExceptionHandler.uncaughtException()
java.lang.NullPointerException
	at jmri.jmrix.roco.z21.Z21Message.<init>(Z21Message.java:40)
	at jmri.jmrix.roco.z21.Z21XPressNetTunnel.message(Z21XPressNetTunnel.java:211)
	at jmri.jmrix.roco.z21.Z21XPressNetTunnel.run(Z21XPressNetTunnel.java:79)
	at java.lang.Thread.run(Thread.java:745)
```

rhwood, 2018-03-16:
OP is no longer concerned.

closed 2018-03-16 by rhwood