JMRI/JMRI #3492: LocoNet RMI Client and Server create own SecurityManagers

rhwood, 2017-05-12:
The LocoNet RMI Client and Server create their own SecurityManagers, which is causing conflicts with Maven Surefire and JUnit 4, and which could also cause conflicts with any application wrapping the JMRI library.

bobjacobsen, 2017-05-31:
> The LocoNet RMI Client and Server create their own SecurityManagers, which is causing conflicts with Maven Surefire and JUnit 4, and which could also cause conflicts with any application wrapping the JMRI library.

(Moved here, to the basic issue, from discussion in #3586)

Are those the only reasons for this?

- "could also cause conflicts with any application wrapping the JMRI library" means that application can't use the RMI client and server, but only that.  Removing them means _nobody_ can. So there's no gain for anybody in addressing that.

- "causing conflict(s) with Maven Surefire" - I don't know the details of those conflicts, specifically how significant they are, so can't comment on the full range of possible solutions.  Maybe this is a separate bug Maven-related that deserves an issue of its own; we certainly have older ones.  But there's an obvious immediate workaround:  In the Maven Surefire environment, don't run the RMI Server and Client tests.  People can and do still run tests outside the Maven environment to find problems, particularly if they touch code related to this. I did that this past weekend while working on PR #3580.

- "causing conflicts with ... JUnit 4" - this might be a bigger issue, though I don't know the details.  Both the `locormi.LnMessageServerTest` and `locormi.LnMessageClientTest` are in JUnit 4 format and work fine in my development environment.  What exactly is broken here?

If those are the only reasons, it seems to me that this is a really low priority. Maybe somebody (@rhwood) wants to _fix_ the problems, and that's fine; additions are welcome. 

But these problems, unless there's a lot more here that I'm not seeing, don't rise to the level of changing a long-standing practice of not removing (with or without a migration) a long-functioning feature from the code.

rhwood, 2017-05-31:
The reasons enumerated above are not the only reasons; they are, however, the only reasons I looked at the LocoNet Server to begin with.

The reasons to migrate users to the LocoNet Over TCP server are:
- The original author has asked in 2008 if we could do what #3586 proposes, although he apparently has never received a response one way or the other.
- RMI requires a security manager and a security policy be instantiated. If any other security manager is in place already, or a conflicting policy is in place, this can cause issues.
- The security manager and RMI code in general are difficult to work with, especially if JMRI is a lib that can be embedded in another application (is it anymore)?
- The LocoNet Server only supports physical LocoNet connections to a DCS 100.

bobjacobsen, 2017-05-31:
OK, let's go through those:

> The original author has asked in 2008 if we could do what #3586 proposes, although he apparently has never received a response one way or the other.

That's not a reason, by itself,  to take the affirmative action of removing something that people have used and may still be using.

> The LocoNet Server only supports physical LocoNet connections to a DCS 100.

Ditto.

And the other two were addressed above.

You may not like this code, but that's a different thing from removing it.  We don't do that.

rhwood, 2018-03-16:
OP is no longer concerned.

closed 2018-03-16 by bobjacobsen